tsduck/tsduck #1162: [BUG] debian package 3.32 not supported on bullseye yet (no zstd support)

ngaullier, 2023-01-20:
The debian package 3.32 uses zstd which is supported on ubuntu but only extremely recently on debian.

Please see
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892664#129

Note that previous package v3.31 is ok (this is my turnaround for now).

Today, my bullseye says:
>dpkg-deb --help|grep 'types'
                                     Allowed types: gzip, xz, none.
=> I miss zstd yet, and it might take some time until it is supported in bullseye I suppose.

Nicolas

lelegard, 2023-01-21:
Hi @ngaullier 

> Today, my bullseye says:
> dpkg-deb --help|grep 'types'
> Allowed types: gzip, xz, none.

Mine says the same and the TSDuck package `tsduck_3.32-2983.debian11_amd64.deb` (and `-dev`) installs flawlessly. This is a standard Debian 11 bullseye with up-to-date packages.

I do not know what you mean by "debian package 3.32 uses zstd". What is your exact problem?

I have 10 different virtual machines with 10 different Linux distros (plus 2 Windows VM and 4 BSD VM, all distinct flavors). I cannot master all the gory details of each of them. They are "standard" installations with all prerequisites to build TSDuck (see script `install-prerequisites.sh`) and, most of the time, up-to-date packages. That's all.

I do my best to provide pre-built binaries for official releases on the latest version of the 4 major distros (Ubuntu, Debian, Fedora, Redhat). For any other version or distro, you need to rebuild.

In case of problem, I recommend to [rebuild from the repo](https://tsduck.io/doxy/building.html). The .deb or .rpm packages (depending on the distro) are built using `make installer`. For other distros (neither .deb nor .rpm), you need to `make install`, directly.

ngaullier, 2023-01-23:
Hi,
First, I have to thank you for all your work, including indeed all the available pre-built binaries, this is a great effort, very rarely seen elsewhere.
I think this issue is minor in that it may be fixed in the coming months (without any action from you), but it may be of interest for users encountering the same problem. I could also have build from the sources (I did not have because of missing disk space on my systems). So, you can close this ticket, for sure, this is so much you already have done.

So just for information, and to dig into the details:
>ar x ./tsduck_3.32-2983.debian11_amd64.deb
extracts 'control.tar.xz' and, as the file extension presumes:
>file ./control.tar.xz 
./control.tar.xz: XZ compressed data

Whereas with v3.33:
>ar x ./tsduck_3.33-3139.ubuntu22_amd64.deb
extracts 'control.tar.zst' and:
>file ./control.tar.zst 
./control.tar.zst: Zstandard compressed data (v0.8+), Dictionary ID: None

When I run 'sudo apt-get install ./tsduck_3.33-3139.ubuntu22_amd64.deb', I get this:
dpkg-deb: error: archive '/home/xxx/tsduck_3.33-3139.ubuntu22_amd64.deb' uses unknown compression for member 'control.tar.zst', giving up

And here it is:
>dpkg-deb --version
Debian 'dpkg-deb' package archive backend version 1.20.12 (amd64).

It seems zstd was added very recently on debian on version 1.21.18:
https://github.com/guillemj/dpkg/commit/ebc2c3def335ac391b010323ab65a302648f3ec7

I am definitely not a distribution expert, so I don't catch how it is possible to get in such a situation...
Anyway, it is good to know about it.

Thank you again for your great tool
Nicolas

lelegard, 2023-01-23:
Is your Debian 11 "up to date"?

My Debian VM is also a Debian 11 bullseye and I do not see the same problem.  Do you have the opportunity to update your system (`sudo apt update; sudo apt upgrade`)?

Moreover, your initial post mentions that 3.32 uses zstd and does not work while 3.31 works. Now your latest post (after I just released 3.33 yesterday) mentions that 3.32 uses xz (and 3.33 uses zstd and does not work). Could you please clarify about 3.32?

lelegard, 2023-01-23:
>  I could also have build from the sources (I did not have because of missing disk space on my systems). 

It is true that the history of the tsduck repo is quite big. This is due to my original mistake of having included Microsoft Office files as regular git files instead of using git-lfs for them.

If you have at least 200 MB free disk, use `git clone --depth 1` to grab the latest state only, skipping the history. The repo size is then about 50 MB only. The complete build process should not use more than 150 MB, probably less.

ngaullier, 2023-01-23:
I am very sorry, it seems I confused the debian and ubuntu packages since I can't reproduce the problem anymore!
I have re-downloaded/installed the latest v33 debian on both my debian VMs with no problem.
Thank you for the git clue, I am still interested in compiling anyway.
Really sorry for having having wasted your time.
I will just try to close the ticket properly now.

closed 2023-01-23 by ngaullier