tsduck/tsduck #1163: [BUG] Fedora 37 libsrt version mismatch

DougLung2000, 2023-01-26:
#### Bug description:
tsduck-3.33-3139.fc37.x86_64.rpm fails to install on Fedora 37 updated as of 25 Jan 2023

#### How to reproduce:
sudo dnf install tsduck-3.33-3139.fc37.x86_64.rpm 

#### Expected behavior:
tsduck should install without errors

#### Errors and logs:
~~~
sudo dnf install tsduck-3.33-3139.fc37.x86_64.rpm 
Last metadata expiration check: 1:02:53 ago on Wed 25 Jan 2023 20:26:08 HST.
Error: 
 Problem: conflicting requests
  - nothing provides libsrt.so.1.4()(64bit) needed by tsduck-3.33-3139.fc37.x86_64
bash-5.2$ sudo ls -l /usr/lib64/libsrt.*
lrwxrwxrwx. 1 root root     13 Oct  2 21:56 /usr/lib64/libsrt.so -> libsrt.so.1.5
lrwxrwxrwx. 1 root root     15 Oct  2 21:56 /usr/lib64/libsrt.so.1.5 -> libsrt.so.1.5.1
-rwxr-xr-x. 1 root root 906808 Oct  2 21:56 /usr/lib64/libsrt.so.1.5.1
~~~

#### Environment:
 - OS:  Fedora 37 Linux
 - OS version: 6.1.7-200.fc37.x86_64
 - TSDuck full version: 3.33-3139
 - Installation type: official binary tsduck-3.33-3139.fc37.x86_64.rpm

#### Additional information:
Thanks for offering a Fedora 37 binary. Hope it can be updated to use the current Fedora 37 libsrt version

lelegard, 2023-01-26:
Hi @DougLung2000 

Thanks for reporting this. However, it is not as straightforward as it seems.

On my Fedora 37, a regular installation without exotic tweak and regular updates, I only see SRT 1.4 in the updates:
~~~
# dnf list 'srt-*'
Last metadata expiration check: 0:13:13 ago on Thu 26 Jan 2023 08:42:16 PM CET.
Installed Packages
srt-devel.x86_64                                                               1.4.4-2.fc36                                                                        @fedora
srt-libs.x86_64                                                                1.4.4-2.fc36                                                                        @fedora
Available Packages
srt-devel.i686                                                                 1.4.4-2.fc36                                                                        fedora 
srt-libs.i686                                                                  1.4.4-2.fc36                                                                        fedora 
~~~

However, if I browse to some Fedora mirror, [here for instance](http://ftp.free.fr/mirrors/fedora.redhat.com/fedora/linux/releases/37/Everything/x86_64/os/Packages/s/), I see this:
~~~
srt-1.5.0-2.fc37.x86_64.rpm             2022-Jul-23 14:36:46    1.0M     application/octet-stream
srt-devel-1.5.0-2.fc37.i686.rpm         2022-Jul-23 14:36:43    36.4K    application/octet-stream
srt-devel-1.5.0-2.fc37.x86_64.rpm       2022-Jul-23 14:36:45    36.3K    application/octet-stream
srt-libs-1.5.0-2.fc37.i686.rpm          2022-Jul-23 14:36:43    393.3K   application/octet-stream
srt-libs-1.5.0-2.fc37.x86_64.rpm        2022-Jul-23 14:36:44    358.2K   application/octet-stream
~~~

I will investigate...

lelegard, 2023-01-26:
For some reason I cannot explain, `dnf` continued to point to Fedora 36 repos while the system identifies itself as Fedora 37 and has all gnome desktop appearance of Fedora 37 and most - but not all - packages are identified as `-fc37`. The upgrade from 36 to 37 went smoothly, as far as I remember. But the system is in some kind of hybrid state. I will fix this and rebuild TSDuck packages.

lelegard, 2023-01-26:
OK, I reinstalled and rebuilt TSDuck. This time, SRT 1.5 is up to date. The packages have been updated and you may retry.
Sorry for the inconvenience.

closed 2023-07-02 by github-actions[bot]