tsduck/tsduck #1164: [Q]

zbynekdrlik, 2023-01-30:
Hi, starting tests with tsduck. Trying send srt stream from vmix to tsp and then forward by rist. 
I am struggling that tsp always exit in few seconds after vmix connect

C:\Users\Video-PC>tsp --verbose  --debug=9 -I srt --statistics-interval=555 -l 7432 -O rist rist://185.14.97.115:9998
tsp: Debug[9]: debug level set to 9
tsp: Debug: ====> tsp --verbose --debug=9
* Debug[9]: debug level set to 9
* Debug[9]: srt: debug level set to 9
* Debug: srt: ====> tsp -I srt --statistics-interval=555 -l 7432
* Debug[9]: rist: debug level set to 9
* Debug: rist: ====> tsp -O rist rist://185.14.97.115:9998
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: srt: calling srt_create_socket()
* Debug: srt: calling srt_setsockflag(SRTO_SENDER, 01, 1)
* Debug: srt: calling srt_setsockflag(SRTO_TRANSTYPE, 00 00 00 00, 4)
* Debug: srt: calling srt_setsockflag(SRTO_MESSAGEAPI, 01, 1)
* Debug: srt: calling srt_setsockflag(SRTO_REUSEADDR, 01, 1)
* Debug: srt: calling srt_bind(0.0.0.0:7432)
* Debug: srt: calling srt_listen()
* Debug: srt: calling srt_accept()
* Debug: srt: connected to 127.0.0.1:60954
* Debug: srt: SRTSocket::open, sock = 0x038B3F40, listener = 0x038B3F41
* srt: SRT statistics:
  Total received: 927,156 bytes, 772 packets, lost: 0 packets, dropped: 0 packets
  Interval received: 927,156 bytes, 772 packets, lost: 0 packets, dropped: 0 packets
  Timestamp-based delivery delay, receive: 3000 ms, RTT: 0.023000 ms
* Debug: srt: initial buffer load: 4 packets, 752 bytes
* srt: unknown initial input bitrate
* Debug: rist: calling rist_sender_create, profile: 1

and exit to cmd

Appreciate any help.

zbynekdrlik, 2023-01-31:
I see that issue is with output rist part. If I put output to drop then cmd is running correctly.

zbynekdrlik, 2023-02-04:
I had similar issue with ffmpeg but after upgrade to nightly ffmpeg build it start working and now I could use ffmpeg for rist. Tsp has still the same behavior on sender side also if I upgrade to nightly build too.

lelegard, 2023-02-08:
Which version of RIST does your latest ffmpeg use when the problem is gone?

The latest release of librist is v0.2.7 and is 10 months old. See https://code.videolan.org/rist/librist/-/releases

This is the version which is used in recent versions of TSDuck. Try `tsp --version=rist` to check.

If ffmpeg uses some not yet released snapshot of librist which fixes a bug, you need to know which one, typically a commit SHA1. Since you seem to use Windows, you will have to rebuild librist using this project: https://github.com/tsduck/rist-installer

The `build.ps1` procedure for Windows has a `-Tag` option to specify a tag or commit from the librist repository. Install this custom version of librist and then rebuild TSDuck.

zbynekdrlik, 2023-02-08:
Hi, I have installed this version 

tsp:
librist version v0.2.7, API version 4.2.0

ffmpeg:
https://github.com/BtbN/FFmpeg-Builds/releases/tag/autobuild-2023-02-04-12-37