tsduck/tsduck #1167: [Q] Is there a way to obtain the EPG data from a transport stream?

alvarollgo, 2023-02-08:
Hi all,

I want to get the EPG data from a transport stream. And the idea would be to be able to save it in an xml file, for example. Is there a way to do it?

Thanks!

lelegard, 2023-02-08:
It depends on the standard: DVB, ATSC or ISDB.

For DVB, just grab all individual EIT sections from PID 18 and save them as XML. For instance:
~~~
tstables file.ts --pid 18 --pack-all-sections --xml events.xml
~~~
Or use the plugin `tables` in the middle of a `tsp` command.
Please RTFM for more details on options.

alvarollgo, 2023-02-10:
Thank you @lelegard  for your quick response. That's right with what you suggested. Finally I did it with the plugin `tables` and the tsp command. Thank you very much!

closed 2023-07-23 by github-actions[bot]