tsduck/tsduck #1168: biss-ca decryption

sojalvo, 2023-02-08:
HI all,
I want to use the TSDuck library to write code that decrypt a mpeg stream encrypted with biss-ca technology
I retrieved the PID of EMM and ECM packets respectively from CAT and PMT tables
My question is: which plug-in I can use to retrieve the encrypted SK and SW data from the EMM and ECM packets?
Can you share an example?
Thank you in advance
Sergio.

lelegard, 2023-02-08:
There is no public plugin to handle TS descrambling based on a given proprietary CAS.

BISS-CA is probably a bit different because this is a public standard and an open-source implementation is technically possible. However, such an implementation would require access to the private RSA key of an entitled receiver. I assume that you have one, don't you?

This private RSA key shall be carefully protected in adequate receiver systems. TSDuck being a fully open portable tool for any operating system, the private RSA key would then be stored on some general-purpose computer. If I were a rights holder (a sports federation for instance), I would immediately revoke the key of a partner who is known to casually store its private on general-purpose computers. But I am not such a rights holder. Maybe they do not care about security. Who knows?

Anyway, I do not want to be involved in such poor security management.

sojalvo, 2023-02-09:
Thank you for your response!

Of course I have a private key and is secured.
My question is how to receive the EMM section data (encrypted! before opening it by the private key!) from the packets.
Something parallel to:
**tsp -I ip 1234 -P tables --json-output emm.json --rewrite-json --tid 0x81 -O drop**
that parse the emm into a json file, but in c++ code using the library

lelegard, 2023-02-09:
The "right" way would be to write a plugin which filters the CAT, then filter the interesting EMM PID. You will get all EM sections one by one. Many plugins work this way. Just have a look at the source code.