tsduck/tsduck #1170: [Q] TSDuck and Windows10 LTSB

EvariSTC, 2023-02-09:
Is TSDuck compatible with Windows 10 LTSB? Whatever exe I run, it doesn't start giving this error message.
[image]
The "missing" DLL is in the folder.

Regards.

lelegard, 2023-02-09:
Hi @EvariSTC 

I do not know what is Windows LTSB precisely. Somewhere I read that this is just a license & support model. Somewhere else, I read that it has technical limitations. I never used it.

I understand only globally the error message (is this Italian?) but I miss the details. This is clearly not a missing DLL issue. This is a missing entry point SetThreadDescription which is called by tsduck.dll. The DLL tsduck.dll is correctly found, then Windwos tries to find the external DLL's which are called by tsduck.dll. SetThreadDescription is a Windows system function and it is called from tsduck.dll. 

Note that the error does not mention a missing DLL. All system DLL which are required by TSDuck are found. But the function is not found in those DLL. According the [Microsoft doc](https://learn.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-setthreaddescription), SetThreadDescription is located in Kernel32.dll, one of the most essential DLL's in Windows. It must be there. Otherwise, nothing would work. However, for some reason, SetThreadDescription is not present on your kernel32.dll. Maybe this is a limitation of Windows LTSB.

lelegard, 2023-02-09:
Look here; https://learn.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-setthreaddescription

> Windows Server 2016, Windows 10 LTSB 2016 and Windows 10 version 1607: SetThreadDescription is only available by [Run Time Dynamic Linking](https://learn.microsoft.com/en-us/windows/win32/dlls/using-run-time-dynamic-linking) in KernelBase.dll.

This means that SetThreadDescription cannot be found in those Windows versions when the .exe is activated. The application needs to search it at runtime using GetProcAddress. What a crazy idea. Maybe this is not due to LTSB itself, but that "long term" means "obsolete Windows version" in practice.

It is true that calling SetThreadDescription was added upon a TSDuck user's request to explicitly identify each thread in tsp. I do not remember the date, but it was probably some time around 2018 or 2020. Maybe older versions of Windows 10 did not have this function.

closed 2023-07-23 by github-actions[bot]