tsduck/tsduck #1171: Can't select 32K carriers for DVB-T2 modulation in VATEK output plugin option "--fft-mode"

skunge2000, 2023-02-11:
#### Bug description:
Despite 32K being declared as the default for --fft-mode setting in the vatek output mode - I can't explicitly select 32K as an option?

When trying to create an Ofcom compatible DVB-T2 mux I get an error.

#### How to reproduce:
tsp -O vatek --frequency 474000000 -bandwidth 8 --fft-mode "32K" --plp0-modulation "256-QAM" --plp0-code-rate "2/3" --t2-guard-interval "1/128"  

Gives an error when I don't think it should. 

#### Expected behavior:
I expect to be able to select 32K mode for DVB-T2 modulation 

#### Errors and logs:
~~~
* Error : vatek: invalid value 32K for option --fft-mode, use one of @"16K", "1K", "2K", "4K", "8K"
~~~

#### Environment:
 - OS: Raspberry Pi OS 
 - OS version: 32-bit v5/15/89-v7l+
 - TSDuck full version: 3.32-2983
 - Installation type: official binary deb

#### Additional information:
Looks like there is code missing here for 32K :

https://github.com/tsduck/tsduck/blob/ead1b63822247096efaa95dcfb922840f5647dcd/src/libtsduck/vatek/tsVatekOutputPlugin.cpp#L274

lelegard, 2023-02-11:
@a83912a, could you have a look on this one?

a83912a, 2023-02-11:
@lelegard  OK. I'll take care of it

a83912a, 2023-02-13:
@skunge2000 Our Modulator Chip doesn't support 32K DVB-T2, the plugin reports 32K is the default was a typo.  We'll delete the description in plugin and upload the corrected version. Sorry for misleading.

skunge2000, 2023-02-13:
@a83912a Thanks for this, apologies, I wasn't aware of that limitation of the modulator in the U3 I'm using.

closed 2024-01-21 by github-actions[bot]