tsduck/tsduck #1172: [BUG] http does not support HTTP/0.9

fe31nz, 2023-02-13:
When connecting to SAT>IP to my minisatip server, I get this:

~~~
root@lith:/tmp# tsp -v -I http "http://satip.jsw.gen.nz:554/?src=1&freq=12358&msys=dvbs2&plts=off&fec=23&pol=h&ro=0.35&sr=22500&mtype=8psk&pids=0,1,307,1007,1107,1907" -P analyze -o drop
* Error: http: download error, Unsupported protocol, Received HTTP/0.9 when not allowed
~~~

Using curl with the same URL produces the same error message:

~~~
root@lith:/tmp# curl "http://satip.jsw.gen.nz:554/?src=1&freq=12358&msys=dvbs2&plts=off&fec=23&pol=h&ro=0.35&sr=22500&mtype=8psk&pids=0,1,307,1007,1107,1907" -o UKTV.ts
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (1) Received HTTP/0.9 when not allowed
~~~

With curl, adding the --http0.9 option allows the connection to work:

~~~
root@lith:/tmp# curl --http0.9 "http://satip.jsw.gen.nz:554/?src=1&freq=12358&msys=dvbs2&plts=off&fec=23&pol=h&ro=0.35&sr=22500&mtype=8psk&pids=0,1,307,1007,1107,1907" -o UKTV.ts
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    97    0    97    0     0     30      0 --:--:--  0:00:03 --:--:--    30^C
~~~

Given the similarity of the error messages, I am guessing that TSDuck is using the same http library as curl does.  If so, then it needs a way of adding the --http0.9 option, or maybe it should always be passing this option to the library.

#### Environment:
 - OS: Ubuntu
 - OS version: Ubuntu 22.04.1 LTS
 - TSDuck full version: 3.34-3144
 - Installation type: Compiled from latest git to .deb files.
 - minisatip version 1.2.72~ad307a0

lelegard, 2023-02-13:
Hi @fe31nz 

Thanks for reporting this. Indeed, TSDuck uses libcurl on all Unix systems (Linux, macOS, *BSD) for all HTTP requests. The similarities in error reporting is not fortuitous. On Windows, it uses the native WinInet library.

However, I assume there is an error in your scenario, maybe a bug in your minisatip server or in the URL. The `http` input plugin connects to a server providing binary TS packets (all TSDuck input plugins are supposed to extract TS packets from their source). According to this [w3.org article](https://www.w3.org/DesignIssues/HTTP0.9Summary.html) or this [other one](https://medium.com/platform-engineer/evolution-of-http-69cfe6531ba0), HTTP 0.9 is a single-line protocol which can return HTML documents only. A server implementing HTTP 0.9 cannot return anything else, not binary data, not TS packets.

This is why there is no need for an `--http0.9` option in a TSDuck input plugin because TS packets cannot be returned over HTTP 0.9.

When you connect to your server using `curl --http0.9`, what is the content of the response? Is this some HTML text containing an error message (if the URL contains some invalid tuning info)? Or does it return binary TS packets? In the latter case, this is a bug of the server which is not allowed to do so.

fe31nz, 2023-02-15:
Thanks for your comments.  When I went back and tried this again, it was obvious what I had done wrong.  In order to use my Sat>IP tuners with TSDuck (or curl), I need to use http protocol instead of Sat>IP RTSP protocol.  When I copied the URL, I changed the protocol, but forgot to change the port number from 554 (Sat>IP) to 8080 for minisatip's http port.  Changing the port makes it work.

closed 2023-02-15 by fe31nz