tsduck/tsduck #1173: [Q] How to check if the ts multicast stream is descrambled properly?

krystiankgb, 2023-02-14:
Hi,

I use different hardware to descramble services from satellite (S2 -> multicast) but they sometimes fail and stream with scrambled multicast. 

I use another software that detect pes errors and restarts that hardware.

Is it possible with TSduck?

lelegard, 2023-02-15:
Hi @krystiankgb 

TSDuck may detect many different kinds of situations in TS packets. You have to translate your need in precise technical terms about TS packets and then TSDuck will probably find a way to detect it.

For instance, saying _"fail and stream with scrambled multicast"_ may mean different kinds of situations:

- Does the equipment completely fail to process scrambled packets? In that case, you need to detect TS packets with a transport scrambling control value which is different from zero.
- Does the equipment still process the packets but the descrambling fails? In that case, the packets may have a clear "scrambling control" and the previous test does not apply. However, you can detect invalid start of PES packets, i.e. packets with "payload unit start indicator" set but the payload does not start with 00 00 01.

All these checks can be performed with `tsp` using combinations of `filter` plugins and labels. At the end of this chain, when the TS is correctly descrambled, no packets are selected. So, add a `bitrate_monitor` plugin at the end of the chain on the selected packets. This plugin can have "alarm" commands which are run when the bitrate goes outside certain limits. In your case, the maximum bitrate is zero, or a very low value.

krystiankgb, 2023-02-15:
Hi @lelegard,
Thank you for the replay.

Option 1: the packets are descrambled properly but the equipment does not set the scrambling control properly - sometimes it's "even" sometimes "odd".

Option 2: The equipment keeps sending scrambled packets (so the bitrate is the same) but the packets have scrambling control set "clear". 

An example:
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|  Transport Stream Id: ....... XXXXX   |  Services: .............. 1  |
|  Bytes: ........................... 613,256     |  PID's: Total: .......... 7  |
|  TS packets: ........................ 3,262   |         Clear: .......... 7  |
|     With invalid sync: .................. 0    |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |          With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|  Broadcast time: ..................................... 2 sec (0 min 2 sec)  |
|  Srv Id  Service Name                             Access          Bitrate  |
|  XX  ChannelName ................................. C    2,135,329 b/s  |
|  Note 1: C=Clear, S=Scrambled                                                |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |

Can you advice me how to do that kind of checks to determine if descrambling process is done properly or not?
Which plugins should I use?

lelegard, 2023-02-15:
Hi @krystiankgb 

Here is a start. For more details, you need to read the user's guide for a complete description of all possible plugins and options.

~~~
tsp -I ... \
    -P filter --audio --video \
    -P filter --negate --clear --set-label 1 \
    -P filter --clear --set-label 2 \
    -P filter --only-label 2 --unit-start --set-label 3 \
    -P filter --only-label 3 --negate --pattern 000001 --search-payload --search-offset 0 --set-label 4 \
    -P filter --label 1 --label 4 \
    ....
~~~

The principles to remember with the `filter` plugins:
- Several filtering options in the same instance of `filter` mean "or". It filters packets matching any of the options. All other packets are removed.
- Option `--negate` is a final "not" on the filtering condition
- Option `--set-label` does not remove packets, it simply marks filtered packets with a label.
- Option `--only-label`, valid for any plugin, means that the plugin processes only the packets with the specified label. Other packets are transparently passed by `tsp` without being passed to the plugin.

As a consequence, in the partial command above,
1. The first filter only keeps packets from audio or video PID's. All other packets are dropped.
2. The next one sets label 1 on scrambled packets (not clear). This is your first invalid condition.
3. The next one sets label 2 on clear packets.
4. The next one processes clear packets only (`--only-label 2`) and sets label 3 on packets with a unit start indicator (start of a PES packet).
5. The next one processes clear packets with unit start only (`--only-label 3`) and sets label 4 on packets which do not have 00 00 01 at the beginning of their payload. This is your second invalid condition.
6. The last filter only keeps packets with any of your invalid conditions and removes all others.

This means that this command removes all packets from a valid clear TS. Any packet going out of this command is an indication of something going wrong according to your invalid conditions.

This may look complicated but it is very logical, with precise filtering conditions. Of course, as usual with `tsp`, put this in a shell script, do not try to type it by hand on a terminal window !

krystiankgb, 2023-02-18:
Hi @lelegard,

First I got familiar with the manual and tested the tsduck tools and was impressed - they are very powerful. 

Your clever way to catch the packet works perfectly, I collected packets and dvb_inspector showed me payload_start:

> 0001 0203 0405 0607 0809 0A0B 0C0D 0E0F 0123456789ABCDEF
> 0x000000 474A 8D1E **A53A 09**73 D5FE 05E6 ABF0 601C GJ...:.s......`.

So I can now identify the "broken" payload_start packets.

The question is: how can I use that? I use tsp to collect CC errors and jitter to my log file and then I parse it and create the graph/notifies/etc.
Is it possible to create a log message with the information f.e. "pes error"?

Thanks a lot for your help.

lelegard, 2023-02-18:
> Is it possible to create a log message with the information f.e. "pes error"?

In my previous example, considering that only invalid packets are passed at the end of the chain, add the following plugin to log a message for each invalid packet:
~~~
    -P count --all --tag 'PES error' \
~~~
Additionally, if you want to automatically restart your failing device on invalid packets, but not more often than every 5 seconds, for instance, to avoid restarting 20 times on a burst of 20 invalid packets, add also this one:
~~~
    -P trigger --min-inter-time 5000 --execute "some shell command that restarts your device" \
~~~
RTFM for the details or more refinements...