tsduck/tsduck #1174: Trimming a TS file

abacklundtvmedia, 2023-02-16:
#### Is your feature request related to a problem?
I want to be able to trim a TS file, for example remove X seconds in the beginning and remove X seconds/minutes/hours from the end. Clean and simple

#### Proposed solution:
Be able to set in and out using timecode similar to ffmpeg

#### Alternatives:
ffmpeg has this ability however it won't retain the SCTE-35 track correctly so most likely a bug in ffmpeg.

lelegard, 2023-02-17:
Hi @abacklundtvmedia 

TSDuck processes a TS, it processes TS packets, not the embedded audio and video (except for a few specialised usages). TSDuck can remove packets at the beginning and the end of a file. See the plugins `skip` and `until`. You need to know the number of packets to remove. If the TS is CBR or not-too-variable VBR, it is easy to convert the duration into packets using the TS bitrate. I have done this many times in scripts. First, let `tsanalyze` evaluate the bitrate. In a script you can collect it using `grep` and `sed` with "normalized" output or using `jq` with JSON output. Then, convert the requested durations into packet counts and remove them.

> ffmpeg has this ability however it won't retain the SCTE-35 track correctly so most likely a bug in ffmpeg.

Precisely because ffmpeg is very good at audio/video processing and very bad at TS processing. The strengths of ffmpeg are the weaknesses of TSDuck and vice-versa.

That being said, since TSDuck is often used in combination with ffmpeg and since ffmpeg is known to destroy everything it does not understand in the TS, there is a way to encapsulate and restore data PID's to preserve them through ffmpeg processing. See the example in section 5.2.28 of the TSDuck user's guide. It preserves the EIT's across ffmpeg processing. You can use a similar principle to preserve SCTE-35 or any other data PID.

abacklundtvmedia, 2023-02-23:
Thank you for the suggestion and explanation 🙏🏻

Are there by the way any real-world examples (that are described somewhere) out there how people tend to use ffmpeg in combination with TSduck?

closed 2024-08-25 by github-actions[bot]