tsduck/tsduck #1175: [BUG] eitinject incorrectly creates following event after event change

przemo1983, 2023-02-20:
#### Bug description:
I have 2 EIT section definitions as follows:
When I start the tool the EIT xml file definition is as follows (present, following and schedule events):
```
<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
	<EIT type="pf" version="1" actual="false" current="true" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42048" start_time="2023-02-20 11:00:00" duration="00:15:00" running_status="running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1200]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42051" start_time="2023-02-20 11:15:00" duration="00:15:00" running_status="not-running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1215]</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
	<EIT type="0" version="1" actual="false" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42050" start_time="2023-02-20 11:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1230]</event_name>
			</short_event_descriptor>
		</event>
...
```
After 15 minutes the EIT xml file is changed correctly to following (1st event schedule becomes following, following event becomes present one):
```
<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
	<EIT type="pf" version="2" actual="false" current="true" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42051" start_time="2023-02-20 11:15:00" duration="00:15:00" running_status="running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1215]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42050" start_time="2023-02-20 11:30:00" duration="00:15:00" running_status="not-running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1230]</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
	<EIT type="0" version="2" actual="false" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42049" start_time="2023-02-20 11:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|20|1245]</event_name>
			</short_event_descriptor>
		</event>
...
```

The first XML file is generated correctly in TS file:
**present**: event id: 42048 - start time 11:00:00 - duration: 00:15:00 - running status: 4 (running)
**following**: event id: 42051 - start time 11:15:00 - duration: 00:15:00 - running status: 1 (not running)
**schedule**: event id: 42050 - start time 11:30:00 - duration: 00:15:00 - running status: 0 (undefined)

**BUT**

After 15 minutes the 2 XML file is generated incorrectly as:
**present**: event id: 42051 - start time 11:15:00 - duration: 00:15:00 - running status: 4 (running)
**following**: event id: 42051 - start time 11:15:00 - duration: 00:15:00 - running status: 1 (not running)

**So the present and following events after the EIT xml file change are with same event id and with same time.** Following event id should have all data from 1st schedule instead.

#### How to reproduce:
Run command as follows:
`    command_line = "tsp \
--bitrate " + str(ts_bitrate) + \
" -I file " + current_directory + "/fifo/first.ts" + \
" -P timeref --system-synchronous \
-P eitinject -f \"" + current_directory + "/eits/eit*.xml\"" \
" -P regulate \
 -O ip " + ip + ":" + str(port) + " &"`

#### Environment:
 - OS version: Ubuntu 20.10
 - TSDuck full version: 3.31-2764

lelegard, 2023-02-20:
Hi @przemo1983 

Thanks for reporting this. Are you sure about the version 3.31-2764 ? The final version 3.31 was -2761 and -2764 was an intermediate version of 3.32.

The current version is [3.33-3139](https://github.com/tsduck/tsduck/releases/tag/v3.33-3139). Could you please try to reproduce your issue with it?

If you use Ubuntu 20, you may need to rebuild the installer since the official binary was provided for Ubuntu 22.10, the latest one. The details are [here](https://tsduck.io/doxy/building.html). The procedure is straightforward:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh  # before building tsduck for the first time
make installer
~~~

If the bug is confirmed, I need a way to reproduce it in order to fix it. Could you please provide the exact command and input files to reproduce it?

lelegard, 2023-02-20:
And by the way, how do you collect the generated EIT XML files?

EIT's are not regular tables, they are sparse sections which should be monitored one by one, without attempting to rebuild complete tables. Otherwise, there is a risk of misinterpretation, aggregating a "present" EITp/f section which _follows_ a "following" section which were sent exactly before and after the event starting time, giving the impression that the event is present twice in the "table", while in fact the "table" does not exist as such.

The way you present the events looks like you did not log EIT sections one by one, thus possibly exhibiting the misleading behaviour I just described. When capturing EIT's, be sure to use option `--all-sections` in text or binary mode and `--pack-all-sections` in XML or JSON format. Please RTFM for more details.

Please retry capturing the XML EIT sections with option `--pack-all-sections` and check if there is an inconsistency in the event scheduling.

przemo1983, 2023-02-21:
OK, I changed the definition of the EIT sections.
All events are now signaled as schedule and put in 3h segments.
I am using latest TSduck version on windows: 3.33-3139

See XML of EIT section:
```
<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
	<EIT type="0" version="6" actual="false" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42148" start_time="2023-02-21 11:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1200]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42151" start_time="2023-02-21 11:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1215]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42150" start_time="2023-02-21 11:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1230]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42149" start_time="2023-02-21 11:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1245]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42152" start_time="2023-02-21 12:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1300]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42155" start_time="2023-02-21 12:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1315]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42154" start_time="2023-02-21 12:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1330]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42153" start_time="2023-02-21 12:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1345]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42156" start_time="2023-02-21 13:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1400]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42159" start_time="2023-02-21 13:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1415]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42158" start_time="2023-02-21 13:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1430]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42157" start_time="2023-02-21 13:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>[]Test Event [2023|02|21|1445]</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
</tsduck>
```
And I see that the EIT p/f and schedule are generated at desired time.
But in case of event present there is running status set to 0 (undefiled) and for following event is the same value 0.

Our device starts recording basing on running-status field (running for current event, not running for following). Could you please fix it? 

I am running eitinject plugin with following command:
`... -P eitinject -f eit.xml ...`

closed 2023-02-21 by przemo1983