tsduck/tsduck #1176: [Q] EITinject plugin make a mess with EIT schedule?

przemo1983, 2023-02-22:
I have a EIT schedule definition as follows (events generated every 15 minutes, packed into 3h segments, from the midnight):
```
<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
	<EIT type="0" actual="false" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="40224" start_time="2023-02-22 00:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0100]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="40227" start_time="2023-02-22 00:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0115]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="40226" start_time="2023-02-22 00:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0130]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="40225" start_time="2023-02-22 00:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0145]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="40228" start_time="2023-02-22 01:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0200]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42211" start_time="2023-02-22 01:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0215]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42210" start_time="2023-02-22 01:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0230]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="40229" start_time="2023-02-22 01:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0245]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42212" start_time="2023-02-22 02:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0300]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42215" start_time="2023-02-22 02:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0315]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42214" start_time="2023-02-22 02:30:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0330]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42213" start_time="2023-02-22 02:45:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0345]</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
	<EIT type="0" actual="false" service_id="101" transport_stream_id="9" original_network_id="70">
		<event event_id="42216" start_time="2023-02-22 03:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|22|0400]</event_name>
[... 634 lines cut ...]
				<event_name>Test Event [2023|02|23|1045]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42344" start_time="2023-02-23 10:00:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|23|1100]</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="42347" start_time="2023-02-23 10:15:00" duration="00:15:00" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Test Event [2023|02|23|1115]</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
</tsduck>
```
After running EITinject plugin with that EIT definition the EIT schedule looks good. All EIT schedule sections have version 0 and correct event times. All are divided of 3h segments as presented in picture below (the last one matches the EIT XML file structure):
[image]

But after 15 minutes of working of EITinject plugin when EITs are being changed I see some strange behavior in EIT schedule. All segments but the segment no 24 have still version 0 and segment no 24 has new version. This invalidates all rest events in the EPG. See below picture.
[image]

At the same time there is no problem with EIT present/following generation. Every 15 minutes all events p/f changes their version.

What am I doing wrong? Is my EIT XML file definition wrong? Plugin has some issues?

lelegard, 2023-02-22:
> But after 15 minutes of working of EITinject plugin when EITs are being changed by the plugin (EIT XML file definition is still the same as at time when plugin has been started) I see some strange behavior in EIT schedule. All segments (apart from segment no 24) have still version 0 and only segment no 24 has new version (with only 4 events in it).

I haven't looked in the details of your EIT before and after the change, but it sounds like a normal behaviour.

Since all your test events are calibrated to 15 mn, one event disappears from the schedule, leading to an update of the EIT/schedule section for that period of time on that service. The section number of _this_ section is updated because that combination of tid / tid-ext / section-number has been updated.

Since EIT's are sparse sections and not full tables, there is no need to update the section numbers of other EIT sections. Their content is unchanged, there is no need to reload a full EPG.

> This invalidates all rest events in the EPG. 

The shouldn't and looks like a bug in the EPG management in the receiver.

przemo1983, 2023-02-22:
Thanks for the explanation. Will look into receiver implementation.

przemo1983, 2023-02-27:
> Since EIT's are sparse sections and not full tables, there is no need to update the section numbers of other EIT sections. Their content is unchanged, there is no need to reload a full EPG. 
Is it possible to add such a feature as an additional option to EITinject?

lelegard, 2023-02-27:
That would be a very bad idea. For about 25 years now, all implementers have made all possible efforts to avoid the avalanche effect of a fully updated EPG. Implementing a fully renewed EPG each time an event terminates is a crazy idea which would kill most receivers in real networks with many services. Long ago, the ETSI TS 101 211 spec has created the unfortunate "midnight effect" where all EIT sections need to be renewed at midnight. That created a lot of problems in many systems and receivers. What you are requesting is a "midnight effect" every 5 minutes. No way...

przemo1983, 2023-02-27:
See the example from the Norway network where many decoders are working without any problems with EITs.
[image]
I have analyzed the stream wit EIT data and each time when any information in EIT is changed all sections change their version number. Not only the one in which the change was made.
All EIT sections for particular service have always the same version as presented in the picture above.

If real networks works in this way why it is bad idea to add such a feature to your tool?

lelegard, 2023-03-03:
Hi @przemo1983 

I see what you mean. All along the development process of TSDuck, we have encountered correct but suboptimal or even completely incorrect behaviours in real networks. At some point, it "works" when the network and the receivers are compatible. This does not mean that this is optimal or correct. TSDuck is full of weird options to handle these peculiarities. We can add one more...

The whole idea about EIT's is that a sub-table is never "complete". There are always holes in it. EIT's are manipulated as individual sections, not as "tables" (or "sub-tables" since they are long sections). This is why an optimal implementation of EPG should track modification (ie. version update) at section level.

I have added option `--synchronous-versions` to `eitinject` to enforce the synchronous update of the version of all sections in an EIT schedule sub-table. However, be aware that this has an impact: each time an event is modified or removed, you have to regenerate 4 days of EPG (32 segments of 3 hours, all sections with same table id). This is why this is suboptimal. When an event is modified, it is technically possible to update only the section where the event is (or the segment - up to 8 sections - if the change shifts events in the segment). 

By default, using `eitinject`, an event is removed from the EPG (and the EIT schedule sections) when it completes. This means that each time an event completes (several times per hour on commercial channels), you have to regenerate 4 days of EPG.

To avoid this non-sense, I also added the option `--lazy-schedule-update`. With this option, a completed event remains in the EPG as long as its 3-hour segment is valid. This means that the 4 days of EPG are regenerated every 3 hours only. I consequently recommend to use `--lazy-schedule-update` whenever you use `--synchronous-versions`.

closed 2023-08-13 by github-actions[bot]