tsduck/tsduck #1179: TSECMG respond with channel_error

nikpakar, 2023-03-04:
Dear all, im testing a new multiplexer with tsecmg. When adding a ECM from mux, i see the channel_setup message received on tsecmg but the same respond with channel_error. This should be surely a problem from mux side as i have a different mux tested ok with tsecmg.

Can some one please assist me to figure out the reason tsecmg respond as channel_error so i can troubleshoot the same from mux side.

Thanks in advance for all the assistance.
-Nik

```
* 192.168.10.18:51337: 2023/03/04 14:17:22: session started
* received message from 192.168.10.18:51337
    channel_setup (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0001
    ECM_channel_id = 0x7740
    Super_CAS_id = 0x000FF001

* sending message to 192.168.10.18:51337
    channel_error (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0005
    ECM_channel_id = 0x7740
    error_status = 0x0000000000000013 (unknown)

* 192.168.10.18:51337: 2023/03/04 14:17:22: session completed
* 192.168.10.18:56322: 2023/03/04 14:18:27: session started
* received message from 192.168.10.18:56322
    channel_setup (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0001
    ECM_channel_id = 0x7748
    Super_CAS_id = 0x000FF001

* sending message to 192.168.10.18:56322
    channel_error (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0005
    ECM_channel_id = 0x7748
    error_status = 0x0000000000000013 (unknown)

* 192.168.10.18:56322: 2023/03/04 14:18:27: session completed
```

lelegard, 2023-03-04:
Hi @nikpakar 

The error status 0x0013 means "channel id in use". This means that the MUX tries to setup a channel with the same id as another channel which is already established. 

There must be some issue in the way tsecmg displays the error. It should not be "unknown". Moreover, the error status should not be so wide, this a 16-bit value, not a 64-bit one. I will have a look at it.

lelegard, 2023-03-04:
Concerning the format of your message, you must have an installation issue.

Normally, you should see this (just tested on opening two channels with same id):
~~~
* sending message to 127.0.0.1:51020
    channel_error (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0005
    ECM_channel_id = 0x0009
    error_status = 0x0013 (ECM_channel_id value already in use)
~~~

If you see `error_status = 0x0000000000000013 (unknown)`, this is because `tsecmg` cannot find the configuration files containing the names and format of ECMG<=>SCS protocol. Since this file also contain thousands of other "names", you must have other errors too.

How did you install TSDuck? Which version of TSDuck? On which operating system?

Otherwise, I confirm that your original problem is due to the MUX tries to setup a channel with an id which is already in use.

closed 2023-10-15 by github-actions[bot]