tsduck/tsduck #1181: [Q] Workflow/Pipeline Optimisation

Desmoss900, 2023-03-09:
Hi Thierry,

Haven't been active in a while but I am still a huge fan of tsduck.  I know you are a busy person and I would appreciate it greatly if you could have a look at my workflow/pipeline.

Couöld my tsp pipeline/workflow be improved as I am still having some "glitches" in my final mpegts stream...my workflow/pipeline is as follows (note that all my .ts files are CBR consisting of 1 x H.264 Video PES, 1 x AC3 Audio PES and 1 x AAC Audion PES and every file has been encoded with the exact same bitrates):

```

Source .ts File -> FFMpeg demux -> FFMpeg TS Mux stdout -> stdin (tsp command 1)

Note: the only reason I have FFMpeg in this pipeline is to that I can gather the current position (time and frame number) of the current video file.

----------------------------------------------------------------------------------------------
tsp command 1:
  tsp --initial-input-packets 7 
    --max-input-packets 7 --max-flushed-packets 7 --buffer-size-mb 0.25 --bitrate 9051033 
    -P regulate --packet-burst 7 --pcr-synchronous --pcr-pid 0x96 
    -P continuity --fix 
    -O ip --rtp --pcr-pid 0x96 --ttl 2 --enforce-burst --packet-burst 7 
    --local-address 127.0.0.1 --local-port 51000 localhost:44000
----------------------------------------------------------------------------------------------
                          ___
localhost:44000 RTP In-->| T |
                         | S |
localhost:44001 RTP In-->| W |-------->RTP Out -> 127.0.0.1:5555
                         | I |
localhost:44001 RTP In-->| T |
                         | C |
                         | H |
                         |___|

TSSwitch ports 1 & 2 are looped .ts files (Tech Difficulties & Not Broadcasting 10 Second Video Files encoded with the exact same parameters as the Source .ts files...albeit with blank audio tracks (tracks exist but have no sound).

----------------------------------------------------------------------------------------------
tsswitch command
  tsswitch -f -i -p 0 --buffer-packets 160 -max-input-packets 84 --max-output-packets 42
    --receive-timeout 2000 --remote 127.0.0.1:4665 --event-udp 127.0.0.1:4666
    -I ip --timestamp-priority rtp-kernel-tsp --local-address 127.0.0.1:44000
    -I ip --timestamp-priority rtp-kernel-tsp --local-address 127.0.0.1:44001
    -I ip --timestamp-priority rtp-kernel-tsp --local-address 127.0.0.1:44002
    -O ip --rtp --pcr-pid 0x96 --ttl 2 --enforce-burst --packet-burst 7 
      --local-address 127.0.0.1 --local-port 52000 127.0.0.1:5555
----------------------------------------------------------------------------------------------

The RTP out from the TSSwitch is then merged into a mpegts stream with the following tsp command (gave the initial null stream 10% more than the merge stream to allow for tables overhead).

  tsp -v -t --bitrate 9913036 --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7
    -I null
    -P regulate --packet-burst 7
    -P inject objects/pat.xml --poll-files --pid 0x0 --bitrate 1504
    -P inject objects/nit.xml --poll-files --pid 0x10 --bitrate 1504
    -P inject objects/sdt.xml --poll-files --pid 0x11 --bitrate 1504
    -P inject objects/tdt.xml=60000 --poll-files --pid 0x14 --bitrate 1504
    -P timeref --start system --system-synchronous --eit
    -P inject objects/pmt.xml --poll-files --pid 0x64 --bitrate 1504
    -P merge --pcr-reset-backwards --incremental-pcr-restamp --acceleration-threshold 0 
      --pass 0x96-0x98 --no-psi-merge --bitrate 9051033  "tsp -v -t --bitrate 9051033
        --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7
        -I ip 127.0.0.1:5555 -P filter --pcr --pid 0x96-0x98"
    -P eitinject -f objects/eit/eit_*.xml --europe --time system --pf --schedule --actual --other --prime-days 7
    -P pcrbitrate
    -P pcradjust --min-ms-interval 40
    -P continuity --fix
    -P fork --ignore-abort "tsp -v -t -O hls -a -l 5 -d 2 -i --live-extra-segments 5 
                           -p video/hls/{uuid}/index.m3u8 video/hls/{uuid}/live.ts"
    -O ip --rtp --pcr-pid 0x96 --ttl 5 --enforce-burst --packet-burst 7 
      --local-address 192.168.8.201 239.255.1.100:10001
----------------------------------------------------------------------------------------------

```

Any and all assistance would be greatly appreciated.

I will also be posting another question with regard to EPG generation and stratigies surrounding that topic.

Regards,

Mark (Desmoss900)

SonnyWalkman, 2023-10-04:
Hey @Mark (Desmoss900),

Did you ever get you implementation working? I'm doing a similar setup without tsswitch at the front end.  I have a h264 and aac video audio TS stream created by ffmpeg with the usual PMT, PAT and SDT. I've injected a NIT and both TDT and TOT into the 5Mbs stream.  using tsduck I have inconsistencies with video continuality errors after the injection process?  I assume the issue I see is with the input bitrate verses NIT, TDT and TOT injection --bitrates?  

Did you get EIT (EPG) injection working? I'd be great full if you could show how, you pulled it off?

Sonny

Desmoss900, 2023-10-06:
Hi Sonny,Sorry for the delay…your continuity errors are probably coming from not having enough „headroom“ in your ts stream.  I do have a setup working correctly (albeit still waiting on any feedback on how to manage eit tables efficiently)..,your ts stream needs to have enough stuffing bandwidth available to inject your EIT, TDT and TOT tables…I am referring to the ts bitrate, not the video/audio bitrate…so if your PMT, PAT, SDT tables + Video-Audio streams sum to 5Mbps, depending on the size of the EIT, you might try to create a 6Mbps ts stream, inject the 5Mbps stream from ffmpeg…this will give you 1Mbps of „stuffing“ which tsduck can then replace with the EIT and TDT/TOT info…when I get a chance I will post my complete pipeline so you can get an idea of how I have accomplished this…I have streams that have been up longer than 90 continuous days with still no errors 😝Regards,MarkDon‘t take life too seriously...no one gets out alive anyway....

SonnyWalkman, 2023-10-06:
Hello @Desmoss900 

Using your method to create a null bitstream the merge my ffmpeg stream then modify SDT, add NIT, TDT, TOT and inject EIT works now with zero discontinuity errors. Was a good lead and Thank-you for sharing. I agree, I'll create a smaller initial null stream like you have suggested say 6,000,000 bps and add what I need then regulate to the final modulation rate to satisfy the modulator.

I'd be appreciate looking at your use case and how your generating EIT? I have a simple 2 events working at startup however I'd like to inject 12 hours worth and continue forward generate 12 hours of EPG.

SonnyWalkman, 2023-10-06:
On another note,
Once I successfully get it working I should add it the TSduck WIKI as I feel it is sparse on real world examples.

Cheers

Sonny

Desmoss900, 2023-10-07:
Hi Sonny,

I do a „rolling“ EIT at the moment as I just stream an entire series…at stream startup I automatically create the entire schedule for all episodes and as one episode completes, delete it from the current position and add it to the end of the list (calculating the start time based on the last entry in the list + the length of the last episode)…my dilemma is that the EventID is a 14 bit field yielding a max value of 16383 and when I roll over to 1 when the max is value is reached, the EIT does not work correctly anymore 😬😬😬

Regards,

Mark