tsduck/tsduck #1182: [FEAT] AIT to signalize a broadband application

Litrona, 2023-03-13:
Hi,
Could you implement an easy AIT creator that signallizes an external broadband app ([https://](https://my_application_root_path/myHbbTV-app/)?
The way this example does: https://github.com/aventuri/opencaster/blob/master/tutorials/hbbtv/hbbtv-http.py
Adding also the AIT stream descriptor inside the existing PMT for that service.

Thanks in advance and best regards.

lelegard, 2023-03-13:
Hi @Litrona 

TSDuck is designed to avoid this. There is no need to create dedicated applications to generate specialized binary tables.

Just create your AIT (or any other table) in XML or JSON format. This a text file which can be created or edited manually or using a script if you need to repeat this many times.

Then, you can directly inject your XML or JSON table in a stream.

You can also edit the PMT the same way. The first method is to extract the PMT in XML or jSON format, edit it and reinject it. The second method is more general: use an XML patch file to edit the PMT on the fly, adding components or descriptors.

As usual, RTFM for more details.

Litrona, 2023-03-25:
ok, done.
Thanks

closed 2023-03-25 by Litrona