tsduck/tsduck #1185: [Q] CPU usage, no scrambling, what is the cause?

thehijacker, 2023-03-17:
Hello,

Using TSDuck to split some multicast MPTS streams to SPTS (5) and only pass multicast to new IP (4). Together 9 instances. What I noticed was high CPU usage after I added more and more instances.

This are the instances:

```
tsp -d --receive-timeout 1000 -I ip -l 1.1.1.1 239.10.12.50:5000 -P svrename -n "Channel 1" -p "TSDuck" -P regulate -O ip -l 172.17.145.212 -e -t 64 239.2.7.10:5000
tsp -d --receive-timeout 1000 -I ip -l 1.1.1.1 235.16.0.39:6000 -P svrename -n "Channel 2" -p "TSDuck" -P regulate -O ip -l 172.17.145.212 -e -t 64 239.2.7.12:5000
tsp -d --receive-timeout 1000 -I ip -l 1.1.1.1 239.19.23.4:5000 -P svrename -n "Channel 3" -p "TSDuck" -P regulate -O ip -l 172.17.145.212 -e -t 64 239.2.7.14:5000
tsp -d --receive-timeout 1000 -I ip -l 1.1.1.1 239.10.2.57:5000 -P svrename -n "Channel 4" -p "TSDuck" -P regulate -O ip -l 172.17.145.212 -e -t 64 239.2.7.5:5000
tsp -d --receive-timeout 1000 -I ip  -l 1.1.1.1 239.40.0.28:5500 -P zap 1 -P svrename -n "MPTS - 1" -p "TSDuck" -P regulate -O ip  -l 172.17.145.212 -e -t 64 239.5.100.1:5000
tsp -d --receive-timeout 1000 -I ip  -l 1.1.1.1 239.40.0.28:5500 -P zap 4 -P svrename -n "MPTS - 2" -p "TSDuck" -P regulate -O ip  -l 172.17.145.212 -e -t 64 239.5.100.2:5000
tsp -d --receive-timeout 1000 -I ip  -l 1.1.1.1 239.40.0.28:5500 -P zap 7 -P svrename -n "MPTS - 3" -p "TSDuck" -P regulate -O ip  -l 172.17.145.212 -e -t 64 239.5.100.3:5000
tsp -d --receive-timeout 1000 -I ip  -l 1.1.1.1 239.40.0.28:5500 -P zap 10 -P svrename -n "MPTS - 4" -p "TSDuck" -P regulate -O ip  -l 172.17.145.212 -e -t 64 239.5.100.4:5000
tsp -d --receive-timeout 1000 -I ip  -l 1.1.1.1 239.40.0.28:5500 -P zap 13 -P svrename -n "MPTS - 5" -p "TSDuck" -P regulate -O ip  -l 172.17.145.212 -e -t 64 239.5.100.5:5000
```

And this is the CPU usage:

[image]

I have read many topics regarding CPU usage when using scrambling. And that if course is normal. But here I am only passing specific PMT channel from MPTS to SPTS. I believe CPU usage should not be that high.

TSDuck binaries compiled from source code:

**tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.33-3139**

Thank you for suggesting how to lower the CPU. I wish to put even more channels to the server but I am already on 65 % of total CPU usage.

lelegard, 2023-03-17:
This kind of process typically uses less than 5% CPU, normally, on a regular desktop/laptop.

The first suggestion (although I don't see the relationship with insane CPU usage) is to remove all `-P regulate` plugins. Why do you use them? Please RTFM. The `regulate` plugin is designed to transform an unregulated input (a file for instance), into a regulated output (IP for instance). When the source is already regulated, using this plugin is at best useless and at worse counterproductive.

Otherwise, to confirm the individual CPU usage, I suggest to  add option `-m` (monitor) to `tsp`. It outputs at regular intervals (less frequent over time) the CPU and virtual memory usage.

Finally, using `top`, type `H` to display each thread on its own line and `V` to display a hierarchy of processes and threads. Go back and forth the pages of display and locate which plugins (threads) consume the most CPU.

thehijacker, 2023-03-17:
Hello. Thank you for your reply and great instructions where to look.

I have remove the regulate plugin. Stream analyzer still says output is OK so you are right, it was not needed, but I needed it at some point. Must have been when I tested local TS files as input and multicast IP output. I have killed all the processes now. Starting with 0.08% CPU usage I have then fired one TSP process at a time until five of them were running. Source was MPTS multicast that I split into five SPTS. When all 5 were running I ran **top** command:

```
top - 12:48:56 up 23 min,  6 users,  load average: 1.94, 2.34, 2.76
Threads: 276 total,   7 running, 269 sleeping,   0 stopped,   0 zombie
%Cpu(s): 19.3 us, 14.9 sy,  0.0 ni, 64.2 id,  0.0 wa,  0.0 hi,  1.5 si,  0.0 st
MiB Mem :  15867.6 total,  15121.7 free,    439.5 used,    306.5 buff/cache
MiB Swap:   4096.0 total,   4096.0 free,      0.0 used.  15142.0 avail Mem

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
   5195 t2admin   20   0  521304  38544  17940 R  32.5   0.2   0:18.59 ip
   5257 t2admin   20   0  521304  38780  18176 S  32.5   0.2   0:14.08 ip
   5281 t2admin   20   0  521304  38544  17940 R  32.1   0.2   0:12.29 ip
   5216 t2admin   20   0  521304  38520  17916 R  31.8   0.2   0:16.73 ip
   5159 t2admin   20   0  521304  38540  17936 S  31.1   0.2   0:20.03 ip
   5282 t2admin   20   0  521304  38544  17940 R  20.2   0.2   0:07.61 zap
   5160 t2admin   20   0  521304  38540  17936 S  19.5   0.2   0:11.69 zap
   5196 t2admin   20   0  521304  38544  17940 S  19.2   0.2   0:10.80 zap
   5217 t2admin   20   0  521304  38520  17916 R  19.2   0.2   0:10.05 zap
   5258 t2admin   20   0  521304  38780  18176 S  19.2   0.2   0:08.26 zap
   5259 t2admin   20   0  521304  38780  18176 R  18.2   0.2   0:07.68 svrename
   5218 t2admin   20   0  521304  38520  17916 S  17.9   0.2   0:09.26 svrename
   5197 t2admin   20   0  521304  38544  17940 S  17.5   0.2   0:09.80 svrename
   5283 t2admin   20   0  521304  38544  17940 S  17.5   0.2   0:06.62 svrename
   5161 t2admin   20   0  521304  38540  17936 S  17.2   0.2   0:10.62 svrename
   5260 t2admin   20   0  521304  38780  18176 S  15.2   0.2   0:06.47 ip
   5219 t2admin   20   0  521304  38520  17916 S  14.9   0.2   0:07.65 ip
   5162 t2admin   20   0  521304  38540  17936 S  14.2   0.2   0:08.78 ip
   5198 t2admin   20   0  521304  38544  17940 S  1

3.9   0.2   0:07.78 ip
   5284 t2admin   20   0  521304  38544  17940 S  13.6   0.2   0:05.21 ip
```

Calculating total CPU usage at around 35 %. I would says it is still too much. I was hoping to add 20+ channels on this server.

Here are outputs from all five threads:

[Relay1.txt](https://github.com/tsduck/tsduck/files/11000988/Relay1.txt)
[Relay2.txt](https://github.com/tsduck/tsduck/files/11000984/Relay2.txt)
[Relay3.txt](https://github.com/tsduck/tsduck/files/11000985/Relay3.txt)
[Relay4.txt](https://github.com/tsduck/tsduck/files/11000986/Relay4.txt)
[Relay5.txt](https://github.com/tsduck/tsduck/files/11000987/Relay5.txt)

And htop output:

[image]

Can you see what else might be the problem? **top** says IP plugin is using the most CPU. Thank you again.

lelegard, 2023-03-17:
Your logs show that each stream is around 112 Mb/s, which is quite a lot for a TS. Considering that IP multicast is the most demanding I/O plugin because of its lack of inherent buffering, what you see looks normal.

With IP multicast, the plugin is wake up every UDP datagram, meaning every 7 TS packets for a well-formed stream. This means that the input thread is wake up 10600 times per second if the TS is 112 Mb/s. Each chunk of TS packets is then processed all long the chain, 10600 times per second, times 20 for 20 processes. What do you expect?

TSDuck is a lab tool. It is designed to be flexible and generic, not to be highly specialized. For multiple highly demanding IP transmissions, you need specially tuned tools. At this rate, this is professional broadcast. Do you expect professional broadcast for free (ie. free tools)?

thehijacker, 2023-03-17:
Source is MPTS multicast IP and same for all five inputs. I planned to use TSDuck to extract each SPTS from it and pass forward to individual multicast output. How exactly TSDuck does this, you explained well and I understand better why CPU is that high.

I do use other application for such tasks (MPTS -> SPTS) like Cesbo Astra, but I like TSDuck as I can fire it from anywhere and it just works and of course that it is free. Thank you.

thehijacker, 2023-03-17:
I have tried the free version of Cesbo Astra. To my suprise it can also do MPTS to SPTS and output is perfect. Same channels I tested with TSDuck. And I added also some other high bitrate inputs (4K) to output only (SPTS->SPTS) and CPU usage is at around 8.3 %. For future readers of this topic. Cesbo Astra is available here:

https://cesbo.com/download/astra/4.4.182/free/

You need to make a script lua files like test.lua and pass it as parameter. For MPTS to SPTS you can have something like this:

`make_channel({ name = "Name",input = {"udp://bond0@239.40.0.28:5500#pnr=10"},output = {"udp://bond0.200@239.5.100.4:5000"})`

[image]

Not sure how Astra processes packets. For sure with less CPU usage. You are right. TSDuck is not meant for this. I have used to in past to analyze the streams. For that it works great and solved many issues with it.

I think we can close it now. Thank you and all the best.

closed 2023-03-17 by thehijacker