tsduck/tsduck #1187: application type in XML starts at 32768 instead of 0

DrakeSeven, 2023-03-22:
#### Bug description:
In the PMT XML, the `<application_signalling_descriptor><application ...>` has an attribute of `application_type`.

When you enter the value `0x00` (`<application application_type="0x00"`), StreamXpert displays `Application type: 32768`.

[image]

Here are some values and the corresponding readout of StreamXpert:
* `0x0000` => `32768`
* `0x0010` => `32784`
* `0x7fff` => `65535`
* `0x8000` => throws an Error in the Log

       Error: inject: '0x8000' must be in range 0 to 32,767 for attribute 'application_type' in <application>, line 6

#### How to reproduce:
1. PMT table

   Create an XML for the PMT: `/home/demo/pmt.xml`.
   Notice that the `application_type` is set to `"0x10"` (`16`).
   ~~~xml
   <?xml version="1.0" encoding="UTF-8"?>
   <tsduck>
     <PMT version="0" current="true" service_id="0x0001" PCR_PID="0x0100">
       <component elementary_PID="0x04D2" stream_type="0x05">
         <application_signalling_descriptor>
           <application application_type="0x10" AIT_version_number="0x00"/>
         </application_signalling_descriptor>
         <stream_identifier_descriptor component_tag="0x03"/>
       </component>
        <component elementary_PID="0x0100" stream_type="0x1B">
         <stream_identifier_descriptor component_tag="0x01"/>
       </component>
       <component elementary_PID="0x0101" stream_type="0x03">
         <stream_identifier_descriptor component_tag="0x02"/>
       </component>
    </PMT>
   </tsduck>
   ~~~
3. AIT table

   Create an XML for the AIT: `/home/demo/ait.xml`.

   ~~~xml
   <?xml version="1.0" encoding="UTF-8"?>
   <tsduck>
     <AIT version="0" current="true" test_application_flag="false" application_type="0x0010">
       <application control_code="0x01">
         <application_identifier organization_id="0x0000000D" application_id="0x00EB"/>
         <transport_protocol_descriptor transport_protocol_label="0x00">
           <http>
             <url base="http://itv.mit-xperts.com/hbbtvtest/"/>
           </http>
         </transport_protocol_descriptor>
         <application_descriptor service_bound="false" visibility="3" application_priority="5">
           <profile application_profile="0x0000" version="1.1.1"/>
           <transport_protocol label="0x00"/>
         </application_descriptor>
         <application_name_descriptor>
           <language code="deu" application_name="hbbtvtest"/>
         </application_name_descriptor>
         <simple_application_location_descriptor initial_path="index.php"/>
       </application>
     </AIT>
   </tsduck>
   ~~~
5. ffmpeg
   Create a DVB-TS Multicast
   ~~~bash
   pmt_start_pid=0x1000
   ffmpeg -f lavfi -i testsrc -f lavfi -i "sine=frequency=1000" -c:v libx264 -c:a mp2 -muxrate 3000k -mpegts_pmt_start_pid $pmt_start_pid -f mpegts "udp://239.6.66.201:1234?pkt_size=1316"
   ~~~
6. tsp
   Add AIT to DVB-TS Multicast

   ~~~bash
   pmt_table="/home/demo/pmt.xml"
   ait_table="/home/demo/ait.xml"
   pmt_start_pid=0x1000

   tsp --debug -P pmt --add-pid 0x04D2/0x05 -P inject --replace -p $pmt_start_pid -s $pmt_table -P inject --inter-packet 1000 -p 0x04D2 -s $ait_table -I ip 239.6.66.201:1234 -O ip 239.6.66.202:1234 --ttl 24
   ~~~
#### Expected behavior:
When you enter the value `0x10` (`<application application_type="0x10"`), StreamXpert should display `Application type: 16`.

#### Environment:
 - OS: Linux
 - OS version: Debian GNU/Linux 11 (bullseye)
 - TSDuck full version: 3.34-3176
 - Installation type: rebuilt binary

#### Additional information:
The same behaviour happens with version `3.33-3139` from the official debian repo.

lelegard, 2023-03-22:
Hi @DrakeSeven 

Congratulation, you found a bug ... in StreamXpert 😉 

Here is the definition of the Application signalling descriptor, from ETSI TS 102 809, section 5.3.5.1.

[image]

As you can see, the application type uses 15 bits. If StreamXpert displays a value above 32767, this is a bug because this cannot be a 15-bit value.

In practice, you can also see that the application type is preceded by one "reserved" bit. Per MPEG specification, all reserved bits shall be '1'. Consequently, the descriptor from your XML sample is generated by TSDuck as `6F 03 80 10 E0`. The MSB of `80` is the reserved bit and not part of the 15-bit application type.

It seems that StreamXpert reads all 16 bits and consider them as the application type. This is a bug of StreamXpert.

Conclusion: when two tools disagree on the interpretation of something, don't make any assumption on which is right. Dump the binary data, read the spec and then decide which is wrong.

DrakeSeven, 2023-03-22:
Thank you very much!I really appreciate the detailed explanation.Best regards

closed 2023-08-27 by github-actions[bot]