tsduck/tsduck #1189: [Q] installed in full on windows 10 but I can't run the program.

jimbo2018, 2023-03-25:
Hi.

Want to get NIT BAT and SDT for a received DTT mux and output changes.

But that may sound very simple it is anything but 

My windows 10 laptop I installed this on has multiple programs which if you double click you either get a command prompt with flashing cursor and sits there doing nothing or the command prompt window flashes for a split second and vanishes 

Without a .bat file thus program us useless as I cannot use it. Either it us a command line program or it isn't. If it us nit a command line program which program do I double click on to make this work please?

Thank you

jimbo2018, 2023-03-25:
And btw where can I go d older versions before 2018? Only goes back to v3.10 do where are v1 and V2 releases?  And which was the last to work with Windows XP?

lelegard, 2023-03-25:
> Either it us a command line program or it isn't. If it us nit a command line program which program do I double click on to make this work please?

TSDuck is a set of command line tools. There is no GUI. Double-clicking on anything is useless. It seems that you need to read the documentation first. This is a flexible, and admittedly complex, set of tools for TV engineers. This is absolutely not a point-and-click tool for beginners.

Since the commands can be complex, you must use a modern shell and shell scripts. Use either Windows PowerShell or Cygwin bash. Using the 40-years-old DOS command line window is hopeless.

I suggest to browse through the [user's guide](https://tsduck.io/download/docs/tsduck.pdf). Since it is pretty big, I recommend to read the table of contents, the intro, the description of the tsp command, browse through the examples chapter.

> where can I go d older versions before 2018? Only goes back to v3.10 do where are v1 and V2 releases? And which was the last to work with Windows XP?

You can't. Binary installers for older versions, before 3.10, were removed long ago. The storage space on GitHub is not infinite. The Git code repository goes backward to version 3.0. Version 1 and 2 were not open source. If you are interested, see [this presentation](https://tsduck.io/download/docs/tsduck-project.pdf) for an overview of TSDuck history.

jimbo2018, 2023-03-25:
Thank you for that response. My windows 10 laptop DOES have Powershell 5 so as far as I know up to date.
  
  
   
  
  
   It just LOOKS LIKE command prompt 
  
  
   
  
  
   I just need to know
  
  
   
  
  
   1 how to set it up to get the tables from my Linux set top box which has an internal IP on my home network;
  
  
   2 let TSDuck read this data;
  
  
   3 output the tables;
  
  
   4 then send only changes to a file or website or email 
  
  
   
  
  
   Thank you again

jimbo2018, 2023-03-25:
With Powershell the file extension should be .PS1 not .exe

If I go to tstables.exe 

All I get is a flashing cursor. Thus won't work unless it was tstables.ps1

That would give a usable interface to use. Instead I get the attached image

Thank you

jimbo2018, 2023-03-25:
Ok. Realised the cursor enables me to type onto it. So I now need to know how to get it to target an IP address on my home network to get the data  to appear 
  
  
   
  
  
   Thanks.

jimbo2018, 2023-03-25:
Even typing --help does nit do anything. So please can you give me an instruction to get the program to read my set top boxes via the local IP addresses 
  
  
   
  
  
   Thank you.

jimbo2018, 2023-03-25:
So I have the latest version installed on windows 10. But if I want to use tstables there should be a command to send to get the data. My satellite receiver us on internal address
  
  
   
  
  
   192.168.0.19
  
  
   
  
  
   So how do I get TSDuck to read that IP address and send the data to a file or email? I followed instructions but it just will not work.
  
  
   
  
  
   Thank you.