tsduck/tsduck #1190: [FEAT] patch-xml for EIT sections

Litrona, 2023-03-25:
I found very useful the --patch-xml option to patch the PMT table contents.
I saw it is also for BAT, CAT, NIT, PAT and SDT.
I would like to have it for EIT sections updates.

In fact my EIT patch would be something like this:
eit-patch.xml:
<tsduck>
  <EIT service_id="0x0001" transport_stream_id="0x0001" original_network_id="0x0001" x-update-service_id="0x0002" x-update-transport_stream_id="0x0002" x-update-original_network_id="0x0002"/>
</tsduck>

Just for updating the SID, ONID and TSID in the EIT descriptors.

Thanks in advance

lelegard, 2023-03-26:
Hi @Litrona 

This is currently not possible for one simple reason: XML / JSON files represent _tables_ while EIT's are sparse _sections_, never complete tables.

An XML file containing a NIT, for instance, contains everything from all sections in the NIT. It is possible to extract the XML file only when all sections of the NIT are collected. Similarly, when an XML file is injected, it is serialised first as one or more binary sections. Typically, in the serialisation operation, a new section is created when the previous one is full, with respect to the documented maximum section size of that table (typically 1024 or 4096 bytes).

In the general case, we can never extract EIT's as XML files because the EIT is never complete as a _table_. The command `tstables` and the plugin `tables` have the specific option `--pack-all-sections` which forces to consider each section as a complete table, overwriting all _section_number_ and _last_section_number_ with zero. This is useful to _extract_ EIT information. However, because the section numbers information is destroyed, it is not useful to reinject it.

lelegard, 2023-03-26:
Hi @Litrona 

Nonetheless, your suggestion is interesting. If it does not fit in the current way of working of XML files, let's change it.

The plugin which processes sections one by one is named ... `sections`.

I have added the option `--patch-xml` to this plugin. In this context, it works a bit differently. In practice, each section is first turned into a table containing just one section. For this, the section and last section numbers shall be first overwritten with zero. The XML patch is normally applied on the "table". Then, the first section of the resulting table is extracted and reinjected in replacement of the input section.

There are several details to take into account:
- The section number and last section number of the input section are saved and restored in the patched section.
- The patch shall not increase the size of the "table" so that it needs two sections. In all cases, only the first section of the patched table is kept.
- Maybe other unexpected side effects may occur.

With this, patching the service id, TS id and network id of an EIT section becomes possible. I have tried and kept it in the non-regression test suite as test 124.

Litrona, 2023-03-27:
Hi @lelegard 

Thanks for considering my suggestion. I guess, it will cover my needs.

However I am not sure, I understand how to apply it in my use case, just to change the SID, TSID and ONID, in all EIT sections whatever the table_id is.
Is that simple, as I wrote below, or do I have to consider any other options to be added ? (eit-patch.xml writtein in my 1st post above)

`tsp -I file input.ts -P sections --patch-xml eit-patch.xml -O file output.ts`

Thanks for your help

lelegard, 2023-03-27:
Look at [test 124](https://github.com/tsduck/tsduck-test/blob/master/tests/test-124.sh) as indicated in my previous post.
~~~
tsp ... -P sections --pid 18 --patch-xml test-124.patch.xml
~~~
and file `test-124.patch.xml` (excerpt):
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT service_id="0x0101" transport_stream_id="0x0001" original_network_id="0x20FA"
       x-update-service_id="0xABCD"
       x-update-transport_stream_id="0x1234"
       x-update-original_network_id="0x5678"/>
</tsduck>
~~~
Add a patch line for each service to rename in EIT's.

This test is now run on each push to the repo, on Linux, macOS and Windows. To make sure it still works.

Litrona, 2023-05-29:
To delete Other EIT contents on-the-fly, would be interesting to have a rule to delete nodes with a different attribute that the one set.

Example, to delete all EITs with TSID different than 0x0001.
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT transport_stream_id!="0x0001" x-node="delete"/>
</tsduck>`

lelegard, 2023-05-29:
The idea of having a "different from" condition in XML patch files is interesting. But in some other form because `transport_stream_id!="0x0001"` is not a valid XML syntax.

However, for your specific need of removing all EIT from a specific TS id, you may use a combination of `--section-content` and `--section-mask`. To remove EIT from TS ID 0x1234, use the following (test it first, I just typed it without testing):
~~~
-P sections --section-content 00000000000000001234 --section-mask 0000000000000000FFFF
~~~

Litrona, 2023-05-29:
ok, perfect, it works your suggestion.

`tsp -I file astra-11127V-2020-08-24.ts -P sections --pid 18 --keep --section-content 000000000000000
00414 --section-mask 0000000000000000FFFF -O file test.ts`

closed 2023-06-02 by Litrona

lelegard, 2023-06-02:
> Please try not to use RTFM as a regular resource in your responses, as I know that is not your intention, it may come across as offensive.

Understood, thanks for raising this.

In several places I worked in the past, RTFM had become a recurrent joke between colleagues.

It is interesting that it is now sometimes perceived as offensive. I see this as a symptom of a society where people consider they have all rights and zero duty.

Let's say that the offensive feeling sometimes comes in the other direction. This is clearly not your case here, you have studied the product in depth and explored many directions. However, in some cases, developers who have spent a tremendous effort to write code (and documentation!) face users who use the product for free, haven't spent a single minute trying to understand what it does or even open the documentation and naively ask the developer to read the doc for them. I perceive this as rather offensive. This is understandable for open source products without doc (there are too many). The idea behind writing a doc is investing in the explanation effort only once and then users become autonomous.

Of course, this addresses only a limited class of users, but it seems to happen more frequently here.

Litrona, 2023-06-02:
You really blessed my life.
God bless you too :-)