tsduck/tsduck #1191: [BUG] VUI not parsed correctly when HEVC Short-term reference picture set is present

bobben1no, 2023-03-27:
#### Bug description:
Tsduck will not parse HEVC VUI correct if there is transmitted Short-term reference picture set.
(not tested with earlier versions)

#### How to reproduce:
By downloading the PQ10 file from https://dvb.org/specifications/verification-validation/hevc-test-content/ 
and dump Sequence Parameter Set/VUI by using -P pes -a -v --avc-access-unit
tsp -I file /home/tsduck/DVB_7680x4320_HEVC_50fps_PQ10.ts -P pes -a -v --avc-access-unit -O drop | grep -i 'vui'

I think the HLG file on dvb.org may have the same issues.    

#### Expected behavior:

 DVBAnalyzer (dvbcontrol.com) dump of Sequence Parameter Set from the PQ10 file  DVB_7680x4320_HEVC_50fps_PQ10.ts:

```

--------------------------------------------------------------------------------------------------------------------------------------

Address         Parameter                                                              Length    Value                 Description
--------------------------------------------------------------------------------------------------------------------------------------
           
0x0000003A,0:          [-] Sequence Parameter Set (SPS)                                <0x95,0> :                       Main 10 profile; Level = 6.1; 4:2:0; 7680 x 4320 pixels
0x0000003A,0:               * zero_byte                                                <0x1,0> : 0x00 (0)            
0x0000003B,0:               * start_code_prefix_one_3bytes                             <0x3,0> : 0x000001            
0x0000003E,0:               * forbidden_zero_bit                                       <0x0,1> : '0'                 
0x0000003E,1:               * nal_unit_type                                            <0x0,6> : 0x21 (33)             Sequence Parameter Set (SPS)
0x0000003E,7:               * nuh_layer_id                                             <0x0,6> : 0x00 (0)            
0x0000003F,5:               * nuh_temporal_id_plus1                                    <0x0,3> : 0x01 (1)            
0x00000040,0:               * sps_video_parameter_set_id                               <0x0,4> : 0x00 (0)            
0x00000040,4:               * sps_max_sub_layers_minus1                                <0x0,3> : 0x03 (3)            
0x00000040,7:               * sps_temporal_id_nesting_flag                             <0x0,1> : '0'                   temporal up-switching is not possible
0x00000041,0:              [+] Profile Tier Level (1, 3)                               <0x10,0> :                       Main 10 profile; Level = 6.1
0x00000041,0:                   * general_profile_space                                <0x0,2> : 0                   
0x00000041,2:                   * general_tier_flag                                    <0x0,1> : '0'                 
0x00000041,3:                   * general_profile_idc                                  <0x0,5> : 2                     Main 10 profile
0x00000042,0:                  [-] general_profile_compatibility_flag[0..31]           <0x5,0> :                       '00100000000000000000000000000000'
0x00000042,0:                       * general_profile_compatibility_flag[ 0]           <0x0,1> : '0'                 
0x00000042,1:                       * general_profile_compatibility_flag[ 1]           <0x0,1> : '0'                 
0x00000042,2:                       * general_profile_compatibility_flag[ 2]           <0x0,1> : '1'                 
0x00000042,3:                       * general_profile_compatibility_flag[ 3]           <0x0,1> : '0'                 
0x00000042,4:                       * general_profile_compatibility_flag[ 4]           <0x0,1> : '0'                 
0x00000042,5:                       * general_profile_compatibility_flag[ 5]           <0x0,1> : '0'                 
0x00000042,6:                       * general_profile_compatibility_flag[ 6]           <0x0,1> : '0'                 
0x00000042,7:                       * general_profile_compatibility_flag[ 7]           <0x0,1> : '0'                 
0x00000043,0:                       * general_profile_compatibility_flag[ 8]           <0x0,1> : '0'                 
0x00000043,1:                       * general_profile_compatibility_flag[ 9]           <0x0,1> : '0'                 
0x00000043,2:                       * general_profile_compatibility_flag[10]           <0x0,1> : '0'                 
0x00000043,3:                       * general_profile_compatibility_flag[11]           <0x0,1> : '0'                 
0x00000043,4:                       * general_profile_compatibility_flag[12]           <0x0,1> : '0'                 
0x00000043,5:                       * general_profile_compatibility_flag[13]           <0x0,1> : '0'                 
0x00000043,6:                       * general_profile_compatibility_flag[14]           <0x0,1> : '0'                 
0x00000043,7:                       * general_profile_compatibility_flag[15]           <0x0,1> : '0'                 
0x00000044,0:                       * general_profile_compatibility_flag[16]           <0x0,1> : '0'                 
0x00000044,1:                       * general_profile_compatibility_flag[17]           <0x0,1> : '0'                 
0x00000044,2:                       * general_profile_compatibility_flag[18]           <0x0,1> : '0'                 
0x00000044,3:                       * general_profile_compatibility_flag[19]           <0x0,1> : '0'                 
0x00000044,4:                       * general_profile_compatibility_flag[20]           <0x0,1> : '0'                 
0x00000044,5:                       * general_profile_compatibility_flag[21]           <0x0,1> : '0'                 
0x00000044,6:                       * general_profile_compatibility_flag[22]           <0x0,1> : '0'                 
0x00000044,7:                       * general_profile_compatibility_flag[23]           <0x0,1> : '0'                 
0x00000045,0:                       * general_profile_compatibility_flag[24]           <0x1,1> : '0'                 
0x00000046,1:                       * general_profile_compatibility_flag[25]           <0x0,1> : '0'                 
0x00000046,2:                       * general_profile_compatibility_flag[26]           <0x0,1> : '0'                 
0x00000046,3:                       * general_profile_compatibility_flag[27]           <0x0,1> : '0'                 
0x00000046,4:                       * general_profile_compatibility_flag[28]           <0x0,1> : '0'                 
0x00000046,5:                       * general_profile_compatibility_flag[29]           <0x0,1> : '0'                 
0x00000046,6:                       * general_profile_compatibility_flag[30]           <0x0,1> : '0'                 
0x00000046,7:                       * general_profile_compatibility_flag[31]           <0x0,1> : '0'                 
0x00000047,0:                   * general_progressive_source_flag                      <0x0,1> : '1'                   source scan type of the pictures in the CVS should be interpreted as progressive only
[... 233 lines cut ...]
0x000000CA,2:                           * cpb_size_value_minus1                        <0x4,3> : 0x39386 (234374)      CPB size = 234375
0x000000CE,5:                           * cbr_flag                                     <0x0,1> : '0'                   the hypothetical stream scheduler (HSS) operates in an intermittent bit rate mode
0x000000CE,6:                   * bitstream_restriction_flag                           <0x0,1> : '0'                   bitstream restriction parameters are NOT present
0x000000CE,7:               * sps_extension_flag                                       <0x0,1> : '0'                   no sps_extension_data_flag syntax elements are present
0x000000CF,0:               * rbsp_stop_one_bit                                        <0x0,1> : '1'                 
0x000000CF,1:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,2:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,3:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,4:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,5:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,6:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'                 
0x000000CF,7:               * rbsp_alignment_zero_bit                                  <0x0,1> : '0'       
 
```        
#### Errors and logs:
No logging done.

#### Environment:
 - OS: Linux 
 - OS version: Ubuntu 22.04
 - TSDuck full version: 3.34-3176
 - Installation type: run from build location (source downloaded and build successfully)

#### Additional information:
None

lelegard, 2023-03-27:
Hi @bobben1no 

Thank you for the detailed technical report.

It seems that the problem is in the deserialization of the short-term reference picture set index 2. There is a mismatch here between what the `pes` plugin displays and what DVBAnalyzer displays. As a consequence, all the rest is desynchronized, including the trailing VUI.

I will look into it. It may take some time.

bobben1no, 2023-03-27:
Thank you !
There is no hurry.

I had a look in the H.265 spec and the tsduck source code yesterday and parsing the short-term reference picture set seems quite complicated compared to all other data in the sequence parameter set.

lelegard, 2023-03-27:
As you wrote, this part of the analysis is quite complex.

What does make you think that TSDuck is wrong here? Do you know for sure the structure of the data in the DVB reference stream? Or do you trust DVBAnalyzer (and assume that TSDuck is wrong when it differs)?

I ask that question because of the recent issue #1187 where a bug report was based on the comparison between TSDuck and a professional expensive tool. And the bug was on the other one... This one was quite easy to spot but yours is much more difficult to analyze. So, before spending a lot of time in complex binary data, I would like to know if the bug in TSDuck is a certainty or if it is based on the assumption that another tool is always correct.

bobben1no, 2023-03-27:
I  created the bug report after I evaluated the VUI data DVBAnalyzer outputs and found it to match 50fps HDR PQ very well. 
The VUI data reported by DVBAnalyzer indicates colour_primaries= 9 (Rec. ITU-R BT.2020-2/BT.2100-2) and transfer_characteristics 16 (Rec. ITU-R BT.2100-2 perceptual quantization (PQ) system).  DVBAnalyzer also reports vui_time_scale 27000000 and vui_num_units_in_tick 540000 which results in 27000000/540000=50fps.   The file "DVB_7680x4320_HEVC_50fps_PQ10.ts" is for verification of HEVC 50p HDR PQ so it did seem logical that DVBAnalyzer reports the VUI correctly. 

It is always nice to verify analyzer output on several tools and now I have also discovered that a quite cheap windows tool TransEdit (DVBViewer.com) reports the same as DVBAnalyzer.   Not sure what source TransEdit is based on but if it is ffmpeg then the ffmpeg function/source for parsing Short-term reference picture set can be viewed here : https://ffmpeg.org/doxygen/4.0/cbs__h265__syntax__template_8c_source.html

Not sure if you need it but here is the TransEdit XML Sequence parameter set export :
```
<?xml version="1.0" encoding="utf-8"?>
  <DVB_7680x4320_HEVC_50fps_PQ10_HEVC_8K_HDR_PQ10_V_V_HEVC_Video>
    <SequenceParameterSet Value="150" Unit="Byte">
      <ZeroByte HValue="0x00"/>
      <StartCodePrefixOne3Bytes HValue="0x000001"/>
      <ForbiddenZeroBit Value="0"/>
      <NALUnitType Value="33"/>
      <NUHLayerID Value="0"/>
      <NUHTemporalIDPlus1 Value="1"/>
      <SPSVideoParameterSetID Value="0"/>
      <SPSMaxSubLayersMinus1 Value="3"/>
      <SPSTemporalIDNestingFlag Value="0"/>
      <ProfileTierLevel>
        <GeneralProfileSpace Value="0"/>
        <GeneralTierFlag Value="0"/>
        <GeneralProfileIDC Value="2" Name="Main 10"/>
        <GeneralProfileCompatibilityFlag>
          <_0 Value="0"/>
          <_1 Value="0"/>
          <_2 Value="1"/>
          <_3 Value="0"/>
          <_4 Value="0"/>
          <_5 Value="0"/>
          <_6 Value="0"/>
          <_7 Value="0"/>
          <_8 Value="0"/>
          <_9 Value="0"/>
          <_10 Value="0"/>
          <_11 Value="0"/>
          <_12 Value="0"/>
          <_13 Value="0"/>
          <_14 Value="0"/>
          <_15 Value="0"/>
          <_16 Value="0"/>
          <_17 Value="0"/>
          <_18 Value="0"/>
          <_19 Value="0"/>
          <_20 Value="0"/>
          <_21 Value="0"/>
          <_22 Value="0"/>
          <_23 Value="0"/>
          <_24 Value="0"/>
          <_25 Value="0"/>
          <_26 Value="0"/>
          <_27 Value="0"/>
          <_28 Value="0"/>
          <_29 Value="0"/>
          <_30 Value="0"/>
          <_31 Value="0"/>
        </GeneralProfileCompatibilityFlag>
        <GeneralProgressiveSourceFlag Value="1"/>
        <GeneralInterlaced SourceFlag Value="0"/>
        <GeneralNonPackedConstraintFlag Value="0"/>
        <GeneralFrameOnlyConstraintFlag Value="0"/>
        <GeneralReservedZero44Bits String="0x000000"/>
        <GeneralLevelIDC Value="183" Name="Level 6.1"/>
        <SubLayer_0>
          <SubLayerProfilePresentFlag Value="0"/>
          <SubLayerLevelPresentFlag Value="0"/>
        </SubLayer_0>
        <SubLayer_1>
          <SubLayerProfilePresentFlag Value="0"/>
          <SubLayerLevelPresentFlag Value="0"/>
[... 366 lines cut ...]
              <CPBCntMinus1 Value="0"/>
              <SubLayerNAL_HRDParameters>
                <SubLayerHRDParameters_0>
                  <BitRateValueMinus1 Value="234374"/>
                  <CPBSizeValueMinus1 Value="234374"/>
                  <CBRFlag Value="0"/>
                </SubLayerHRDParameters_0>
              </SubLayerNAL_HRDParameters>
              <SubLayerVCL_HRDParameters>
                <SubLayerHRDParameters_0>
                  <BitRateValueMinus1 Value="234374"/>
                  <CPBSizeValueMinus1 Value="234374"/>
                  <CBRFlag Value="0"/>
                </SubLayerHRDParameters_0>
              </SubLayerVCL_HRDParameters>
            </HRDSubLayer_3>
          </VUI_HRDParametersPresentFlag>
        </VUITimingInfoPresentFlag>
        <BitstreamRestrictionFlag Value="0"/>
      </VUIParametersPresentFlag>
      <SPSExtensionFlag Value="0"/>
      <RBSPStopOneBit Value="1"/>
    </SequenceParameterSet>
  </DVB_7680x4320_HEVC_50fps_PQ10_HEVC_8K_HDR_PQ10_V_V_HEVC_Video>
```

lelegard, 2023-03-30:
Hi @bobben1no

I spent part of last night on this. This was tricky and silly at the same time. In the complex encoding of HEVC parameter sets, the vast majority of optional bits default to zero when not present. Except for two of them which default to one when not present. This is specified in the middle of a long and dense documentation. This changes the rest of the deserialization.

The ouput of the `pes` plugin now matches the structure from DVBAnalyzer. Test 125 has been added in the non-regression test suite to verify this.

Let us know if you find some other unexpected HEVC analysis.

closed 2023-05-16 by lelegard

bobben1no, 2023-11-27:
I have discovered that FTV UHD service on satellite Hotbird 13G is having some issues that may be related to HEVC Short-term reference picture set.  For this service no VUI can not be found with the current version of tsduck.

DVBAnalyzer seems to read a normal VUI but Transedit is not able to read VUI from this service.
It may be an encoding error or issue that may related to short term reference picture set.

I have a ts recording which is cut down to 1.4MByte, zipped and attached:
[FTV_UHD.zip](https://github.com/tsduck/tsduck/files/13472768/FTV_UHD.zip)

DVBAnalyzer shows this:

```
======================================================================================================================================
| PID: 0121 [PES -> Video]   length: 100895[0x18A1F]   status: VALID |
--------------------------------------------------------------------------------------------------------------------------------------

Address         Parameter                                                              Length    Value                 Description
--------------------------------------------------------------------------------------------------------------------------------------
0x00000000,0:  [+] PID: 0121 [PES -> Video]                                            <0x18A1B,0> :                       (374 nodes in 6 levels)
0x00000000,0:       * start_code_id                                                    <0x4,0> : 0x000001E0            Video
0x00000004,0:       * PES_packet_length                                                <0x2,0> : 0x0000 (0)          
0x00000006,0:       * 10                                                               <0x0,2> : '10'                
0x00000006,2:       * PES_scrambling_control                                           <0x0,2> : '00'                  No scrambling of PES packet payload
0x00000006,4:       * PES_priority                                                     <0x0,1> : '0'                   Normal priority
0x00000006,5:       * data_alignment_indicator                                         <0x0,1> : '1'                   PES packet header is immediately followed by the ES
0x00000006,6:       * copyright                                                        <0x0,1> : '0'                   Not defined if the associated PES packet payload is protected by copyright
0x00000006,7:       * original_or_copy                                                 <0x0,1> : '0'                   Contents of the associated PES packet payload is a copy
0x00000007,0:       * PTS_DTS_flags                                                    <0x0,2> : '11'                  Both PTS and DTS fields shall be present
0x00000007,2:       * ESCR_flag                                                        <0x0,1> : '0'                   No ESCR fields are present
0x00000007,3:       * ES_rate_flag                                                     <0x0,1> : '0'                   No ES_rate field is present
0x00000007,4:       * DSM_trick_mode_flag                                              <0x0,1> : '0'                   No trick mode field is present
0x00000007,5:       * additional_copy_info_flag                                        <0x0,1> : '0'                   No additional_copy_info field is present
0x00000007,6:       * PES_CRC_flag                                                     <0x0,1> : '0'                   No CRC field is present
0x00000007,7:       * PES_extension_flag                                               <0x0,1> : '0'                   No extension field is present
0x00000008,0:       * PES_header_data_length                                           <0x1,0> : 0x0A (10)           
0x00000009,0:      [-] PTS                                                             <0x5,0> :                       PTS: 7494344128 => Time: 83270.490311 sec => (hh:mm:ss.ms) 23:07:50.490
0x00000009,0:           * 0011                                                         <0x0,4> : '0011'              
0x00000009,4:           * PTS [32..30]                                                 <0x0,3> : 0x06 (6)            
0x00000009,7:           * marker_bit                                                   <0x0,1> : '1'                 
0x0000000A,0:           * PTS [29..15]                                                 <0x1,7> : 0x7D65 (32101)      
0x0000000B,7:           * marker_bit                                                   <0x0,1> : '1'                 
0x0000000C,0:           * PTS [14..0]                                                  <0x1,7> : 0x1DC0 (7616)       
0x0000000D,7:           * marker_bit                                                   <0x0,1> : '1'                 
0x0000000E,0:      [-] DTS                                                             <0x5,0> :                       DTS: 7494340528 => Time: 83270.450311 sec => (hh:mm:ss.ms) 23:07:50.450
0x0000000E,0:           * 0001                                                         <0x0,4> : '0001'              
0x0000000E,4:           * DTS [32..30]                                                 <0x0,3> : 0x06 (6)            
0x0000000E,7:           * marker_bit                                                   <0x0,1> : '1'                 
0x0000000F,0:           * DTS [29..15]                                                 <0x1,7> : 0x7D65 (32101)      
[... cut ...]
0x000000AF,0:               * first_slice_segment_in_pic_flag                          <0x0,1> : '1'                   the slice segment is the first slice segment of the picture in decoding order
0x000000AF,1:               * no_output_of_prior_pics_flag                             <0x0,1> : '0'                 
0x000000AF,2:               * slice_pic_parameter_set_id                               <0x0,1> : 0x00 (0)            
0x000000AF,3:               * slice_type                                               <0x0,3> : 0x02 (2)              I slice
0x000000AF,6:               * Payload last (HEVC)                                      <0x1896F,2> :

```

lelegard, 2023-11-27:
Hi @bobben1no 

Please do not add info to an issue which was closed six months ago. Nobody will notice this. If there is another case, please open a new issue.

Also please specify the TSDuck command you used, the TSDuck version, and what is wrong with this command. Here, we just have a zip file and "DVBAnalyzer shows this".