tsduck/tsduck #1194: [FEAT] HTTP output plugin

Marware, 2023-03-29:
#### Is your feature request related to a problem?
Many mpegts streams are transmitted over http, it will be useful for tsduck to have a the ability to serve output mpegts over http

#### Proposed solution:
Add HTTP output plugin

I can look into it if no objections to the feature

lelegard, 2023-03-29:
Hi @Marware 

> serve output mpegts over http

So, you mean that the output plugin would behave as an HTTP server, right ? 

The problem that I see here is what kind of HTTP server should we embed? Writing an HTTP server is incredibly complicated, not just responding to a `GET` line on a TCP socket. There are zillions of options and corner cases. You need to support SSL/TLS with lots of cryptographic suites. And this must remain portable on Linux, macOS, Windows, BSD, just like the rest of TSDuck. And this must be easily integrated to be compiled in scripts in the CI/CD pipeline, on all mentioned operating systems.

For HTTP _clients_ (input plugins `http`, `hls` and a few others), there are native libraries on all operating systems, namely WinInet on Windows and libcurl elsewhere. These libraries are already available on all OS and easy to integrate. That is how TSDuck and many other applications can be generic HTTP clients, facing all situations and protocols. 

How would you do it for an HTTP server?

Note that this is why the `hls` output plugin does not serve the HTTP protocol. It generates and renews the playlists and segments files but let an independent application serve them (typically an Apache server).

Marware, 2023-03-29:
I can imagine the challenges of adding an HTTP server since I use HTTP servers all the time and created a minimal one. However, there are some good projects building lightweight and embeddable HTTP server that cover all your mentioned concerns, also they can be stripped down to the use case of video only and embed it to tsduck library like SRT as a listener

lelegard, 2023-03-29:
OK, so here are the requirement for a library to be embedded:

- Reputed library (stable, reliable, secure). This kind of server library, if not designed with security in mind, may contain vulnerabilities.
- Available on Linux, macOS, Windows (optionally on BSD systems)
- Binary packaging available through installers (`apt install`, `dnf install`, `brew install`, etc) and a clear URL or Web page where the development binaries for Windows are available.

In the case of Windows, the binary libraries must be compiled with the native compiler MSC, not some weird mingw installation. If a binary installer for the libraries is not available on Windows, I can make one (I did it for SRT and RIST) but the compilation of the library must be automatable using a script (typically, it must be build using a provided MSBuild/VC solution file, or cmake or meson).

If you have a candidate library with all these characteristics, I will have a look.

Marware, 2023-03-29:
Thanks for the specs, on it!

lelegard, 2023-03-31:
And I forgot to mention the licensing requirements. It must be a permissive license, compatible with TSDuck BSD license;

- No GPL, restrictive license, rarely used in libraries but some of them use it (readline for instance)
- LGPL possible if binary libraries using shared objects / DLL's are available everywhere, with separate installation.
- Preferably BSD, MIT, Apache and the like.

closed 2024-01-21 by github-actions[bot]