tsduck/tsduck #1196: [Q] Printing the elementary stream header in hex

kmathewmk, 2023-03-31:
I would like to print the elementary stream header bytes for audio/video/private streams in hex/binary

For eg The following command produces only a human readable form
```
> tsp -I file output_1_5.ts -P pes -p 482 -a -O drop
* PID 0x01E2, stream_id 0xBD (Private stream 1), AC-3 audio attributes:
  AC-3, 3/2 (L,C,R,SL,SR), @48,000 Hz, complete main, bsid 6

```
The -b option in addtion to -a does not seem to help. 

Is there a way to get the bytes corresponding to the attributes printed above ?

lelegard, 2023-04-02:
Hi @kmathewmk 

The usual answer applies: RTFM.

> I would like to print the elementary stream header bytes 

For the PES header, use option ... `--header`.

~~~
$ tsp ... -P pes -p 0x82 -a --header  ...
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES header:
    0000:  00 00 01 BD 0C 08 87 80 05 25 E6 C7 FD AF
* PID 0x0082, stream_id 0xBD (Private stream 1), AC-3 audio attributes:
  E-AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 16
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES header:
    0000:  00 00 01 BD 0C 08 87 80 05 25 E6 CB 0B AF
~~~

And for the PES payload (where A/V data are), use option ... `--payload`.

It may be useful to also specify `--max-dump-size` because the PES payloads can be very large.

~~~
$ tsp ... -P pes -p 0x82 -a --payload --max-dump-size 48 ...
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES payload (truncated):
    0000:  0B 77 00 FF 34 85 FF A8 E3 90 00 CC 18 84 21 84  .w..4.........!.
    0010:  A6 28 5F 9E 69 21 3C 81 A4 94 11 07 07 06 06 07  .(_.i!<.........
    0020:  07 41 81 C1 C1 C1 81 FB 20 55 4C 8A 5A 4B 25 F5  .A...... UL.ZK%.
* PID 0x0082, stream_id 0xBD (Private stream 1), AC-3 audio attributes:
  E-AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 16
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES payload (truncated):
    0000:  0B 77 00 FF 34 85 FF 88 E3 90 20 4C 09 00 10 4D  .w..4..... L...M
    0010:  15 8A 17 DF 9A 42 4A 3D 48 C5 04 41 C1 C1 C1 C1  .....BJ=H..A....
    0020:  C1 D0 70 70 70 70 70 7E A9 1E D4 72 BA B3 F4 B1  ..ppppp~...r....
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES payload (truncated):
    0000:  0B 77 00 FF 34 85 FF 88 E3 90 20 CC 00 00 00 2E  .w..4..... .....
    0010:  A2 C5 0B F1 CD 22 A4 1C 24 93 82 20 E0 81 21 01  ....."..$.. ..!.
~~~

> The -b option in addtion to -a does not seem to help.

Again, RTFM will tell you _"Include binary dump in addition to hexadecimal."_ So, if you don't ask for hexadecimal dump (through `--header` or  `--payload`), `-b` won't give you anything more.

~~~
$ tsp ... -P pes -p 0x82 -a --payload --max-dump-size 48 -b ...
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES payload (truncated):
    0000:  0B 77 00 FF 34 85 FF A8  00001011 01110111 00000000 11111111 00110100 10000101 11111111 10101000  .w..4...
    0008:  E3 90 00 CC 18 84 21 84  11100011 10010000 00000000 11001100 00011000 10000100 00100001 10000100  ......!.
    0010:  A6 28 5F 9E 69 21 3C 81  10100110 00101000 01011111 10011110 01101001 00100001 00111100 10000001  .(_.i!<.
    0018:  A4 94 11 07 07 06 06 07  10100100 10010100 00010001 00000111 00000111 00000110 00000110 00000111  ........
    0020:  07 41 81 C1 C1 C1 81 FB  00000111 01000001 10000001 11000001 11000001 11000001 10000001 11111011  .A......
    0028:  20 55 4C 8A 5A 4B 25 F5  00100000 01010101 01001100 10001010 01011010 01001011 00100101 11110101   UL.ZK%.
* PID 0x0082, stream_id 0xBD (Private stream 1), AC-3 audio attributes:
  E-AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 16
* PID 0x0082, stream_id 0xBD (Private stream 1), size: 3086 bytes (header: 14, payload: 3072)
  PES payload (truncated):
    0000:  0B 77 00 FF 34 85 FF 88  00001011 01110111 00000000 11111111 00110100 10000101 11111111 10001000  .w..4...
    0008:  E3 90 20 4C 09 00 10 4D  11100011 10010000 00100000 01001100 00001001 00000000 00010000 01001101  .. L...M
    0010:  15 8A 17 DF 9A 42 4A 3D  00010101 10001010 00010111 11011111 10011010 01000010 01001010 00111101  .....BJ=
    0018:  48 C5 04 41 C1 C1 C1 C1  01001000 11000101 00000100 01000001 11000001 11000001 11000001 11000001  H..A....
    0020:  C1 D0 70 70 70 70 70 7E  11000001 11010000 01110000 01110000 01110000 01110000 01110000 01111110  ..ppppp~
    0028:  A9 1E D4 72 BA B3 F4 B1  10101001 00011110 11010100 01110010 10111010 10110011 11110100 10110001  ...r....
~~~

I recommend to RTFM carefully, this is where the answers are.

kmathewmk, 2023-04-03:
Thanks for you response. 

My quest was to know if there was an option that would specifically dump the ES header rather that the PES header. In my case that would be the bytes pertaining to 
```
* PID 0x01E2, stream_id 0xBD (Private stream 1), AC-3 audio attributes:
  AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 6
```

I kind of managed to get that by using the following command:
```
tsp -I file output_1_5.ts -P pes -p 482 --save-es - -O drop | hexdump | more
00000000: 0B 77 14 B6 1C 30 43 1F - F7 37 24 92 00 00 FF D7 | w   0C  7$     |
00000010: D7 0C 24 AD D9 C9 83 C5 - 58 1E 92 AA 5B EF 3E 86 |  $     X   [ > |
```
The bytes of interest are 
1C 30 43 .... 
corresponding to Tables 5.1 (syncinfo) and 5.2 (bsi) from [ATSC A52](https://prdatsc.wpenginepowered.com/wp-content/uploads/2021/04/A52-2018.pdf), excluding syncword (16 bits), crc1 (16 bits) of syncinfo and spanning variable number of bytes for bsi

closed 2023-04-03 by kmathewmk