tsduck/tsduck #1197: [FEAT] Report error/warning on trimmed PES packets

alex3d, 2023-03-31:
#### Is your feature request related to a problem?

When actual PES packet length (a data size between two PUSI marks) is less than PES packet length in PES header it is not considered a problem and silently ignored:
https://github.com/tsduck/tsduck/blob/f6a0aa02e664a7a0cb01451dbc70fcb6345860bf/src/libtsduck/dtv/pes/tsPESPacket.cpp#L234

#### Proposed solution:

Report a problem in a same way like "spurious trailing bytes" reported in case when actual PES size greater than PES packet length in PES header.

#### Alternatives:

#### Additional information:

If this behavior is not intentional I could probably make a PR.

alex3d, 2023-04-11:
@lelegard ping?

lelegard, 2023-04-17:
Hi @alex3d 

This has been added as part of `-P pes --trace-packets`.
Please confirm that it works for you.

AlfonZ42, 2023-05-01:
Is it possible to add a message to _analyze_'s _error analysis report_ as well, or is this out of scope of what _analyze_ should report there?
Thanks.

lelegard, 2023-05-03:
> Is it possible to add a message to analyze's error analysis report as well

Done, but I have no TS to test right now. Could you please test and report here if it works as expected?

AlfonZ42, 2023-05-04:
I believe inputs for tests 002, 059, 072, 089, 092 contain invalid PES packets, but _analyzer_'s _error analysis_ doesn't seem to be invoked in any tests.
On my end I'll try tomorrow after the nightly build is done.

lelegard, 2023-05-04:
> I believe inputs for tests 002, 059, 072, 089, 092 contain invalid PES packets, 

Indeed:
~~~
$ tsanalyze --error-analysis test-059.ts
TITLE: ERROR ANALYSIS REPORT
INFO: Transport Stream Identifier: 4 (0x0004)
TS:4:0x0004: No BAT
TS:4:0x0004: No TDT
PID:0:0x0000: Discontinuities (unexpected): 3
PID:16:0x0010: Discontinuities (unexpected): 1
PID:17:0x0011: Duplicated TS packets: 2
PID:18:0x0012: Discontinuities (unexpected): 1
PID:21:0x0015: Discontinuities (unexpected): 1
PID:300:0x012C: Discontinuities (unexpected): 5
PID:320:0x0140: Discontinuities (unexpected): 22
PID:330:0x014A: Discontinuities (unexpected): 14
PID:330:0x014A: Invalid PES packets: 1     <===================
PID:331:0x014B: Discontinuities (unexpected): 16
PID:331:0x014B: Invalid PES packets: 2     <===================
PID:332:0x014C: Discontinuities (unexpected): 14
PID:332:0x014C: Invalid PES packets: 1     <===================
PID:333:0x014D: Discontinuities (unexpected): 17
PID:333:0x014D: Invalid PES packets: 2     <===================
PID:340:0x0154: Discontinuities (unexpected): 1
PID:370:0x0172: Discontinuities (unexpected): 1
PID:371:0x0173: Discontinuities (unexpected): 1
PID:400:0x0190: Discontinuities (unexpected): 1
PID:420:0x01A4: Discontinuities (unexpected): 8
PID:420:0x01A4: Duplicated TS packets: 3
PID:430:0x01AE: Discontinuities (unexpected): 9
PID:430:0x01AE: Duplicated TS packets: 1
PID:430:0x01AE: Invalid PES packets: 1     <===================
PID:431:0x01AF: Discontinuities (unexpected): 8
PID:431:0x01AF: Duplicated TS packets: 1
PID:431:0x01AF: Invalid PES packets: 1     <===================
PID:432:0x01B0: Discontinuities (unexpected): 8
PID:432:0x01B0: Duplicated TS packets: 1
PID:432:0x01B0: Invalid PES packets: 1     <===================
PID:440:0x01B8: Discontinuities (unexpected): 3
PID:441:0x01B9: Discontinuities (unexpected): 1
SUMMARY: Error count: 34
~~~

closed 2023-05-16 by lelegard