tsduck/tsduck #1198: Wants to analyze stream is found or not .

prashant-09-tech, 2023-04-03:
Wants to check stream is found or not.
Currently  I am using this command -> tsp -I ip 10202 -P until -s 8 -P analyze   .
but It is not working  when stream is not found .

 @lelegard So please  suggest which plugin is useful to detect stream is coming or not.

lelegard, 2023-04-03:
Hi @prashant-09-tech 

Use `-I ip --receive-timeout 8000 ...` with 8000 being a number of milliseconds.

Using any other plugin in the chain won't work because the chain of plugins is started after some input packets were initially received (to have an idea of the TS bitrate if possible). In the case where there is no input, the plugin chain will never be started. This is why the absence of input must be detected in the input plugin. Several input plugins have a `--receive-timeout` option: `ip`, `http`, `hls`, `dvb`, `dektec`, etc.

prashant-09-tech, 2023-04-03:
@lelegard  It is working .
Thank you

closed 2023-04-03 by prashant-09-tech