tsduck/tsduck #1201: [BUG] make file with NOPCSC NODTAPI NOVATEC still adding pcsc dektec and vatec headers in tsduck.h

Hakai-Shin, 2023-04-08:
#### Bug description:

make command with NOPCSC=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 build successful but using the development library in a sample app throws error.

running tsconfig --cflag give the following output:
-I/home/tsduck/include/tsduck -I/usr/include/PCSC --std=c++11

#### How to reproduce:

1) sudo make -j10 NOPCSC=1 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOEDITLINE=1 NOSRT=1 NODTAPI=1 NOGITHUB=1

2) sudo make NOPCSC=1 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOEDITLINE=1 NOSRT=1 NODTAPI=1 NOGITHUB=1 install install-devel

3) run make on sample code provided at 
https://github.com/tsduck/tsduck/tree/master/sample/sample-app

#### Expected behavior:

build sample app

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

[image]

#### Environment:
 - OS: CENTOS
 - OS version: CentOS Linux release 7.8.2003 (Core)
 - TSDuck full version: tsduck-3.32
 - Installation type: build from source

#### Additional information:

maceyldn, 2023-04-08:
I'm seeing the same thing building in Docker.

RUN git clone https://github.com/tsduck/tsduck.git
WORKDIR tsduck
RUN make distclean
RUN bash -x scripts/install-prerequisites.sh
RUN make -j10 NOPCSC=1 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 install

then build my app.

#25 3.892 In file included from /usr/include/tsduck/tsPCSC.h:65,
#25 3.892                  from /usr/include/tsduck/tsduck.h:499,
#25 3.892                  from src/xxxx.h:17,
#25 3.892                  from src/xxxx.cpp:1:
#25 3.892 /usr/include/PCSC/winscard.h:41:10: fatal error: pcsclite.h: No such file or directory
#25 3.892    41 | #include <pcsclite.h>
#25 3.892       |          ^~~~~~~~~~~~
#25 3.892 compilation terminated.

lelegard, 2023-04-09:
Hi @Hakai-Shin and @maceyldn 

The `NOxxx` symbol in the `make` command line were primarily added to build binaries, not to install a development environment. There were added upon users' requests to remove dependencies on some libraries they did not want to install on "production" systems (ie. not developments systems). 

I have just made some modifications to fix this and make sure that a development environment which was built and installed with `NOxxx` symbols would work now.

However, please consider that TSDuck is [tested](https://tsduck.io/doxy/testing.html) in its full-featured version only. The number of combinations of `NOxxx` symbols is too large to be tested. It would take days in the CI/CD pipeline and GitHub would probably complain... These symbols were added to help a few users but I cannot guarantee that every possible combination of symbols (and operating systems) is correct. This is why you ran into that problem.

Please let us know if the recent modifications fix your problems.

Additionally, @maceyldn, there is some inconsistency in the way you build your container. You build with several `NOxxx` symbols but you run `scripts/install-prerequisites.sh`. This script is a convenience to install _everything_ that is required by a full-featured build. So, you build to avoid some dependencies but you install those dependencies anyway. Specifically, the PCSC tools and services will be installed and started. Maybe not a big deal, just a bit inconsistent. However, I wonder why your build reports an error on a PCSC header since `scripts/install-prerequisites.sh` had installed it. Do you build in the same container?

maceyldn, 2023-04-09:
Thanks @lelegard, yes my fault.

turns out install-prereqs was never actually being run for some reason.

RUN git clone https://github.com/tsduck/tsduck.git && cd tsduck && \
make -j10 install NOPCSC=1 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 SYSPREFIX=/usr/local

Seems to now work, however...
the SRT input/output modules never get built.

I'm installing the dependency from source.
RUN git clone https://github.com/Haivision/srt.git
WORKDIR srt
RUN ./configure && make -j8 && make install

before running the tsduck make. Any thoughts?

Thanks

lelegard, 2023-04-09:
TSDuck is built with SRT if SRT is detected. Looking into the makefile, you can see that SRT is considered present if header files are found in directory `/usr/include/srt/`. This is where the SRT binary packages install their headers.

Why do you prefer to install SRT from sources instead of using a binary package? SRT is now available on most Linux distros? Which distro do you use in your container? If this is Alpine Linux, as in most containers, use package `libsrt-dev`.

Otherwise, if you prefer to keep using the sources, just check that SRT is installed in `/usr` and not some other exotic location.

Hakai-Shin, 2023-04-10:
it is working, thank you @lelegard

closed 2023-04-10 by Hakai-Shin