tsduck/tsduck #1203: [Q] Build tsduck for arm64 debian or ubuntu?

tthil, 2023-04-12:
How can I build tsduck for arm64 debian or ubuntu?

Do I need any specific parameters?

lelegard, 2023-04-12:
Hi @tthil 

I just run `make -j10` on my Ubuntu 22.10 virtual machine on a MacBook M1 host (arm64 CPU) and it works fine. 

`make installer` should build the `.deb` package. 

For the fun, I even tuned the crypto part of the code using the specialized arm64 instructions for AES and SHA-1 and -2. I also used the CRC special arm64 instructions. Computing a CRC32 is 20 times faster with the CRC instructions than the portable C++ code. Considering that computing a CRC32 is quite common with MPEG sections, this is a nice gain. 

Just to mention that TSDuck is active on Arm64, Linux and macOS. 

Not tested yet on arm64 Windows. I have an arm64 Windows 11 VM but the environment is poor. Many usual products are missing. 

What are your versions of Ubuntu, Debian and GCC ? Which CPU? Which command to build TSDuck?

If you use Ubuntu 22.10, you may try the prebuilt arm64 package for TSDuck 3.33.

tthil, 2023-04-13:
gcc (Debian 10.2.1-6) 10.2.1 20210110

I am trying to build this on Orange Pi 3 LTS with CPU:
Allwinner H6 quad-core 64-bit 1.8GHz high-performance Cortex-A53 processor

Everything goes fine until:
  [LD] /root/tsduck/bin/release-aarch64-orangepi3-lts-static/tsmux
  [GEN] tsduck.tables.model.xml
make[3]: *** [Makefile:67: /root/tsduck/bin/release-aarch64-orangepi3-lts-static/tsduck.tables.model.xml] Illegal instruction
make[2]: *** [Makefile:38: default] Error 2
make[1]: *** [Makefile:61: default] Error 2
make: *** [Makefile.inc:576: static] Error 2

So I guess it's a problem to generate tsduck.tables.model.xml!?

closed 2023-04-13 by tthil

lelegard, 2023-05-04:
This is a fixed. See a [complete explanation](https://github.com/tsduck/tsduck/issues/1205#issuecomment-1534495595) in #1205.