tsduck/tsduck #1205: [Q] tsduck.tables.model.xml - Illegal instruction

jof-cw, 2023-04-18:
Hello Lelegard

I'm trying to install TSDuck on a Okdo Rock 4C+ (similar to Raspberry 4). I'm using Debian 11 64bit.

The scenario is:
uname -a
Linux rock-4c-plus 5.10.110-1-rockchip #27b5bf1db SMP Mon Apr 3 11:17:08 UTC 2023 aarch64 GNU/Linux

Linux is Debian 11 64bits.

First of all I executed the script _install-prerequisites.sh_ without problems; Next I execute only "make" command and after some time it show this error message:

....
 [CXX] utestXML.cpp
  [LD] /home/a2t/Descargas/tsduck/bin/release-aarch64-rock-4c-plus/utest
  [LD] /home/a2t/Descargas/tsduck/bin/release-aarch64-rock-4c-plus/utest_static
  [DEP] tsprofiling.cpp
  [DEP] tsmux.cpp
  [DEP] setpath.cpp
  [CXX] tsmux.cpp
  [LD] /home/a2t/Descargas/tsduck/bin/release-aarch64-rock-4c-plus/tsmux
  [CXX] tsprofiling.cpp
  [LD] /home/a2t/Descargas/tsduck/bin/release-aarch64-rock-4c-plus/tsprofiling
  **[GEN] tsduck.tables.model.xml
make[2]: *** [Makefile:67: /home/a2t/Descargas/tsduck/bin/release-aarch64-rock-4c-plus/tsduck.tables.model.xml] Illegal instruction
make[1]: *** [Makefile:38: default] Error 2
make: *** [Makefile:61: default] Error 2**

What's wrong? I searched in the issues but I found nothing; thank in advance.

jof-cw, 2023-04-18:
Sorry, I just founded a similar case #1203

lelegard, 2023-04-18:
Hi @jof-cw 
Also involving @tthil as he reported a similar issue first.

The file `tsduck.tables.model.xml` is generated using the TSDuck tool `tsxml`. So, the build starts with the libraries, then the applications and, finally, the model file using the previously generated application. If there is a problem with the generated code of the application, building the final configuration file will fail.

If two unrelated people have the same "illegal instruction" issue, this is not an accident and there must be something to investigate.

On my side, I did tests on a Mac M1 using macOS and Linux Ubuntu. The M1 is an Arm64 chip but more advanced in terms of Arm features. The M1 uses Apple Firestorm cores at Arm8.5-A level while the Rockchip RK3399-T uses a mix of Cortex A72 and A53 at Armv8.0-A level. I wonder if this difference has an impact on the binary.

Specifically, I customized the CRC32 and crypto code to use the accelerated Arm64 instructions, when available. The compiled code contains the two versions, with and without acceleration. The presence of the special instructions is checked at runtime and the corresponding code path is taken only when the instructions are present. So, in theory, it should work in all cases. However, I wonder if there could be some hidden issue.

More details on Arm64 features here: https://github.com/lelegard/arm-cpusysregs/blob/main/docs/features.md

jof-cw, 2023-04-19:
Hello Lelegard; thank you for your response.

I done the same as @tthil with the source zip file and the compilation was done without problems.

I tried to compile the tool that you told me (arm-cpusysregs) but the compilator showed a lot of erros related with some variables; if you need a feedback I would open a new issue on that place.

Best regards; Jesus

lelegard, 2023-04-19:
> I done the same as @tthil with the source zip file and the compilation was done without problems.

What did you do exactly the first time? Which zip file did you download? Is it a zip of the current state of the repo or the zip of the latest release?

Cloning the repo or getting an archive of the current state of the repo shall produce _exactly_ the same result. If one operation consistently pulls garbage, there is a problem that we need to investigate.

> I tried to compile the tool that you told me (arm-cpusysregs) but the compilator showed a lot of erros related with some variables; if you need a feedback I would open a new issue on that place.

This was just a pointer to additional information on the various Arm features. If you are only interested in TSDuck, this project is useless to you. It is only useful to study some technical aspects of the Arm64 internals. That being said, you should not get compilation errors on an Arm64 Linux system, unless you use an old compiler.

jof-cw, 2023-04-19:
Hello Lelegard

> What did you do exactly the first time? Which zip file did you download? Is it a zip of the current state of the repo or the zip of the latest release?

The first time I cloned the repository with _git clone_ <...> 

> This was just a pointer to additional information on the various Arm features. If you are only interested in TSDuck, this project is useless to you. It is only useful to study some technical aspects [...]

I am interested in TSDuck only; but for your interest, this Debian distro use gcc version 10.2.1 20210110 (Debian 10.2.1-6), maybe this is old.

Thank you so much; Jesus

lelegard, 2023-04-19:
> The first time I cloned the repository with git clone <...>

This is worrying. Especially because two independent users got the same problem around the same time frame. If you still have the cloned repo, could you make a recursive diff with the unzipped archive? They should be identical. But they are probably not since one works and the other doesn't.

lelegard, 2023-04-19:
I just cloned the repo and downloaded the master zip. The contents are identical, just as expected:
~~~
$ git clone --depth 1 https://github.com/tsduck/tsduck tsduck-git-clone
$ curl https://codeload.github.com/tsduck/tsduck/zip/refs/heads/master -o tsduck-master.zip
$ unzip tsduck-master.zip
$ diff -Naur -x .git tsduck-git-clone tsduck-master
$
~~~
What did you do differently?

lelegard, 2023-05-04:
> This is worrying. Especially because two independent users got the same problem around the same time frame.

I was right to worry, there is an actual issue (that I just fixed). This is due to the lack of flexibility of GCC in terms of code generation for Arm64. This is a bit tricky and deserves some explanations.

A few months ago, I drastically improved the performances of CRC32 and crypto on Arm64 by using the corresponding specialized instructions (CRC, AES, SHA-1/256/512). These instructions are optional in the Arm architecture. So, the code anticipates this: the level of features in the current Arm CPU is checked. If the corresponding instructions are present, we use the accelerated version of the algorithm. Otherwise, we use the standard portable implementation. This is dynamically decided in one single binary.

So far, so good, we have a universal binary which can run on all Arm64 CPU's and gets the best of the CPU on which it executes. The problem is the GCC compiler. If we want to compile the `asm()` code with the specialized instructions (or the equivalent intrinsics), we _must_ specify a minimum level of architecture, for instance `-march=armv8.2-a+crypto+sha2+sha3`. Otherwise, GCC refuses to compile the specialized parts of the code.

However, this has undesired side effects. Using `-march=armv8.2-a+...`, GCC feels free to generates post-Armv8.0 instructions here and there. When you use an Armv8.0 CPU such as the Arm Cortex A72, the CPU core in the BCM2711 SoC as used in the Raspberry Pi 4, it fails with "illegal instruction".

I only had Armv8.4 and Armv8.5 CPU's to test and the problem is invisible at these levels of architecture. Trying on a Raspberry Pi 4, I was able to determine that the code crashes on an atomic add instruction (ldaddab), which is part of Arm FEAT_LSE, which an Armv8.1 feature. This explains the problem.

I would expect the compiler to accept optional features (`crc`, `crypto`, `sha2`, etc) without being obliged to raise the level of architecture. Ideally, some pragma should allow to change the level of architecture and features in specific portions of source code which are called in a specific context. Unfortunately, GCC is rather limited in that matter. The Arm code generation is still too influenced by the embedded world where you recompile everything for each target. Too rigid, not flexible enough.

I consequently modified the structure of the source code wherever specialized instructions are used. The accelerated parts of the code - which are called only when the corresponding feature is implemented in the current CPU - were moved in separate modules. Each of these specialized modules is built with a customized `-march` option (see `src/libtsduck/Makefile`). The rest of the code is compiled with the default Arm64 level of architecture.

Now, the same binary, same installer, works at all level of architectures. To view the level of features of the current platform, use the following command.

On a Raspberry Pi 4:
~~~
$ tsversion --version=acceleration
CRC32: yes, AES: no, SHA-1: no, SHA-256: no, SHA-512: no
~~~

On a MacBook M1:
~~~
$ tsversion --version=acceleration
CRC32: yes, AES: yes, SHA-1: yes, SHA-256: yes, SHA-512: yes
~~~

So, you may now clone the repo and rebuild.

BTW, cloning the repo or downloading a zip does not make any difference. The source code and the binaries are identical and exhibit the same issue. I suspect that you both downloaded the zip from the previous release, 3.33, instead of a zip of the current state of the repo. Version 3.33 did not implement the specialized instruction optimization and did not have the problem.

jof-cw, 2023-05-04:
Hello Lelegard

Sorry for my long silence; I was working out of office.

I just done your commands and now it works fine.

Thank you so much.

jof-cw, 2023-05-04:
.

jof-cw, 2023-05-04:
Hello Lelegard

I apologize for my long silence; I was working out of office all this time.

I tried the commands you told me and now it works fine. Thank you so much for your work and patience :)

closed 2023-05-16 by lelegard