tsduck/tsduck #1206: [Q] Using TSDuck's SRT functionality but output is incorrect.

jace-23, 2023-04-19:
Hi 
I'm using TS Duck on a CentOS9 VM as a caller / receiver for an SRT connection.
The listener / sender is an SRT device that is sending encapsulated multicast data over the internet to TSDuck.
[image]

Connectivity is all working, but I am not seeing correct output, where the I am seeing the de-encapsulated raw data at the output instead of the multicast data.

I am not sure if the issue sits with my SRT command - as my understanding is SRT should not natively de-encapsulate data - and the output should just be multicast. We have a similar setup in parallel that is only using 2 SRT devices and the output is multicast data.

Would you be able to have a look at my command below? I have even tried removing the 'fork' plugin that is duplicating the stream - this still has the same issue even with 1 output.

tsp --realtime -I -srt --caller x.x.x.x:y --local-interface 10.10.10.1 -O ip 239.10.10.1:1000 --local-address 10.10.20.1 -P fork "tsp -O ip 239.10.20.1:1000 --local-interface 10.10.30.1" &

Using either tcpdump orwireshark at the physical outputs (10.10.20.1 and 10.10.30.1) both show the PIDs and raw data instead of the multicast data.

x.x.x.x:y is the IP:port of our listener SRT device
10.10.10.1 is the interface im calling out to my SRT device on
10.10.20.1 is the 1st interface I need to output the multicast data (239.10.10.1:1000)
10.10.30.1 is the 2nd interface I need to output the multicast data (239.10.20.1:1000)

Let me know if you can assist.

Thanks 
Kind Regards
Jace

lelegard, 2023-04-19:
Hi @jace-23 

What do you mean by _"raw data instead of the multicast data"_ ? How would you define "raw data" and "multicast data"?

TSP manipulates TS packets. These packets are encapsulated in various protocols such as SRT, RIST, RTP or direct multicast. In the latter case, the UDP datagrams only contain up to 7 TS packets in their payload, nothing else. Unless you specify `--rtp` which adds RTP headers.

jace-23, 2023-04-19:
Hi @lelegard 

My understanding is that the raw data is the data before it is encapsulated into a multicast TS. 

I'm trying to send send our Multicast TS over SRT and at the output - instead of just seeing the multicast TS, I am seeing the de-encapsulated data, but we are seeingthe packets de-encapsulated.

Essentially i think my above TSDuck command is de-encapsulating the Multicast TS after it has been sent over SRT.
Where all I am trying to do send a simple multicast TS over SRT.

Let me know your thoughts

Thanks
Kind Regards 
Jace

jace-23, 2023-04-19:
Hi @lelegard 

My understanding is the raw data (PIDs) is before encapsulation - and the Encapsulator turns it into a multicast TS over IP.

I think that seemed to work. I added --rtp for the output and that solved my issue - now i am only seeing the multicast IP output after SRT. Thanks

Just to follow up, I am not sure if the 'fork' plugin is working in my command. 

I think the fork is supposed to duplicate the TS at the output - but the bit rate of the forked output is not matching - even becoming jittery and unstable.

Is there something i am missing in the fork command to make it exactly match the first output TS?

[image]

tsp --realtime -I -srt --caller x.x.x.x:y --local-interface 10.10.10.1 -O ip --rtp 239.10.10.1:1000 --local-address 10.10.20.1 -P fork "tsp -O ip --rtp 239.10.20.1:1000 --local-interface 10.10.30.1" &

Kind Regards
Jace

lelegard, 2023-04-20:
There is a lot of confusion here. When you mention "encapsulation", do you mean MPE? There is no such decapsulation in `tsp`, unless you use the `mpe` plugin of course. Adding `--rtp` just adds RTP headers, it does not change the data which remain TS packets.

lelegard, 2023-04-20:
About the `fork` plugin, I assume that the jitter is due to the buffering of the pipe. The default buffer maximizes the performance, as the expense of the latency (high throughput and low latency are incompatible, you have to choose). For an IP multicast output, I suggest to use `-P fork --buffered-packets 7` to make sure that the equivalent of one UDP datagram is pushed as soon as possible.

jace-23, 2023-04-23:
Hi @lelegard 

Yes correct, I thought that the data looked like it was being decapsulated by tsp, as on a packet capture, the data looked exactly as it did before encapsulation. However - upon adding the RTP headers, as you said, I now only see the IP multicast output address that I defined.

With the fork plugin, that worked, I also added in a bitrate using the 'regulate' plugin at the start of the fork option, which I set to what the actual bit rate of the stream was.

Thanks for your help on this matter.

closed 2023-10-01 by github-actions[bot]