tsduck/tsduck #1207: Add Managed Private UPID structure to TSDuck SCTE-35 capabuilities under segmentation_upid_type

KnoxSwift, 2023-04-20:
#### Is your feature request related to a problem?
TSDuck currenlty supports segmentation_upid_type with raw HEX entries. However, if using type 0x0C or 0x0D adds new variables that  TSDuck will error on if trying to add.

MPU() {
format_identifier  
private_data

TSduck should allow abuility to add multiple entries of MPU data with different Identifires under segmentation_upid_type 0x0C and 0x0D

#### Proposed solution:
Allow New tags
<MPU> format_identifier=HEX Value  segmentationUpidFormat=Base65/Hex/Decimal    private_data=Raw data per format type</>

#### Alternatives:
Allow reported MPU varriables under <SegmentationUpid> tag? 

#### Additional information:
MPU is section 10.3.3.3. MPU() definition and semantics om tje SCTE-35 2022 Specification.

KnoxSwift, 2023-04-20:
On further investigation it is possible to create Multiple MPU() strings within a single segmentation_upid TAG by using SCTE35 segmentation_upid type 0x0D (aka. MID) and in the RAW HEX add the MPUs with type C and correct Identifier then desired data.

There is room for improvement of TSDuck by allowing the tags: 
segmentationUpidType="12" formatIdentifier="0" segmentationUpidFormat=
under segmentation_upid
And allow under Format the use of Text/HEX/Base65 instaed of just raw hex?

The original reason this ticket was open is resolved.

Example of work-a-round:
<segmentation_upid type="0x0D" >
0C5E544553547b2261737365744964223a22313233343536375f363030313131222c2263756544617461223a7b2263756554797065223a226672616d655f6164222c226b6579223a227062222c2276616c7565223a226672616d656164227d7d
0C5E544553547b2261737365744964223a22313233343536375f363030313131222c2263756544617461223a7b2263756554797065223a226672616d655f6164222c226b6579223a227062222c2276616c7565223a226672616d656164227d7d
  </segmentation_upid>

closed 2023-04-20 by KnoxSwift

lelegard, 2023-04-21:
Hi @KnoxSwift 

The XML syntax of the signalisation, tables and descriptors, is documented in appendix D of the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf). This is where you get all structures and attributes. Section D.16 describes SCTE-defined descriptors.

In SCTE 35 2022b, the syntax diagram is clear, only one segmentation_upid() structure is allowed. This is why you get "contains 2 <segmentation_upid>, allowed 1 to 1". The XML syntax is just a consequence of this.

The internal structure of the <segmentation_upid> is left as raw hex since there are many different possible interpretations, still changing from versions to versions of the standard.

It could be possible to define as many XML sub-structures as defined in the standard and maintain this with new versions of the standard. This is just work. But you have to realise that the splice_segmentation_descriptor is just one out of 280 descriptors from various standards, MPEG, DVB, ATSC, SCTE, ISDB and private descriptors from various providers, for a total of approximately 400,000 lines of code. With one single maintainer.

Improving details in TSDuck is always welcome through [code contributions](https://tsduck.io/doxy/contributing.html). The last major addition in the descriptor set was provided by a user who needed it, implemented it and contributed it back.

futzu, 2023-06-22:
__From 2022b__
[image]