tsduck/tsduck #1208: Dektec ASI Output Stuffing not working[Q]

jtetzschner, 2023-04-21:
We are trying to ZAP a service (vbr service) from a MPTS stream, and then send it out using a Dektec 2160 Card on a ASI port. The Dektec output should be able to do stuffing, if stuffing and a bitrate is set. But this does not work. The fifo will fill up on the service vbr, and then play out on the defined bitrate fast and then go into fifo underflow.  

We use this commandstring:

/usr/bin/tsp -d -I dektec --device 0 --preload-fifo --io-standard ASI -P zap 9901 -P pcrbitrate -O dektec --bitrate 49840000 --stuffing --device 0 -c 1 --io-standard ASI --preload-fifo --maintain-preload

The Output:

 zap: found service id 0x26AD, PMT PID is 0x0B6C
* Debug: zap: adding standards DVB to MPEG
* pcrbitrate: new bitrate from PCR analysis: 9,067,349 b/s
* dektec: new input bitrate: 178,992,004 b/s
* Debug: dektec: input: got bitrate 178,992,004 b/s
* dektec: DTA-2160 output FIFO load is 6,710,660 bytes, starting transmission
* pcrbitrate: new bitrate from PCR analysis: 3,465,962 b/s
* dektec: Pausing transmission temporarily in order to maintain preload.
* dektec: new input bitrate: 178,759,643 b/s
* Debug: dektec: input: got bitrate 178,759,643 b/s
* pcrbitrate: new bitrate from PCR analysis: 6,158,343 b/s
* dektec: new input bitrate: 178,605,148 b/s
* Debug: dektec: input: got bitrate 178,605,148 b/s
* pcrbitrate: new bitrate from PCR analysis: 6,235,272 b/s
* dektec: DTA-2160 output FIFO load is 6,710,660 bytes, starting transmission
* dektec: Got FIFO underflow.
* dektec: Pausing transmission temporarily in order to maintain preload.
* dektec: new input bitrate: 178,372,375 b/s
* Debug: dektec: input: got bitrate 178,372,375 b/s
* pcrbitrate: new bitrate from PCR analysis: 8,141,326 b/s
* dektec: new input bitrate: 179,069,869 b/s
* Debug: dektec: input: got bitrate 179,069,869 b/s
* pcrbitrate: new bitrate from PCR analysis: 8,444,603 b/s
* dektec: DTA-2160 output FIFO load is 6,710,660 bytes, starting transmission
* dektec: Got FIFO underflow.
* dektec: Pausing transmission temporarily in order to maintain preload.
^C* tsp: user interrupt, terminating...

Can this be a error in the Dektec output? 

The documentation on the Dektec output --stuffing option says:

Automatically generate stuffing packets if tsp fails to provide packets fast enough.

lelegard, 2023-04-22:
Hi @jtetzschner 

The Dektec output card can automatically add a few stuffing packets "if tsp fails to provide packets fast enough". On the other hand, on the input of the card (the input of the output card, not the input card that you are separately using with the input plugin), there is a hardware FIFO which must be maintained at a reasonable level. It must not be empty; otherwise the output card will starve.

This used to be important with older ASI Dektec cards, I do not know if this is still the case. Maybe @EricDeJong can comment on this.

In your case, the bitrate before the Dektec output card is around 8 Mb/s and you program the ASI output to 49.84 Mb/s. The difference is too high. Maybe, on the output side, the card can provide stuffing but, on the input side, the input FIFO of the card is permanently starving. This is due to the buffering effect inside tsp. Packets are not passed one by one from plugin to plugin. That would eat all the CPU. There is some buffering and packets are passed in slices. Since these "slices" are produced at 8 Mb/s and consumed at 50 Mb/s, each slice is immediately consumed and the FIFO becomes empty, probably creating troubles in the ASI card (again, it was clearly the case with older models but may no longer be true).

The issue with your case is that the service is way much slower than the output bitrate. My recommendation is to feed the Dektec card with a bitrate which is slightly below 49.84 Mb/s, the output bitrate. Thus 1) the FIFO will never starve, 2) it will not overflow either if this is slightly below the output bitrate, 3) the low difference between input and output bitrate will be compensated by the stuffing option if the card.

I would recommend something like this:
~~~
tsp -I dektec ... \
    -P zap 9901 --stuffing \
    -P reduce --target-bitrate 49,800,000 \
    -P pcrbitrate \
    -O dektec --bitrate 49,840,000 --stuffing ...
~~~

The input bitrate is very high, around 179 Mb/s. Using `-P zap --stuffing`, you don't remove packets, you just replace them with stuffing packets, keeping the high bitrate. Then, `-P reduce --target-bitrate 49,800,000` removes some stuffing, but not all, to reach 49.8 Mb/s, just 40 kb/s below the final output. These 40 kb/s will be compensated by the Dektec card using its stuffing option.

I let you try and fine tune the parameters. Also RTFM to check additional parameters which can be useful.

jtetzschner, 2023-05-02:
Hi Lelegard

It seems to work quite well. When I zap several services it seems to make some cc errors. I then added the  -d to the pcrbitrate, so that the recomputed bitrate is made from the decoded timestamp instead. This only because I in the documentation read under the PCR verify plugin that:

_By default, each PCR is compared to its expected theoretical value as computed from the previous
PCR value and the transport bitrate. Note that this method works perfectly on constant bitrate streams
(CBR) only. With variable bitrate streams (VBR) with high amplitudes of variation, the reported errors
are usually meaningless.
For real-time streams, it is possible to use the input timestamps as reference which is a better
alternative for VBR streams._  

Since the services we are zap'ing are VBR I gues the -P pcrbitrate -d is the best way to do it if the same is correct for pcrbitrate as for pcrverify. Or do you know whats the best to use, dts or pcr?