tsduck/tsduck #1210: [Q] [GEN] tsduck.tables.model.xml fail on update

Dando-Real-ITA, 2023-04-27:
Hi, sometimes when updating code from github, and running a new build, tsduck fails with the following error:
```
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tstabdump
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tstabcomp
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsterinfo
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tstestecmg
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsxml
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsversion
  [DEP] tsprofiling.cpp
  [DEP] tsmux.cpp
  [DEP] setpath.cpp
  [CXX] tsmux.cpp
  [LD] /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsmux
  [GEN] tsduck.tables.model.xml
/opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsxml: symbol lookup error: /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsxml: undefined symbol: TSDUCK_LIBRARY_VERSION_3_34_3194
make[2]: *** [Makefile:67: /opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsduck.tables.model.xml] Error 127
make[1]: *** [Makefile:38: default] Error 2
make: *** [Makefile:61: default] Error 2
```
If I run `/opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/tsp --version` it shows the new version:
`tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.34-3194`
So the build is correct, it is only the generation that fails

lelegard, 2023-04-29:
Hi @Dando-Real-ITA 

The file `tsduck.tables.model.xml` is generated at the very last stage of the build because we create it using the binary `tsxml` which was just built in the previous stage. What you see is an error in the execution of the `tsxml` which was just built.

To enforce the consistency between the TSDuck shared library and its executables, the shared library exports a symbol with the library version (here `TSDUCK_LIBRARY_VERSION_3_34_3194`) and all executables reference this symbol. Thus, if there is some mess in the system or libraries and a TSDuck executable accidentally references another version of the TSDuck library, there is a symbol lookup error. This is exactly what happens here.

Inconsistencies between libraries and executables using C++ classes in that library have created a lot of obscure and hard to debug issues in the past. TSDuck uses this trick to prevent execution in case of inconsistency.

Please provide more information. Which operating system and which version do you use? Which build command? Do you already have TSDuck installed somewhere and how? Do you have defined any environment variable which affect the shared library order such as `$LD-LIBRARY-PATH`?

lelegard, 2023-04-29:
I just tried to make the build more robust against rogue definitions of `LD_LIBRARY_PATH`. Could you please update from the repo and retry the build?

Dando-Real-ITA, 2023-04-30:
Hi,
yes you are right, because my is a multi step build script, I set at start the LD_LIBRARY_PATH to the final ( post install ) lib directory, which during an update build still holds the previous version
The fix allowed the build to complete and it updated to version 3.34-3195
Thank you for all your work with tsduck!

closed 2023-04-30 by lelegard