tsduck/tsduck #1211: [Q] failing to create an SRT endoint

odev1952, 2023-04-28:
Hello,
I'm trying to create an SRT endpoint using TSP but I get an error message when running the following command:
tsp -I file --infinite t_1.ts -P regulate -O srt --listener <IP>:11340 --transtype live --messageapi
Error: srt: error during srt_bind(): Connection setup failure: unable to create/configure SRT socket: Cannot assign requested address

the destination <IP> is reachable as well as the destination por. I'm not sure what I'm not doing right here.
The command works when I create an endpoint on localhost:
tsp -I file --infinite t_1.ts -P regulate -O srt --listener localhost:11340 --transtype live --messageapi

Thanks in advance for the help.

lelegard, 2023-04-29:
Hi @odev1952 

What is `'my_endpoint_ip'`?

In listener mode, this must be a local address of your system. Is this the case here?

~~~
  -l [address:]port
  --listener [address:]port
      Use SRT in listener (or rendezvous) mode. The parameter specifies the
      IPv4 local address and UDP port on which the SRT socket listens. The
      address is optional, the port is mandatory. If --caller is also
      specified, the SRT socket works in rendezvous mode.
~~~

odev1952, 2023-04-30:
Hello! thanks for the quick reply.
I'm quite new to SRT protocol. My goal is to push an SRT stream to a distant server's public IP.
That server will then be used in caller mode:
tsp -I srt --caller locahost:11340 -P until --seconds 10 -P analyze -O drop

Is that something I can achieve?

lelegard, 2023-04-30:
I am not an SRT specialist either. The input and output plugins for SRT were initially developed by a user who needed them and he offered his code to the project.

As far as I understand your problem, there is a confusion about the words "server" and "client". What you call a "server" is probably a media server for some other protocols but it acts a client in the SRT link. So, it cannot use `-I srt --caller locahost:11340`, it must use `-I srt --caller X.X.X.X:11340` where `X.X.X.X` is an address of the machine which runs `-O srt --listener 11340`. The "SRT server" (or listener) is the latter and only defines a local port, the "SRT client" (or caller) is the former and needs to specify the IP address of the server / listener.

You may want to get more information at the SRT project: https://github.com/Haivision/srt

odev1952, 2023-04-30:
Thanks! this is actually very helpful

closed 2026-04-28 by github-actions[bot]