tsduck/tsduck #1212: --final-wait 0 is not working for a srt protocol how to make it possible using tsduck [Q]

prabhavathis08, 2023-04-29:


prabhavathis08, 2023-04-29:
The command I have used 
tsp -v -m --synchronous-log --realtime=yes -i -I srt --caller $INPUT_URL --latency 5000000  \
 -P analyze -i 30 -o report.txt  \
 -O hls --duration 2 --live 15 -p index.m3u8 index.ts

lelegard, 2023-04-29:
Please describe exactly the conditions and analysis of the problem. The command you provide does not use `--final-wait`, you do not specify any information on the context of the disconnection, etc.

By design, the option `--final-wait` works exactly the same way on all input plugins, upon a normal end of input.

prabhavathis08, 2023-05-02:
Hi ,
I have used the following command 

tsp -v -m --synchronous-log --final-wait 0 -I srt --caller $INPUT_URL --rcvbuf 48234496 \
 -P analyze -i 30 -o report.txt  \
 -P regulate --pcr-synchronous \
 -O hls --duration 15 --live 120 -p index.m3u8 index.ts 

The following output I got while the input is cut or the internet gets dropped the connection between two servers gets disconnected.

* [MON] 2023/05/02 10:39, VM: 763 MB (+516 MB), CPU:7.90% (average:7.90%)
* hls: creating media segment index-000001.ts
* [MON] 2023/05/02 10:39, VM: 763 MB (stabilizing), CPU:6.80% (average:7.35%)
* [MON] 2023/05/02 10:40, VM: 763 MB (stabilizing), CPU:7.00% (average:7.23%)
* hls: creating media segment index-000002.ts
* [MON] 2023/05/02 10:40, VM: 763 MB (-528,384 B), CPU:3.60% (average:6.32%)
* [MON] 2023/05/02 10:40, VM: 763 MB (stabilizing), CPU:0.30% (average:5.11%)
* [MON] 2023/05/02 10:40, VM: 763 MB (stabilizing), CPU:0.30% (average:4.31%)
* [MON] 2023/05/02 10:40, VM: 763 MB (stabilizing), CPU:0.30% (average:3.74%)
* [MON] 2023/05/02 10:40, VM: 763 MB (stabilizing), CPU:0.20% (average:3.29%)
* [MON] 2023/05/02 10:41, VM: 763 MB (stabilizing), CPU:0.40% (average:2.97%)
* [MON] 2023/05/02 10:41, VM: 762 MB (-823,296 B), CPU:0.20% (average:2.69%)
* [MON] 2023/05/02 10:41, VM: 762 MB (stabilizing), CPU:0.00% (average:2.45%)
* [MON] 2023/05/02 10:41, VM: 762 MB (stabilizing), CPU:0.00% (average:2.24%)

The stream is not running anymore...

prabhavathis08, 2023-05-02:
Okay, When internet was dropped how to reconnect the SRT caller to listener? @lelegard

lelegard, 2023-05-03:
> Okay, When internet was dropped how to reconnect the SRT caller to listener? @lelegard

I often wonder why people always want me to give _my_ time for free instead of investing some of _their_ time to search for a solution, starting with reading the documentation for instance, or doing some tests.

Did you read the documentation? Did you find any reconnection option in the documentation for the `srt` input plugin?

There is none. An end of stream is an end of stream and `tsp` terminates. If you want to restart it, you may write a script around your `tsp` command.

closed 2023-10-29 by github-actions[bot]