tsduck/tsduck #1214: question: how to capture entire ts file from a DVBS transponder via tbs5530 and tsduck

stulluk, 2023-05-02:
Hi,

First of all, I managed to achieve this under ubuntu22.04 with mumudvb but I want to learn to use tsduck.

I have a TBS5530 and a diseqc multiswitch with multiple satellites. I asked this question to chatgpt but it gave me no more than wrong commandline usage of tsdump :)

Here is my command that can capture whole TS with mumudvb:

```bash
mumudvb -dt -c 11053h8000.conf --dumpfile tmpdir/new6.ts
```

And here is the config file contents:

```bash
tuner=1
freq=11053
pol=h
srate=8000
delivery_system=DVBS
sat_number=1
#switch_input=1
diseqc_repeat=1
autoconfiguration=full
```

I checked tsdump utility code but I couldn't understand how does it set delivery_system as DVBS and how to set diseqc commands  ?

is it possible to achieve this with tsduck utilities ?

lelegard, 2023-05-03:
Hi @stulluk 

The command `tsdump` reads a TS file (or pipe). Its purpose is to analyze TS packets one by one, a real hexadecimal dump of packets. Probably not what you need.

There are many utilities and plugins in TSDuck. You need to browse through the [user's guide](https://tsduck.io/download/docs/tsduck.pdf) first. Have a look at the reference doc for the `tsp` command (the transport stream processor), it `dvb` plugin (to read a stream from a tuner) and the chapter with sample usages.

stulluk, 2023-05-03:
Hi @lelegard 

Thank you for your fast response. I am reading page 225 in the userguide.

So I tried:

```bash
stulluk ~ $  tsp -I dvb --adapter 0 --delivery-system DVB-S --frequency 11054 --polarity vertical --symbol-rate 30000
* Error: dvb: deliver system DVB-S not supported on tuner /dev/dvb/adapter0
stulluk ~ $
```
TBS5530 has two frontends. One DVBS and one DVBT/C.

Here is how my device looks like:

```bash
[10218.009531] usb 1-2: USB disconnect, device number 2
[10218.009966] dvb-usb: TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x successfully deinitialized and disconnected.
[10223.188118] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[10223.414331] usb 1-2: New USB device found, idVendor=734c, idProduct=5530, bcdDevice= 0.01
[10223.414338] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[10223.416394] dvb-usb: found a 'TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0' in cold state, will try to load a firmware
[10223.416402] dvb-usb: downloading firmware from file 'dvb-usb-id5530.fw'
[10223.416404] tbs5530: start downloading tbs5530 firmware
[10223.812550] tbs5530: could not restart the USB controller CPU.
[10223.920234] usb 1-2: USB disconnect, device number 5
[10225.664093] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[10225.901106] usb 1-2: New USB device found, idVendor=734c, idProduct=5530, bcdDevice= 0.01
[10225.901115] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10225.901118] usb 1-2: Product: TBS 5530
[10225.901120] usb 1-2: Manufacturer: TBS-Tech
[10225.912180] dvb-usb: found a 'TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0' in cold state, will try to load a firmware
[10225.912188] dvb-usb: downloading firmware from file 'dvb-usb-id5530.fw'
[10225.912190] tbs5530: start downloading tbs5530 firmware
[10226.412088] dvb-usb: found a 'TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0' in warm state.
[10226.412226] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[10226.412228] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[10226.422470] dvbdev: DVB: registering new adapter (TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0)
[10227.957916] dvb-usb: MAC address: 00:22:ab:c0:78:89
[10227.972911] i2c i2c-3: Detect CXD2878/CXD6802(SiP) chip.
[10227.972917] usb 1-2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0 DVB-T/T2/C/C2,ISDB-T/C,ATSC1.0)...
[10228.203895] i2c i2c-3: found a 'Montage m88rs6060' in cold state
[10228.203948] i2c i2c-3: downloading firmware from file 'dvb-demod-m88rs6060.fw'
[10230.548690] i2c i2c-3: found a 'Montage m88rs6060' in warm state
[10230.548697] i2c i2c-3: firmware version:30
[10230.696053] usb 1-2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC DVB-S/S2/S2X)...
[10230.703018] dvb-usb: TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0 successfully initialized and connected.

^C
stulluk ~ $  cd ~
stulluk ~ $  
stulluk ~ $  
stulluk ~ $  sudo tslsdvb --debug
tslsdvb: Debug: debug level set to 1
tslsdvb: Debug: ====> tslsdvb --debug
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/idVendor = "734c" (success)
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/idProduct = "5530" (success)
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/manufacturer = "TBS-Tech" (success)
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/product = "TBS 5530" (success)
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/version = "2.00" (success)
tslsdvb: Debug: /sys/devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-2/dvb/dvb0.frontend0/device/serial = "" (failure)
tslsdvb: error opening /dev/dvb/adapter0/frontend1: Device or resource busy
stulluk ~ $  mumudvb -l
MuMuDVB Version 2.1.0_5e9959e366b815ac498f2dc2823ac028d59a5d5f
 --- Build information ---
Built with CAM support.
Built without SCAM support.
Built with ATSC support.
Built with ATSC long channel names support.
Built with support for DVB API Version 5.11.
Built with support for DVB-T2.
---------
Originally based on dvbstream 0.6 by (C) Dave Chapman 2001-2004
Released under the GPL.
Latest version available from http://mumudvb.net/
Project from the cr@ns (http://www.crans.org)
by Brice DUBOST (mumudvb@braice.net)

Info:  DVB:  ==================================
Info:  DVB:          DVB CARDS LISTING
Info:  DVB:  ==================================
Info:  DVB:  =========== Card 0 - Tuner 1 ===========
Info:  DVB:   Frontend : TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC DVB-S/S2/S2X
Info:  DVB:   Satellite (DVB-S) card
Info:  DVB:   Frequency: 950000 kHz to 2150000 kHz
Info:  DVB:   Symbol rate: 1000 k symbols/s to 45000 k symbols/s 
Info:  DVB:  
Info:  DVB:  =========== Card 0 - Tuner 0 ===========
Info:  DVB:   Frontend : TurboSight TBS 5530 DVB-T2/T/C/S/S2/S2x,ISDB-T,ATSC1.0 DVB-T/T2/C/C2,ISDB-T/C,ATSC1.0
Info:  DVB:   Terrestrial (DVB-T) card
Info:  DVB:   Frequency: 45000 kHz to 868000 kHz
Info:  DVB:  
stulluk ~ $  
```

So, in the tsp -I dvb commandline, how can I select frontend ???

lelegard, 2023-05-03:
> `$  tsp -I dvb --adapter 0 --delivery-system DVB-S --frequency 11054 --polarity vertical --symbol-rate 30000`
> `* Error: dvb: deliver system DVB-S not supported on tuner /dev/dvb/adapter0`
> TBS5530 has two frontends. One DVBS and one DVBT/C.

Then the DVB-S one is probably on adapter1. Try command `tslsdvb` to list all tuners and their characteristics.

stulluk, 2023-05-03:
@lelegard  I already put tslsdvb output above ?

stulluk, 2023-05-03:
For future readers, I managed to capture whole TS from a transponder via following oneliner:

```bash
tsp --debug  -I dvb -d /dev/dvb/adapter0:1:0:0 --delivery-system DVB-S2 --satellite-number 0 --lnb Extended --frequency 11054000000 --polarity vertical --symbol-rate 30000000 > outfile.ts
```
Created outfile.ts can be played with VLC / TSReader...etc.

@lelegard thank you.
In the Fking Manual, it mentions "--satellite-frequency" can be used if the intermediate frequency is not known.

```bash
stulluk ~ $  tsp --debug  -I dvb -d /dev/dvb/adapter0:1:0:0 --delivery-system DVB-S2 --satellite-number 0 --lnb Extended --satellite-frequency 11054000000 --polarity vertical --symbol-rate 30000000 > outfile.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: trying to load "/usr/bin/tsplugin_dvb.so"
* Debug: /usr/bin/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/dvb.so"
* Debug: /usr/bin/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_dvb.so"
* Debug: /usr/lib64/tsduck/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/dvb.so"
* Debug: /usr/lib64/tsduck/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_dvb.so"
* Debug: /usr/lib64/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/dvb.so"
* Debug: /usr/lib64/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_dvb.so"
* Debug: registering input plugin "dvb", status: ok
* Debug: dvb: debug level set to 1
* Error: dvb: unknown option --satellite-frequency
* Error: dvb: no parameter allowed, use options only
* Debug: dvb: ====> tsp -I dvb -d /dev/dvb/adapter0:1:0:0 --delivery-system DVB-S2 --satellite-number 0 --lnb Extended --satellite-frequency 11054000000 --polarity vertical --symbol-rate 30000000
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file 
* Debug: error when initializing the plugins
* Debug: Report logging thread terminated
stulluk ~ $
```
Or maybe I am still not reading carefully...

Anyway, I will try more TP's from both Turksat & Hotbird, lets see what we can do with it.

lelegard, 2023-05-03:
> In the Fking Manual, it mentions "--satellite-frequency" can be used if the intermediate frequency is not known.
> Or maybe I am still not reading carefully...

Not carefully enough indeed... 😄 

There is no option `--satellite-frequency` in the `dvb` plugin. The value of `--frequency` is always the satellite frequency. The intermediate frequency is automatically computed based on the `--lnb` description.

There is an option `--satellite-frequency` in the `dektec` input and output plugins. This is because the Dektec devices are more frequently used in labs where we speak in terms of "frequency on the wire" and there is often no antenna at all. In that case, there is no LNB, no distinct frequencies, just directly connected modulators and tuners and `--frequency` is the direct frequency on the device. In that context, `--satellite-frequency` is a feature which was added later, in case the lab guys really insist to use the equivalent frequencies if there was a satellite in the lab...

The `dvb` plugin, on the other hand, is dedicated to cheap consumer products where most users don't even know what an intermediate frequency is.

stulluk, 2023-05-03:
Great explanation. Thank you @lelegard 

I hope this issue helps someone else trying to achieve the same thing in the future.

Closing.

closed 2023-05-03 by stulluk