tsduck/tsduck #1215: [Q] UDP input to UDP output weird behavior

Litrona, 2023-05-03:
I am testing a hardware TS muxer, and found a weird behavior when passed thru tsp.
I am using tsp version: 3.34-3197
OS: Linux  5.10.0-17-amd64 #1 SMP Debian 5.10.136-1 (2022-08-13) x86_64 GNU/Linux

I get the TS from the muxer at 239.2.2.2:2001 and just use tsp to send it to 239.2.2.2:2002
I use TSReader Lite to analyze the stream.

Input stream in TSReader says the mux bitrate is 19,914,423 bps (very near to the one set in the hw 19,906,000 bps.
However when analyzing the output from tsp, the mux rate found by TSReader is 39,931,005 bps.

What is supposed to be happenning here ?.

Thanks in advance and regards

Litrona, 2023-05-03:
TSReader says to measures:
- Mux bitrate, based on PCR contents
- Last sec, based on bytes received on latest bytes received.

I also saw lots of continuity errors in the ip output of tsp.

`tsp -I ip 239.2.2.2:2001 -O ip 239.2.2.2:2002`

Measures are very awesome, how can be the ip output bitrate almost double the ip input bitrate ?

Litrona, 2023-05-03:
OK, I found the issue.
I have a cmdline UDP measure tool.
If I measure the UDP 239.2.2.2:2001 from the muxer, it says: 
bps 19666304 size 1316

Once I run this cmdline:
`tsp -I ip 239.2.2.2:2001 -O drop`
The measure of UDP 239.2.2.2:2001 stream doubles:
bps 39806370 size 1316

lelegard, 2023-05-03:
I didn't really get if TSReader measures the same source as tsp or if it measures the output of tsp, after relaying its input. In my previous response, I assumed that TSReader measured the output of tsp.

If TSReader and tsp runs on the same system and receive the same source, they both subscribe to the same multicast stream. Depending on the OS, Linux, macOS, Windows, the behavior is "variable" (meaning shitty). Sometimes, you receives one stream, sometimes twice, sometimes half of it. On a given OS, the behavior is consistent but not the same on all OS.

Litrona, 2023-05-13:
I guess what is happening here.
I have 2 ifaces, so if I run:
`tsp -I ip 239.2.2.2:2001 -O ip 239.2.2.2:5000`
and don't specify a local address to bind to, the socket will subscribe the mcast group 239.2.2.2 on both ifaces, instead of just one of them.
I would also disable IP_MULTICAST_LOOP =0 under IPPROTO_IP, to disable loopback of multicast packets.

lelegard, 2023-05-14:
> I would also disable IP_MULTICAST_LOOP =0 under IPPROTO_IP, to disable loopback of multicast packets.

Do you suggest we should add this to the `ip` output plugin? This would prohibit test scenarios where a receiver runs on the same system. Or we may add a command line option for IP_MULTICAST_LOOP.

lelegard, 2023-05-14:
Added option `--disable-multicast-loop` to ip output plugin in commit 1399bff65966e81724885e8b6afb399f493a1aa5.

Litrona, 2023-05-15:
Great, thanks a lot

Litrona, 2023-05-15:
Hmm, I have tested it, and seems not having effect at all.
Are you sure, that must be set in the IP output, not in the IP input ?

lelegard, 2023-05-15:
> Are you sure, that must be set in the IP output, not in the IP input ?

It seems that this option applies to the _output_ socket on Linux and the _input_ socket on Windows. See https://learn.microsoft.com/en-us/windows/win32/winsock/ip-multicast-2

I have always been amazed by the ability of Windows to integrate Unix concepts with a carefully incompatible API or semantics (or both).

Since you use Linux, yes, this is an output socket option. I also added a debug trace just before `setsockopt()` and, yes `--disable-multicast-loop` sets `IP_MULTICAST_LOOP`to 0.

lelegard, 2023-05-15:
Considering the Windows oddity, I added `--disable-multicast-loop` on UDP input, for Windows systems only. It shouldn't have any effect on Linux.

closed 2023-10-15 by github-actions[bot]