tsduck/tsduck #1219: tsscan.exe unable to use tuner in Windows

DualTachyon, 2023-05-14:
#### Bug description:
I have a TBS6904SE quad DVB-S2 tuner. tsscan fails with a QueryInterface error (only visible with --debug=XXX), which leads to a "Program aborted: UninitializedVariable: uninitialized variable".

#### How to reproduce:
bin\tsscan --nit-scan -f 10729000 -s 22000 --debug=255

#### Expected behavior:
I'm not sure if my command line above is complete enough to run a scan, but I would expect it to either do a scan or tell me some other parameter is missing.

#### Errors and logs:
~~~
C:\Storage\TSDuck>bin\tsscan --nit-scan -f 10729000 -s 22000 --debug=255
tsscan: Debug[255]: debug level set to 255
tsscan: Debug: ====> tsscan --nit-scan -f 10729000 -s 22000 --debug=255
tsscan: Debug: looking for DVB adapter number ""
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: CreateClassEnumerator: success
tsscan: Debug[10]: IMoniker::BindToStorage: success
tsscan: Debug[10]: IPropertyBag::Read: success
tsscan: Debug: found tuner filter "TBS 6904SE DVBS/S2 Tuner 1"
tsscan: Debug[10]: IMoniker::GetDisplayName: success
tsscan: Debug: tuner device path: @device:pnp:\\?\pci#ven_544d&dev_6178&subsys_00206904&rev_00#6&18819866&0&0010003a#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{8aaa8912-fa33-448c-b78d-1a84379c0ca3}
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: IUnknown::QueryInterface: success
tsscan: Debug[10]: IUnknown::QueryInterface: success
tsscan: Debug[10]: IUnknown::QueryInterface: success
tsscan: Debug[10]: IMoniker::BindToObject: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: IUnknown::QueryInterface: success
tsscan: Debug: DirectShowGraph init, graph build is valid: true, media control is valid: true
tsscan: Debug: adding filter "NetworkProvider", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug: adding filter "Tuner", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug[10]: ITunerCap::get_SupportedNetworkTypes: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: error setting CarrierFrequency: success
tsscan: Debug[10]: error setting Modulation: success
tsscan: Debug[10]: error setting InnerFEC: success
tsscan: Debug[10]: error setting InnerFECRate: success
tsscan: Debug[10]: error setting OuterFEC: success
tsscan: Debug[10]: error setting OuterFECRate: success
tsscan: Debug[10]: error setting SymbolRate: success
tsscan: Debug[10]: error setting SignalPolarisation: success
tsscan: Debug[10]: error setting WestPosition: success
tsscan: Debug[10]: error setting OrbitalPosition: success
tsscan: Debug[10]: error setting Azimuth: success
tsscan: Debug[10]: error setting Elevation: success
tsscan: Debug[10]: error setting DiseqLNBSource: success
tsscan: Debug[10]: error setting LocalLNBSwitchOverride: success
tsscan: Debug[10]: error setting LocalOscillatorOverrideLow: success
tsscan: Debug[10]: error setting LocalOscillatorOverrideHigh: success
tsscan: Debug[10]: error setting LocalSpectralInversionOverride: success
tsscan: Debug[10]: error setting SignalRollOff: success
tsscan: Debug[10]: error setting SignalPilot: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug: initializing tuning space "TSDuck DVB-S Tuning Space"
tsscan: Debug[10]: error setting UniqueName on TSDuck DVB-S Tuning Space: success
tsscan: Debug[10]: error setting FriendlyName on TSDuck DVB-S Tuning Space: success
tsscan: Debug[10]: error setting _NetworkType: success
tsscan: Debug[10]: error setting SystemType: success
tsscan: Debug[10]: error setting NetworkID: success
tsscan: Debug[10]: error setting LNBSwitch: success
tsscan: Debug[10]: error setting LowOscillator: success
tsscan: Debug[10]: error setting HighOscillator: success
tsscan: Debug[10]: error setting SpectralInversion: success
tsscan: Debug[10]: error setting DefaultLocator: success
tsscan: Debug: user-specified receiver filter name: ""
tsscan: Debug: trying direct connection from tuner (no receiver)
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug: adding filter "Tee", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: failed to connect pins: No combination of intermediate filters could be found to make the connection.
tsscan: Debug: failed to connect filters
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IFilterGraph::RemoveFilter: success
tsscan: Debug[10]: IFilterGraph::RemoveFilter: success
tsscan: Debug[10]: IFilterGraph::RemoveFilter: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: CreateClassEnumerator: success
tsscan: Debug[10]: IMoniker::BindToStorage: success
tsscan: Debug[10]: IPropertyBag::Read: success
tsscan: Debug: trying receiver filter "TBS 6904SE DVBS/S2 Capture 1"
tsscan: Debug[10]: IMoniker::BindToObject: success
tsscan: Debug: adding filter "Receiver", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug: adding filter "Tee", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug: adding filter "Sink/Capture", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug: adding filter "Demux", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug: adding filter "TIF", graph valid: true, filter not null: true
tsscan: Debug[10]: IFilterGraph::AddFilter: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: IBaseFilter::EnumPins: success
tsscan: Debug[10]: failed to connect pins: success
tsscan: Debug: using receiver filter "TBS 6904SE DVBS/S2 Capture 1"
tsscan: Debug: IBDA_DigitalDemodulator in tuner: 2
tsscan: Debug: IBDA_DigitalDemodulator2 in tuner: 2
tsscan: Debug: IBDA_SignalStatistics in tuner: 2
tsscan: Debug: IKsPropertySet in tuner: 2
tsscan: Debug: found tuner device "TBS 6904SE DVBS/S2 Tuner 1"
using default deliver system DVB-S
tsscan: Debug: adding standards DVB to none
tsscan: Debug[10]: setting tuning space TSDuck DVB-S Tuning Space: success
tsscan: Debug[10]: cannot create DirectShow tune request: success
tsscan: Debug[10]: IUnknown::QueryInterface: success
tsscan: Debug[10]: error setting ONID: success
tsscan: Debug[10]: error setting TSID: success
tsscan: Debug[10]: error setting SID: success
tsscan: IUnknown::QueryInterface: No such interface supported
tsscan: Debug[10]: CoCreateInstance: success
tsscan: Debug[10]: error setting CarrierFrequency: success
Program aborted: UninitializedVariable: uninitialized variable
~~~

#### Environment:
 - OS: Windows
 - OS version: 11, with latest updates
 - TSDuck full version: 3.34-3197
 - Installation type: Binaries from your github release page

#### Additional information:

lelegard, 2023-05-14:
Hi @DualTachyon,

Thanks for the details report.

There are several topics here:

1. A bug (I just fixed it)
2. A workaround for that bug, until you get a fixed version.
3. Incorrect parameters values
4. Probably missing parameters

[1] The bug is specific to DVB-S on Windows. For each system delivery type (DVB-S, DVB-S2, DVB-T, etc), there are mandatory modulation parameters. When one of them is missing from the command line, a default is provided. These default values are documented in the user's guide or with option `--help`. Using DVB-S, only one modulation is permitted: QPSK. So, this is not really a parameter for DVB-S and no default is supplied. On Linux, there is no need to set the modulation on a DVB-S transponder. On Windows, however, the driver requires the modulation to be explicitly specified, even though QPSK is the only allowed modulation. Since no default was supplied and none was set on the command line, there was this "uninitialized" exception.

I just fixed it. A default value for modulation is now always specified for DVB-S. The Windows binaries will be available tomorrow, after the nightly build, in the [prerelease area](https://tsduck.io/download/prerelease/).

[2] In the meantime, the workaround is obvious: add `--modulation qpsk` on the command line.

[3] `-f 10729000 -s 22000` are probably incorrect parameters. I assume you mean 10 GHz and 22 mega-symbols/s. However, TSDuck uses "real" or "basic" units, meaning Hz and sym/s. So, you should use `-f 10,729,000,000 -s 22,000,000` (the commas are optional and ignored). Your values are expressed in units which are specific to DVB-S. DVB-T uses different ranges of values. Not to mention ATSC, ISDB, etc. Each one has its own "usual" subunits (kHz, deci-mega-whatever). As a universal tool with universal units, TSDuck always uses the basic units without implicit k, M or G.

[4] Most DVB-S tuners need _all_ parameters to be specified in order to tune to a transponder. So, you probably need to also specify `--fec-inner`, `--polarity` and maybe `--spectral-inversion`, unless the documented default values match yours. Note that this is different from DVB-T where most tuners are able to "guess" the other parameters when you specify the frequency.

DualTachyon, 2023-05-14:
I'm glad you fixed the problem, I will give the new binary a try when it comes out. Thanks.

I specified the modulation and it does indeed work (-a 2 because the default tuner is not the one I expected), see below:

C:\Storage\TSDuck>bin\tsscan --nit-scan -f 10729000000 -s 22000000 --modulation 8-PSK -a 2
* Frequency: 11,214,250,000 Hz H (DVB-S2, 8-PSK), locked: yes, strength: 0.09 dB
  Transport stream id: 1001, 0x03E9
* Frequency: 11,229,000,000 Hz V (DVB-S2, 8-PSK), locked: yes, strength: 0.09 dB
  Transport stream id: 1002, 0x03EA
* Frequency: 11,243,750,000 Hz H, locked: yes, strength: 0.09 dB
  Transport stream id: 1003, 0x03EB
* Frequency: 11,258,500,000 Hz V (DVB-S2, 8-PSK), locked: yes, strength: 0.09 dB
  Transport stream id: 1004, 0x03EC
* Frequency: 11,273,250,000 Hz H (DVB-S2, 8-PSK), locked: yes, strength: 0.09 dB
  Transport stream id: 1005, 0x03ED
* Frequency: 11,288,000,000 Hz V (DVB-S2, 8-PSK), locked: yes, strength: 0.09 dB
  Transport stream id: 1006, 0x03EE
* Frequency: 11,302,750,000 Hz H (DVB-S2, 8-PSK), locked: yes, strength: 0.089 dB
* 
...

lelegard, 2023-05-14:
Great. And it must be DVB-S2 if the modulation is 8-PSK.

DualTachyon, 2023-05-15:
I can confirm your fix works fine, the uninitialised error is no longer appearing.

closed 2023-05-15 by DualTachyon