tsduck/tsduck #1221: [FEAT]

deejaydeeF, 2023-05-21:
#### Is your feature request related to a problem?

Several informations are missing in the output provided by tsanalyze (or tsp -P analyze) plugin. I find them very useful to better identify streams & programs carried by the TS.
Could you please add the Network name description field in the NIT PID description ?
Could you also add the CA-descriptor id from PMT table, when available ? (it is necessary to identify some specific encryption like BISS, otherwise some programs are marked as encrypted without knowing about the encryption type)
 
Many thanks in advance and all my gratitude for this tremendous job !

lelegard, 2023-05-21:
Hi @deejaydeeF 

> Could you please add the Network name description field in the NIT PID description ?

Could be done.

> Could you also add the CA-descriptor id from PMT table, when available ?

The EMM and ECM PID are already identified by CAS id, or name and usage when known. See an [EMM example](https://github.com/tsduck/tsduck-test/blob/04915a891b9b6254dd30316a2c15bee58ddf5603/reference/test-001/test-001.tsanalyze.full.txt#L51) and an [ECM example](https://github.com/tsduck/tsduck-test/blob/04915a891b9b6254dd30316a2c15bee58ddf5603/reference/test-001/test-001.tsanalyze.full.txt#L86)

deejaydeeF, 2023-05-21:
> Hi @deejaydeeF
>
>
> Could be done.
>
>
> The EMM and ECM PID are already identified by CAS id, or name and usage when known. See an [EMM [...]

Sorry but there is no EMM and ECM for BISS encryption, it is only represented as a parameter of the PMT information (checked with Wireshark, one example below:)
Descriptor Tag=0x09
    Descriptor Tag: CA Descriptor (0x09)
    Descriptor Length: 4
    System ID: 0x2600
    111. .... .... .... = Reserved: 0x7
    ...1 1111 1111 1111 = CA PID: 0x1fff

lelegard, 2023-05-22:
> Sorry but there is no EMM and ECM for BISS encryption

I do not know BISS, just heard of it. That must be a fixed-key encryption if there is no ECM, a very unusual CAS configuration (and a joke in terms of security).

The command `tsanalyze` gives an overview of the structure of the TS. It cannot give the tiny details. There are zillions of potential information in a PMT. They cannot appear in the `tsanalyzer` report, in which case it would no longer be an overview. The vast majority of CAS (in fact all but BISS) are materialized by ECM and EMM PID's which are listed in the report. Otherwise, you need more in-depth analysis of the signalization. The command `tstables` (and the corresponding `tables` plugin) can fully analyze the various PMT's.

BTW, the network id and name have been added in the NIT PID in commit f6de30c.

closed 2023-10-22 by github-actions[bot]