tsduck/tsduck #1223: Wants to take input through memory

prashant-09-tech, 2023-05-24:
Hi @lelegard ,
I want to take input from memory and I am using below command but it is working .
**tsp -I memory  -e 1028  -P pcrbitrate -P regulate -O ip 127.0.0.1:10201**  .
 And I am using this command for memory output . 
tsp -I file -i  "D:\workspace\SCTE\test_scte\sample.ts" -P pcrbitrate -P regulate -O memory 

So please suggest  how can i take input from memory .

lelegard, 2023-05-24:
Hi @prashant-09-tech 

Please RTFM, especially the reference section for the `memory` plugins (input and output):

> This is a developer plugin. It is useful only to C++, Java or Python developers who run an instance of TSProcessor pipeline inside their applications and want this application to directly interact with the input of the pipeline. Using this plugin in a tsp command line does nothing.
>
> In practice, this plugin is useful to Java and Python developers only since it is not possible to develop TSDuck plugins in these languages. To develop some custom form of transport stream input, C++ developers should rather develop their own input plugin. This is simpler and more efficient than handling input events for the memory plugin.
>
> Developers may start from the sample code which is available for the three languages in the TSDuck source code tree in the directory sample/sample-memory-plugins.

Please note _"Using this plugin in a tsp command line does nothing."_

This is rather logical. What does mean "getting TS packets from memory"? Which memory? In a `tsp` command, the "memory" is the virtual address space of the process running the `tsp` executable, over which the user has no control.

As explained in the FM, these plugins are useful only in Python and Java applications which use the TSDuck library (not the `tsp` executable). In that case, "memory" means the virtual address space of the user's Python or Java application. The user's application may produce TS packets internally and pass them to an embedded plugin chain using the `memory` plugin. Again, I insist that this plugin is useful only in the context of a user's application running an embedded plugin chain, not in the context of the `tsp` application.

As also explained in the FM, C++ developers may use this plugin as well but it is much easier to write their own input plugin since C++ is the native language of TSDuck. The `memory` input and output plugins act as inter-language gateways for Python and Java developers.

prashant-09-tech, 2023-05-24:
Thank you for these solution

closed 2023-05-24 by prashant-09-tech