tsduck/tsduck #1224: Make fails on Ubuntu 18.04.6

sewelln95, 2023-05-25:
Hi @lelegard 

When trying to build the software using the following on Ubuntu 18.04.6: 

`make NOPCSC=1 NOCURL=1 NODTAPI=1 NOSRT=1` 
`make install`

The operation fails with the following:

~~~[LD] /home/tsduck/bin/release-x86_64-[REDACTED]/tssmartcard
/home/tsduck/bin/release-x86_64-[REDACTED]/objs-tstools/tssmartcard.o: In function `MainCode(int, char**)':
tssmartcard.cpp:(.text+0x1493): undefined reference to `ts::pcsc::GetStates(long, std::vector<ts::pcsc::ReaderState, std::allocator<ts::pcsc::ReaderState> >&, unsigned int)'
tssmartcard.cpp:(.text+0x182f): undefined reference to `ts::pcsc::Transmit(long, unsigned int, void const*, unsigned long, void*, unsigned long, unsigned short&, unsigned long&)'
/home/tsduck/bin/release-x86_64-[REDACTED]/objs-tstools/tssmartcard.o: In function `(anonymous namespace)::Check(long, (anonymous namespace)::Options&, ts::UString const&) [clone .constprop.98]':
tssmartcard.cpp:(.text+0x3475): undefined reference to `ts::pcsc::StrError(long)'
collect2: error: ld returned 1 exit status
../../Makefile.inc:554: recipe for target '/home/tsduck/bin/release-x86_64-[REDACTED]/tssmartcard' failed
make[2]: *** [/home/tsduck/bin/release-x86_64-[REDACTED]/tssmartcard] Error 1
Makefile:43: recipe for target 'install-tools' failed
make[1]: *** [install-tools] Error 2
Makefile:156: recipe for target 'install' failed
make: *** [install] Error 2
~~~

lelegard, 2023-05-25:
Hi @sewelln95 

I assume that the errors occur during `make install`, the second command. Building the tools and libraries is a requirement for installation. Thus, `make` rebuilds first any missing tool or library. Since `tssmartcard` was not built during the first `make` command because of `NOPCSC=1`, it tries to rebuild it. And fails because the smartcard-related features were not built into the TSDuck library.

You have to repeat the same exclusion options in the `make install` command:
~~~
make NOPCSC=1 NOCURL=1 NODTAPI=1 NOSRT=1 install
~~~

sewelln95, 2023-05-26:
Thank you that got me a bit further, then it failed with the following:
~~~/home/tsduck/bin/release-x86_64-[REDACTED]/include/tsduck.h /usr/include/tsduck
echo '#define TS_NO_PCSC 1' >>/usr/include/tsduck/tsPreConfiguration.h
echo '#define TS_NO_DTAPI 1' >>/usr/include/tsduck/tsPreConfiguration.h
echo '#define TS_NO_VATEK 1' >>/usr/include/tsduck/tsPreConfiguration.h
echo '#define TS_NO_CURL 1' >>/usr/include/tsduck/tsPreConfiguration.h
echo '#define TS_NO_SRT 1' >>/usr/include/tsduck/tsPreConfiguration.h
echo '#define TS_NO_RIST 1' >>/usr/include/tsduck/tsPreConfiguration.h
  File "/home/tsduck/scripts/get-version-from-sources.py", line 59
    print('invalid option: %s' % arg, file=sys.stderr)
                                          ^
SyntaxError: invalid syntax
install -d -m 755 /usr/share/pkgconfig
sed -e 's|{{VERSION}}||g' \
    -e 's|{{PREFIX}}|/usr|g' \
    -e 's|{{LIBDIR}}|/usr/lib|g' \
    -e 's|srt||g' \
    -e 's|librist||g' \
    -e 's|libpcsclite||g' \
    -e 's|libcurl||g' \
     \
     \
     \
     \
     \
    -e 's| *-lprocstat *| |g' \
    -e 's| *-lkvm *| |g' \
    -e 's|:[ ,][ ,]*|: |g' -e 's|,[ ,][ ,]*|, |g' -e 's|[ ,]*$||' -e '/: *$/d' \
    tsduck.pc >/usr/share/pkgconfig/tsduck.pc
chmod 644 /usr/share/pkgconfig/tsduck.pc~~~

lelegard, 2023-05-26:
This is probably because you do not have python3 installed on your system and/or is not the default python command.

Python 3 was released 15 years ago, in 2008, and many distros continue to use python2 as only or default Python, creating many errors like yours in recent (or not so recent) scripts.

Before building TSDuck, the doc recommends to run `scripts/install-prerequisites.sh` to make sure everything is correctly installed and configured. However, you specify to ignore several dependencies such as PCSC and I assume you did not run the prerequisites script. This is fine but you should read it and make sure that everything you need, minus the dependencies your explicitly exclude, are present and configured.

lelegard, 2023-05-27:
For convenience, I have updated the install prerequisites script to accept the same exclusion options as `make`. It installs all prerequisites to build TSDuck, except those which are not necessary because of your options.

After pulling the current state of the repo, you can now use:
~~~
scripts/install-prerequisites.sh NOPCSC=1 NOCURL=1 NODTAPI=1 NOSRT=1
make NOPCSC=1 NOCURL=1 NODTAPI=1 NOSRT=1
make NOPCSC=1 NOCURL=1 NODTAPI=1 NOSRT=1 install
~~~

closed 2023-10-29 by github-actions[bot]