tsduck/tsduck #1226: Pulled a new version, built and install but tsp now fails

joolzg, 2023-06-02:
sudo make NOEDITLINE=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOPCSC=1 NOSRT=1 NORIST=1 -j16 install

tsp -I file a.ts -P analyze -O file b.ts
* Error: processor plugin analyze not found

No plugins found anymore

Any ideas

lelegard, 2023-06-02:
Hi @joolzg 

Please specify the operating system and version. Please attach a build log, especially the installation part.

Did you define environment variables TSPLUGINS_PATH or LD_LIBRARY_PATH ?

Please post the output of `tsp --debug -I file a.ts -P analyze -O file b.ts`.

joolzg, 2023-06-02:
Solved, plugins are installed in /usr/lib/tsduck but with the debug it shows its looking at /usr/bin

tsp --debug -I file a.ts -P analyze -O file b.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file a.ts
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file b.ts
* Debug: trying to load "/usr/bin/tsplugin_analyze.so"
* Error: processor plugin analyze not found
* Debug: error when initializing the plugins
* Debug: Report logging thread terminated

closed 2023-06-02 by joolzg