tsduck/tsduck #1227: [BUG] tables stuck in an infinite loop within Service Prominence Descriptor

AlfonZ42, 2023-06-04:
#### Bug description:

_tstables_ (or _tables_ plugin) can get stuck in an infinite loop when fed certain data.

#### How to reproduce:

`tstables test.ts > test.txt`
The used file is [test.zip](https://github.com/tsduck/tsduck/files/11644485/test.zip) (gotten from #9 via `tsp -I file T2MI_CAPITAL.ts -P skip 14428 -P until -p 27 -O file test.ts`).
The last output is:
~~~
...
- Descriptor 0: Extension Descriptor (0x7F, 127), 68 bytes
  Extended descriptor: Service Prominence (0x22, 34)
  SOGI flag: false, priority: 1908, service id: 55917
  SOGI flag: true, priority: 2966
  SOGI flag: true, priority: 1143, service id: 55160
  SOGI flag: false, priority: 304, service id: 45243
  SOGI flag: false, priority: 813, service id: 53709
  Primary region: 191, secondary region: 254, tertiary region: 18944
  Primary region: 223, secondary region: 217
  Country: &:, primary region: 185
  Country: s
  Primary region: 179, secondary region: 73, tertiary region: 6235
  Country: , primary region: 23, secondary region: 11, tertiary region: 49939
  Country: ^e3, primary region: 171, secondary region: 152
  Country: !, primary region: 218, secondary region: 207
  Country: o
  Country: , primary region: 255
  SOGI flag: false, priority: 0
  SOGI flag: false, priority: 0
  SOGI flag: false, priority: 0
~~~
with the last line repeating ad infinitum.

#### Expected behavior:

_tables_ seems not to expect such data, so GIGO kinda applies, but should not get stuck.

#### Errors and logs:

#### Environment:
 - OS: Windows
 - OS version: 10, x64
 - TSDuck full version: 3.35-3218
 - Installation type: portable nightly build

#### Additional information:

lelegard, 2023-06-04:
Hi @AlfonZ42 

Thank you for this precise bug report. I just fixed it.

The bug occurred on garbage input only. Nonetheless, resisting to incorrect input is a general requirement and this was indeed a bug.

The update nightly build should be available tomorrow.

AlfonZ42, 2023-06-04:
I believe there is a typo:
```diff
--- a/src/libtsduck/dtv/descriptors/tsDVBServiceProminenceDescriptor.cpp
+++ b/src/libtsduck/dtv/descriptors/tsDVBServiceProminenceDescriptor.cpp
@@ -260,7 +260,7 @@ void ts::DVBServiceProminenceDescriptor::DisplayDescriptor(TablesDisplay& disp,
                     const bool country_code_flag = buf.getBool();
                     const uint8_t region_depth = buf.getBits<uint8_t>(2);
                     bool drawn = false;
-                    if (country_code_flag && buf.canReadBits(3)) {
+                    if (country_code_flag && buf.canReadBytes(3)) {
                         disp << margin << "Country: " << buf.getLanguageCode();
                         drawn = true;
                     }
```

lelegard, 2023-06-05:
You are right. This is fixed.

Concerning the XML model, only the structure matters. The content of attributes is only informational, not used in the code. Except in the automatic XML-to-JSON translation where the prefix `uint` (but not the size) is recognized to generate a JSON number instead of a string.

AlfonZ42, 2023-06-06:
I created [test.xml](https://github.com/tsduck/tsduck/files/11659010/test.zip), which (via `tspacketize | tstables`) could replace the above _test.ts_ in regards to the descriptor testing. 
In particular, using it with the old version 3218, several sub-tests exhibited the infite loop; with version 3220 they successfully passed.  
The xml does not replace the _test.ts_ in regards to `tables` mis-interpreting it as tables to be processed.

Using `tstabcomp`s to compile+decompile the xml, the result seemed to be correct - every valid input was recognized, some invalid inputs were partially recognized, some invalid inputs were rejected and remained as *generic_descriptor*s.

Note: There are three sub-tests that check *(existence of country_code or primary_region_code)* in xml, which I commented out, as they error out, as designed. 
I think it is possible to separate them into different inputs, so each of them gets tested separately. I'm not sure what would be the best solution here though.

AlfonZ42, 2023-06-06:
Using `tspacketize|tstables` and `tstabcomp --compile|--decompile`, with input

```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT network_id="0xFF02">
    <transport_stream transport_stream_id="0x0002" original_network_id="0xFF01">
      <generic_descriptor tag="0x7F">
        <!-- one item in SOGI_list, target_region_flag set, one item in region_loop, country_code_flag set, too few data (1/2) -->
        22 05 50 00 02 FC 41
      </generic_descriptor>
      <generic_descriptor tag="0x7F">
        <!-- one item in SOGI_list, target_region_flag set, one item in region_loop, country_code_flag set, too few data (2/2) -->
        22 06 50 00 03 FC 41 42
      </generic_descriptor>

      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0x000">
          <target_region country_code="GHI" secondary_region_code="0x56"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0x000">
          <target_region country_code="JKL" tertiary_region_code="0x57"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0x000">
          <target_region country_code="DEL" primary_region_code="0x55" tertiary_region_code="0x57"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0x000">
          <target_region country_code="GHL" secondary_region_code="0x56" tertiary_region_code="0x57"/>
        </sogi>
      </service_prominence_descriptor>
    </transport_stream>
  </NIT>
</tsduck>
```

version 3220 produced:
```
    - Descriptor 0: Extension Descriptor (0x7F, 127), 7 bytes
      Extended descriptor: Service Prominence (0x22, 34)
      SOGI flag: false, priority: 0
      Country: 
    - Descriptor 1: Extension Descriptor (0x7F, 127), 8 bytes
      Extended descriptor: Service Prominence (0x22, 34)
      SOGI flag: false, priority: 0
      Country: 
```
```xml
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0">
          <target_region country_code="GHI"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0">
          <target_region country_code="JKL"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0">
          <target_region country_code="DEL" primary_region_code="85"/>
        </sogi>
      </service_prominence_descriptor>
      <service_prominence_descriptor>
        <sogi SOGI_flag="false" SOGI_priority="0">
          <target_region country_code="GHL"/>
        </sogi>
      </service_prominence_descriptor>
```

whereas version 3221 produced:
```
    - Descriptor 0: Extension Descriptor (0x7F, 127), 7 bytes
      Extended descriptor: Service Prominence (0x22, 34)
      SOGI flag: false, priority: 0
      Reserved bits incorrectly set:
        Byte 8, bit #1 should be '1'
        Byte 8, bit #3 should be '1'
        Byte 8, bit #4 should be '1'
        Byte 8, bit #5 should be '1'
    - Descriptor 1: Extension Descriptor (0x7F, 127), 8 bytes
      Extended descriptor: Service Prominence (0x22, 34)
      SOGI flag: false, priority: 0
      Primary region: 66
      Reserved bits incorrectly set:
        Byte 8, bit #1 should be '1'
        Byte 8, bit #3 should be '1'
        Byte 8, bit #4 should be '1'
        Byte 8, bit #5 should be '1'
```
```xml
      <generic_descriptor tag="0x7F">
        22 07 50 00 04 FD 47 48 49
      </generic_descriptor>
      <generic_descriptor tag="0x7F">
        22 07 50 00 04 FD 4A 4B 4C
      </generic_descriptor>
      <generic_descriptor tag="0x7F">
        22 08 50 00 05 FE 44 45 4C 55
      </generic_descriptor>
      <generic_descriptor tag="0x7F">
        22 07 50 00 04 FE 47 48 4C
      </generic_descriptor>
```

The first two cases are a change in dealing with invalid input, so it might be acceptable.

The rest four cases look like there is a difference between what serialization produces vs. deserialization accepts.

lelegard, 2023-06-06:
I think that we should not spend too much time in this.

The input is invalid because `tstables` interprets as sections the content of a T2-MI PID (a very specific binary format which is neither PES nor sections). The important point is not to fail in the presence of invalid input: no crash, no infinite loop as there was here, etc. The output is whatever `tstables` can understand from that invalid input and it will never be correct.

Note that, in the general case, `tstables` can safely extract sections without telling it which PID to analyze because, in the vast majority of transport streams, a PID carries PES packets or sections, nothing else. And the PES format is carefully designed to be different from any possible section. So, `tstables` detects PES data, ignores them, and safely analyzes sections. Here, we have a PID carrying T2-MI data, a raw binary format which cannot be detected as non-section. So, `tstables` decodes possible sections, producing weird results (but not failing). With such rare transport streams, we must tell which PID to analyze: `tstables --pid ... --pid ...`.

Concerning your last point, the XML is incorrect because while primary, secondary and tertiary regions are optional, they can be specified in addition to the previous one only. Thus, this incorrect input produced an incorrect binary. I have changed this so that the consistency of the XML is now enforced:
~~~
$ tspacketize -p 100 test.xml -o test.ts
tspacketize: secondary_region_code cannot be used without primary_region_code in <target_region>, line 16
tspacketize: Error in descriptor <service_prominence_descriptor> at line 14
tspacketize: tertiary_region_code cannot be used without secondary_region_code in <target_region>, line 21
tspacketize: Error in descriptor <service_prominence_descriptor> at line 19
tspacketize: tertiary_region_code cannot be used without secondary_region_code in <target_region>, line 26
tspacketize: Error in descriptor <service_prominence_descriptor> at line 24
tspacketize: secondary_region_code cannot be used without primary_region_code in <target_region>, line 31
tspacketize: Error in descriptor <service_prominence_descriptor> at line 29
tspacketize: Error in table <NIT> at line 3
~~~

closed 2023-11-05 by github-actions[bot]