tsduck/tsduck #1228: [BUG] Incorrect (CR CR LF) newline sequence of characters under Windows in specific tables output

AlfonZ42, 2023-06-10:
#### Bug description:

Tests 012, 062, 065, 120 in the test suite fail in my Windows+Cygwin environment with the following type of difference:
```diff
8,9c8,9
<       Byte 10, bit #1 should be '1'
<       Byte 10, bit #2 should be '1'
---
>       Byte 10, bit #1 should be '1'
>       Byte 10, bit #2 should be '1'
```
There is no visible difference. The difference is in the presence of `CR` characters.

Newline under Linux is `LF`, under Windows `CR LF`. This difference is usually handled by `diff --strip-trailing-cr` (as is elsewhere in the output). 
However, in this specific case, the exact output is `CR CR LR`, which *diff* considers different from `LF` (and `CR LF`).

This extra `CR` occurs only in the sections listing incorrectly set reserved bits, which leads to `Buffer::ReservedBitsErrorString`, which adds environment-dependent newline (`UString::EOL`). 
I think the `LF` in this `CR LF` then later gains another `CR`.

#### How to reproduce:

Run `tspacketize --pid 200 test.xml | tstables > test.txt` under Windows, where *test.xml* is
```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT network_id="0xFF02">
    <transport_stream transport_stream_id="0x0002" original_network_id="0xFF01">
      <generic_descriptor tag="0x7F">
        22 04 00 00 00 00
      </generic_descriptor>
    </transport_stream>
  </NIT>
</tsduck>
```
and investigate line endings.
Alternatively, running one of tests 012, 062, 065, 120 under Windows and investigating line endings might work as well. 

These tests seem to pass in CI. 
I tried to install *MSYS2*, which utilizes MinGW, I believe; same as CI. 
There was a difference in versions of diff (package *diffutils* had version *3.9*, package *mingw-w64-diffutils* version *3.6*). 
MSYS' version *3.9* behaved the same as my Cygwin's version (also *3.9*). 
Version *3.6* seemed to perform CR-stripping even with no extra arguments, so with `--strip-trailing-cr`, it found no difference between `CR CR LF` and `LF`. 
I tried to install Git *v2.41.0*, its diff version was *3.9*, no extra CR-stripping. 
So I am not sure why the behaviour on the CI machine is different.

#### Expected behavior:

I think the extra `CR` should not be present.

#### Errors and logs:

#### Environment:
 - OS: Windows
 - OS version: 10, x64
 - TSDuck full version: 3.35-3222
 - Installation type: portable nightly build

#### Additional information:

lelegard, 2023-06-10:
Hi @AlfonZ42 

Thank you for digging into the gory details that way.

> This extra CR occurs only in the sections listing incorrectly set reserved bits, which leads to Buffer::ReservedBitsErrorString, which adds environment-dependent newline (UString::EOL). I think the LF in this CR LF then later gains another CR.

This is the usual dilemma in Windows. LF in the "normal world" (Linux, macOS, BSD, all Unix flavours and everywhere except Windows) is considered as CR/LF in Windows. But who is responsible from translating LF into CR/LF? If every software layer which is crossed by a text turns each LF into CR LF, without checking if there is already a preceding CR, we may get text lines ending in CR CR CR CR CR LF.

It is known that using `std::endl` in C++ generates a CR LF sequence on Windows. But what about string data containing `'\n'`? Does the Microsoft C++ runtime library inspect all string variables and translate all internal `'\n'`? Apparently yes. What about other C++ RTL? This is just a mess.

I changed the `EOL` you spotted (which has a CR/LF interpretation on Windows) with a plain `LINE_FEED`. Let us know if this fixes the problem.

> I tried running the mentioned tests under them and found a difference - sed -i (performed in test_text) did the CR-stripping as well, resulting in files without any trailing CRs, hiding the difference. Under Cygwin it leaves them intact.

Do you mean that `sed -i` behaves as `dos2unix` with MinGW and not with Cygwin? This is funny how open and portable tools end up with inconsistent behaviours on Windows.

All developers now use git. On Windows, they naturally use the so-called "git bash" tool which is a MinGW environment. So, Cygwin is not fully explored by most developers. Furthermore, we always make sure to use only the "git bash" environment and never Cygwin. This is useful to make sure we do not use,  in our scripts, common Unix tools which are present in Cygwin but not in the predefined "git bash" environment. If I used Cygwin and used such tools, the CI/CD would fail on GitHub.

> I think the extra CR should not be present.

I think that no CR should be present anywhere in Windows. At all. Nowhere.
Sometimes, it is tempting to think that Windows itself should not be present...

AlfonZ42, 2023-06-10:
> Do you mean that `sed -i` behaves as `dos2unix` with MinGW and not with Cygwin? This is funny how open and portable tools end up with inconsistent behaviours on Windows.

Yes, running `sed --debug "" test.txt` with MinGW:
```
INPUT:   'test.txt' line 12
PATTERN:         Byte 4, bit #4 should be '1'\r
END-OF-CYCLE:
        Byte 4, bit #4 should be '1'
INPUT:   'test.txt' line 13
PATTERN:         Byte 6, bit #4 should be '1'
END-OF-CYCLE:
        Byte 6, bit #4 should be '1'
```
vs. Cygwin:
```
INPUT:   'test.txt' line 12
PATTERN:         Byte 4, bit #4 should be '1'\r\r
END-OF-CYCLE:
        Byte 4, bit #4 should be '1'

INPUT:   'test.txt' line 13
PATTERN:         Byte 6, bit #4 should be '1'\r
END-OF-CYCLE:
        Byte 6, bit #4 should be '1'
```

The difference happens on reading the file.

This thread looks relevant: https://cygwin.com/pipermail/cygwin/2017-June/thread.html#233121 . 
It seems to me Cygwin's goal is to behave as much as Linux (leaving *CR*s as they are, unless told otherwise), whereas MinGW tries to "hide" it is actually running on Windows.

AlfonZ42, 2023-06-11:
> I changed the `EOL` you spotted (which has a CR/LF interpretation on Windows) with a plain `LINE_FEED`. Let us know if this fixes the problem.
Yes, version 3223 fixed it.

closed 2023-06-12 by AlfonZ42