tsduck/tsduck #1229: [BUG] Debian package declares incorrect libsrt version dependency

ajkq, 2023-06-10:
#### Bug description:
The TSDuck deb package compiled on Debian 11 and 12 declares a dependency on either libsrt1.4-openssl or libsrt1.5-openssl.
However, on Debian 11 only libsrt1.4-openssl is available.
On Debian 12, only libsrt1.5-openssl is available.

This causes TSDuck 3.34-3197.debian11 that was originally installed on Debian 11 to continue to function after an upgrade to Debian 12, since the library remains installed.
Once the TSDuck package is upgraded to 3.34-3197.debian12 with dependencies satisfied by executing `apt install --fix-broken`, tsp fails to execute with the below error message. Since the package declares a dependency on either libsrt1.4-openssl or libsrt1.5-openssl, the existing libsrt1.4-openssl package from before the upgrade causes dpkg to not complain about the missing package the binary is linked against.
`tsp: error while loading shared libraries: libsrt.so.1.5: cannot open shared object file: No such file or directory`

#### How to reproduce:
1. Install Debian 11 with 'Standard System utilities' and 'SSH server' selections
2. ```
   apt install curl
   curl -LJO https://github.com/tsduck/tsduck/releases/download/v3.34-3197/tsduck_3.34-3197.debian11_amd64.deb
   dpkg -i tsduck_3.34-3197.debian11_amd64.deb
   ```
   dpkg shows dependency problems
3. `apt install --fix-broken` installs libpcsclite1, libsrt1.4-openssl
4. Observe `tsp --version` working correctly
5. Upgrade to Debian 12
6. `dpkg -i tsduck_3.34-3197.debian12_amd64.deb` (See below how package was built)
7. dpkg shows a dependency error on librist
8. `apt install --fix-broken` installs libcjson1, libmbedcrypto7, librist4, but not libsrt1.5-openssl
9. Observe `tsp --version` failing to execute due to missing libsrt.so.1.5

#### Expected behavior:
Debian 11 package should only declare a dependency on libsrt1.4-openssl, Debian 12 on libsrt1.5-openssl. This would cause dpkg to show an error message about unmet dependencies, and the command `apt install --fix-broken` could be used to automatically install missing dependencies.

Packages compiled on either OS version list the same dependencies.
```
dpkg -I tsduck_3.34-3197.debian11_amd64.deb  | grep libsrt
  libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl | libsrt1.5-openssl,

dpkg -I tsduck_3.34-3197.debian12_amd64.deb  | grep libsrt
  libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl | libsrt1.5-openssl,
```

The binary depends on library version 1.4 or 1.5, depending on the OS version the package was compiled on. Output is from a Debian 12 system, where libsrt1.4-openssl was removed:
```
dpkg -l | grep tsduck; ldd /usr/bin/tsp | grep libsrt
ii  tsduck                                           3.34-3197.debian11                        amd64        MPEG Transport Stream Toolkit
	libsrt.so.1.4 => not found

dpkg -l | grep tsduck; ldd /usr/bin/tsp | grep libsrt
ii  tsduck                                           3.34-3197.debian12                        amd64        MPEG Transport Stream Toolkit
	libsrt.so.1.5 => /lib/x86_64-linux-gnu/libsrt.so.1.5 (0x00007fdcb2a73000)
```

#### Environment:
 - OS: Debian
 - OS version: 12
 - TSDuck full version: 3.34-3197
 - Installation type: Deb package built via Dockerfile with the following commands:
```
apt-get --no-install-recommends install -y curl ca-certificates lsb-release sudo locales
echo "en_US.UTF-8 UTF-8" >>/etc/locale.gen && locale-gen
./scripts/install-prerequisites.sh
./scripts/dtapi-config.sh --download
make deb
```

lelegard, 2023-06-11:
Hi @ajkq 

Thanks for reporting this. 

This is an incorrect handling of a few dependencies. On Ubuntu and Debian (and unlike Fedora and Red Hat), these packages change their names as well as their version. It is a bit complicated to find the right package name, automatically for all OS versions. Giving a list of "any of the known alternatives" is finally not correct as you demonstrated.

I have fixed this. Now, the dependencies of the `.deb` packages are the exact package names which were used during the build.

Please let us know if this fixes the problem for you.

ajkq, 2023-06-11:
Yes, this fixes the problem.
Thanks!

closed 2023-06-11 by ajkq

impleotv, 2023-06-14:
Hi guys,

I tried to run the latest version on Ubuntu 22.04 and got this error:
```
tsp: error while loading shared libraries: libsrt.so.1.5: cannot open shared object file: No such file or directory
```
Thanks,

Alex

lelegard, 2023-06-14:
> I tried to run the latest version on Ubuntu 22.04 and got this error:

Which package did you try to install? If this is the latest one, named `tsduck_3.34-3197.ubuntu23_amd64.deb`, you may notice "ubuntu23" in the name. If you install it on Ubuntu 22, there is a mismatch in the binaries.

TSDuck works well on Ubuntu 22 but you have to rebuild it yourself. I cannot produce binaries for all versions of all distros. I just build a packages for the latest version of a few common distros.

If you never rebuilt TSDuck, this is trivial:
~~~
got clone --depth 1 https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make installer
~~~

If you already have setup the build environment for TSDuck, just use `make installer`.

The `.deb` packages are dropped in the `installers` directory [More details here](https://tsduck.io/doxy/building.html).

impleotv, 2023-06-14:
Thanks a lot! 

Alex