tsduck/tsduck #1230: [Q] How to insert a binary buffer as a ts packet?

impleotv, 2023-06-12:
Hi all,

Is it possibly to insert a data buffer coming from an external udp source (ideally, splitting it, if it is > 188 bytes) as a data pid?
I tried the following command line and it re-sends the ts stream, properly adding the tables with klv descriptor. 

```
tsp -I ip 227.1.1.2:30122  -P encap -p 0x102 --output-pid 7777 --pes-mode fixed -P pmt --add-pid 7777/0x15 --add-pid-registration 7777/0x4B4C5641 -O ip 227.1.1.1:30120
```

Is it possibly to add/merge the actuall klv data (sent as a buffer by the external application)?

Thanks,

Alex

lelegard, 2023-06-12:
Hi @impleotv 

>  insert a data buffer coming from an external udp source (ideally, splitting it, if it is > 188 bytes) as a data pid

What kind of data do you receive? Is this already a TS, a suite of 188-byte packets with proper TS header? Or other generic data? In the latter case, you need to encapsulate the data into some TS-compatible format such as MPE.

The plugin `encap` you used encapsulates a few PID's from the TS, ie. TS packets, not generic data, into another "tunnel" PID. This is typically a way to cross TS-processing tools, such as ffmpeg, which may damage or remove PID's they don't understand.

impleotv, 2023-06-12:
Hi Thierry,

Thanks for your prompt reply. The idea was to use tsduck to create/process STANAG 4609 streams. Since your tool can insert tables for both ASYNC and SYNC Klv metadata, it already supports STANAG. I wanted to find a way to use it to add KLV packets to a ts stream that doesn't initially have KLV metadata. These packets are just binary buffers (which may be longer than 188 bytes) - the payload, not the ts packets. This is how HW vendors send them - RAW KLV. I was looking for a way to receive such data over UDP, packetize it, and add it to the ts stream. This method would also work for adding any type of private data in general.

So, based on your reply, I understand that if I produce a ts stream with these packets (with a single data PID, without any video), I would be able to use the tool (I have zero experience with it, but it seems like it is able to handle all possible scenarious)...

Thanks,

Alex

impleotv, 2023-06-13:
Hi Thierry,

There is some info about these streams in your User's manual (at the bottom of the page 242) [https://impleotv.com/2017/02/17/klv-encoded-metadata-in-stanag-4609-streams/](url)  which actually leads to our website :-)

I beleive it should work, just haven't fully figured out the correct syntax... Instead of using the **encap** plugin, I should have used the **merge** plugin.

if I simply rebroadcast the ts stream, modifying the pmt, I receive all the correct tables I need.

```
tsp -I ip 227.1.1.2:30122 -P pmt --add-pid 65/0x15 --add-pid-registration 65/0x4B4C5641 -O ip 227.1.1.1:30120
```
Now, I just have to send the klv data as a separate ts stream (using another application), with only one data pid that contains the klv, and merge that data pid.

So, something like this should work: I rebroadcast the ts stream, modify the pmt and add the klv pid using **merge** plugin:
```
tsp -I ip 227.1.1.2:30122 -P pmt --add-pid 65/0x15 --add-pid-registration 65/0x4B4C5641 -P merge -p 65 "tsp -I ip 227.1.1.4:12345" -O ip 227.1.1.1:30120
```

Not sure how to do it properly, as it doesn't seem to work. The klv data pid in the external stream is 65.

Thanks,

Alex