tsduck/tsduck #1235: Eit workflow[Q]

tenkavideo, 2023-06-21:
hello !
I need to inject eit with this worflow : present and follow events in td 78 P/F must not be present in the td80 schedule
i tries with a custom tsduck xml with p/f and scdule nodes but using -P inject .... work but the td78 have the present and follow events in the same section(0) section 1 is not gerated
this the p/f part of the input xml:
 <?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT type="pf" version="0" current="true" actual="true" service_id="0x264A" transport_stream_id="0x98E1" original_network_id="0x217C" last_table_id="0x4E">
    <metadata PID="18"/>
    <event event_id="0x001A" start_time="2023-06-21 15:54:01" duration="01:27:10" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>yyyyyyy>
        <text>xxxxx  xxxxxx. </text>
      </short_event_descriptor>
    </event>
    <event event_id="0x001B" start_time="2023-06-21 17:21:11" duration="00:08:38" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>zzzzzzzz>
        <text>xxxxxxxxxx </text>
      </short_event_descriptor>
    </event>
  </EIT> 

with using of -P eitinject both p/f section are generated but there are no way to remove current p/f events from schdule
this my input xml :
 <EIT type="pf" version="1" actual="true" service_id="9802" transport_stream_id="39137" original_network_id="8572">
    <event event_id="29" start_time="2023-06-20 18:59:06" duration="00:32:58" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>aaaaaaaaa</event_name>
        <text>Iaasasa </text>
      </short_event_descriptor>
    </event>
    <event event_id="30" start_time="2023-06-20 19:32:04" duration="00:00:25" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>asasas</event_name>
        <text />
      </short_event_descriptor>
    </event>
  </EIT>
  <EIT type="0" actual="true" service_id="9802" transport_stream_id="39137" original_network_id="8572">
    <event event_id="31" start_time="2023-06-19 22:02:57" duration="00:25:55" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>zzzzzzz</event_name>
        <text>dsdsdsdsd. </text>
      </short_event_descriptor>
    </event>
    <event event_id="32" start_time="2023-06-19 22:28:52" duration="00:00:25" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="nor">
        <event_name>sdsdsdsds</event_name>
        <text />
      </short_event_descriptor>
    </event>

any suggestion? thx so much

lelegard, 2023-06-21:
Hi @tenkavideo 

First, the plugin `inject` is not suitable for EIT injection. The plugin `inject` repeatedly injects the same tables while EIT's are dynamic sections which are updated when time passes (ETSI TS 101 211). You shall use `eitinject`.

With the plugin `eitinject`, the structure of the input XML tables are ignored, only the events are extracted. The EIT sections are generated on the fly, with the appropriate events, according to the current time.

When you generate EIT p/f and schedule, the two are independently generated. They are only mixed in the output PID. An event is removed from EIT schedule sections when the event is completed or when its 3-hour "segment" terminates (option `--lazy-schedule-update`).

I see no identified use case for removing an event from EIT schedule when the event is current (ie. when the event is also in an EITp/f). Typically, the two types of EIT's are processed in two different applications in the receiver: the EPG for EIT schedule, the info bar (or whatever name is used) for EIT p/f.

Most operators generate different contents for the same event in EIT p/f (short description) and EIT schedule (complete description). In such a context, if you stop cycling EIT schedule for current events, their description in the EPG will become less accurate. 

Note that this is not possible with the plugin `eitinject`, the content of an event is the same in both types of EIT.

If I understand correctly, you would like to remove an event from the EIT schedule when it appears in the EIT p/f. Is that right? This is not possible with `eitinject` (and I see no valid use case beyond over-engineered specifications). An EPG application can even receive some events from the past and may ignore them. This is the case when you choose to renew EIT sections segment per segment instead of event per event, to avoid stressing the EIT section demux.

tenkavideo, 2023-06-21:
thx so much , is a very professional answer nad u get fully my question
let me ask one thing in last, i have an application that manage the  eit tsduck xml  ,so things like add  - remove events from p/f and schedule etc are automaticly done 
the only things i cannot do with the inject plugin is to genate p/f with section 0 an section 1, always only section 0 is created with the present and follow event, or i can generate 2 section both 0, can u help me on this? if not possible u can close and..........thx so much for your quicky answer !

lelegard, 2023-06-22:
> the only things i cannot do with the inject plugin is to genate p/f with section 0 an section 1, always only section 0 is created with the present and follow event inside,

This is not the only problem. The section layout of your EIT schedule cannot be compliant with ETSI TS 101 211 because `inject` can only inject complete tables and not sparse sections.

You shall not use `inject` to inject EIT's. Forget about it. You must use `eitinject`. If you do not like the idea of having the current events in EIT p/f and schedule at the same (which is a normal configuration), the result of using `inject` will be worse anyway.

And if you _really_ want to implement the non-standard use case of removing events from EIT schedule when they become present or following, you may modify the `eitinject` plugin for your own usage. It's open source ! 😄 

But the first thing to do is to carefully think _why_ you want to do this. Sometimes, people insist in implementing a specific way of working, forgetting that the actual target is more important and can be reached by a different, safer, path.

tenkavideo, 2023-06-22:
Again thx for you time and compentence , sure i ll take a look on eitinject code , and pls close this thread because i don t want create confusion to other users, this a custom workflow 
In last , how to donate ? 
thx again L

lelegard, 2023-06-22:
>  i ll take a look on eitinject code

https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/epg/tsEITGenerator.h
https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/epg/tsEITGenerator.cpp

> In last , how to donate ?

https://tsduck.io/donate/