tsduck/tsduck #1236: [FEAT] Enhancement to decap tool to support plain encapsulation including sync byte (Abertis)

mrwish7, 2023-06-22:
Hi there,

First of all, thank you for an excellent set of tools!

I have a use case for which I'm wondering whether an addition to the decap tool would be feasible and straightforward.

The Spanish digital terrestrial operator uses encapsulated feeds via satellite (30°W) to feed their DVB-T transmitters. They encapsulate each full DVB-T MPEG-TS into a single PID inside the DVB-S2 transponder MPEG-TS. This is very similar to the method used by the tsduck encap tool, except that the 0x47 sync byte is not removed - the full TS is encapsulated without any modification whatsover. This system tends to be informally known online as "Abertis", after the original name of the provider of these feeds.

This is quite simple to decapsulate as we just have to strip the packet headers/AF from the outer PID and then we have a playable 'inner' MPEG-TS. The first output packet being a fragment (as the packet starts obviously don't necessarily line up) doesn't seem to cause any problem in practice. The PUSI bit being set or not set doesn't seem to always correspond with there being the start of a new packet inside the outer packet or not, and there's no pointer to where it starts. The data is just raw.

Would it be feasible to update the decap tool to handle this system? Seeing as the system is at least fairly similar at a high level to the existing encap system. I guess there may be more work required though if we then need to handle each output packet separately rather than just dumping the raw stream and worrying about it later in any process. Maybe we could sync to 0x47 at the start, and then we wouldn't need to worry after that as the data has no other unexpected things in it once the header/AF is stripped?

Another solution could be another separate tool for this purpose, I know there's precedent for this with a dedicated t2mi tool for example (the Abertis system is sometimes seen as a kind of precursor to T2-MI).

For extra context, here's a simple tool to strip the Abertis encapsulation written in C and kindly shared on a satellite forum - https://sat-universe.com/index.php?threads/spanish-abertis-dtt-sat-hispasat-30w-chat-how-to-guides.306087/page-73#post-2037173416

I only have experience of using tsduck and I'd need to do some learning to contribute to the code, but I'm happy to help with this if it is a feature worth considering!

Thanks

lelegard, 2023-06-22:
Hi @mrwish7 

Interesting case. With a few remarks:

First, we need reference materials. Please provide a capture of the complete satellite stream which contains the encapsulated TS. A couple of minutes would be good, probably a couple of GB.

Do you have any reference documentation or specification on this encapsulation? I guess no but we need to ask. Any research article would be good too.

A potential decapsulation plugin would be more like the `t2mi` plugin, not the `decap` plugin. The `encap` and  `decap` plugins are used to encapsulate a part of the TS, a few PID's from the TS inside one single tunnel PID (the typical use case is crossing an ffmpeg command while preserving all data PID's which would be otherwise destroyed by ffmpeg). So, the `decap` plugin replaces the content of the tunnel PID with the decapsulated PID's, merging them with the rest of the TS. The `t2mi` plugin, on the other hand, extracts a complete and distinct TS from the tunnel PID. This extracted TS (the inner TS) completely replaces the outer TS which contained the tunnel PID.

The provided sample code is rather rudimentary. It blindly extracts the contents of the TS payloads without managing any form of synchronization. In the general case, the result is an invalid TS since it potentially starts in the middle of a packet. We need to do better and restore the packet synchronization. They must have some synchronization mechanism in their encapsulation format. If we do not have a specification, we need at least a large sample TS to look for this synchronization mechanism.

mrwish7, 2023-06-22:
Hi @lelegard 

Thank you for getting back to me. That makes complete sense for it to be like the t2mi plugin, as that's the exact functionality that would be required (the extracted distinct full TS replacing the outer TS).

Here is a sample recording of 2 minutes from one of the DVB-S2 transponders that use the system (30°W 12548.5V) -

https://mega.nz/file/IIQiUZZC#umsYGW2HBhl6v6id4Vq5hKLMl_Ltw4yQ5WmlAcPOrMo

This contains three PIDs that each contain a tunnelled full TS - PIDs 701, 702 and 703. Note that I have already processed the TS with tsduck in the following way, as the three PIDs are encrypted with BISS -

`tsp -I file 12548.5v.ts -P descrambler -c [CW REMOVED] -p 701-703 -O file 12548.5v_dec.ts`

With a compiled version of the tool linked above called `abertis` , it's then possible to do the following -

`cat 12548.5v_dec.ts | abertis 702 > 702.ts`

The resulting 702.ts file then plays successfully, although as you rightly say - it's certain to nearly always include a partial TS packet fragment at the start which isn't neat. Hopefully the provided sample will include enough to work out the synchronisation method used by the broadcaster.

[image]

Unfortunately I don't have any other documentation, what is known is largely what's been found out through experimentation by satellite enthusiasts.

Thanks again,
Adam

lelegard, 2023-06-24:
Hi @mrwish7 

I had a quick look at the TS. Technically, this is crap.
- They do not even know how to build a CA_descriptor in a CAT: the reserved bits are 0 instead of 1.
- Ironically, the reserved bits in the CA_descriptors in the PMT are correct. But the ECM PID is always the null PID. A CA system using no ECM (and consequently fixed keys) does not deserve any respect.
- They place the service_descriptor in the PMT while it should be in the SDT. And there is not SDT.
- They use private descriptors in the PMT without prior private data specifier descriptor.
- Ironically (again), they correctly use a private data specifier descriptor in the CAT. But they uselessly repeat it.

I know that this TS is probably a private feed, not for reception on generic receivers, but there is no reason to produce such a crap.

I will have a look.

jof-cw, 2023-06-25:
Hello all.

That feed from Cellnex Telecom (in the past Abertis) is used for transport the TS of national public and private channels in Spain. Cellnex is the most important broadcaster in Spain and they are member of DVB and other associations. Maybe they are done this signalling on purpose to avoid pirate transmitters; their satellite receivers are customized.

Best regards.
Jesus

lelegard, 2023-06-25:
> Maybe they are done this signalling on purpose to avoid pirate transmitters; their satellite receivers are customized.

I don't believe it a single minute. 25 years in CAS and DRM systems have taught me a few lessons:
- If you want to implement security, do it properly. Using a fixed key is breaking the very first rule of security.
- If you want to implement obfuscation, there is room for a lot of imagination. Placing a service descriptor in the wrong table does not belong to this.
- When you are astonished by something weird, if you hesitate to classify the cause as malice or incompetence, the latter wins the vast majority of times.

lelegard, 2023-06-25:
Plugin `feed` added. Please try it. RTFM for more details.

lelegard, 2023-06-26:
> If I use the IP output plugin instead (to stream out UDP) it seems to struggle more, with the re-synchronisation messages appearing frequently -

Probably because you miss some UDP datagrams. With `ip` input, the input plugin thread is stressed every UDP datagram. With `dvb` input, it fills large buffers. Your system is probably too busy as you noticed. You may try `-I ip --buffer-size <something-larger>`.

> ... for around 10 years now ... and nothing has changed in that time (CW etc)

Same CW in 10 _years_ seriously ? 😆 
Standard operation rules have used 10 _seconds_ from the beginning, they just missed the unit...