tsduck/tsduck #1237: How do I analyze PID packets for SCTE-35 Splice info?

RGCrutwell, 2023-06-22:
Hi TSDuck Team, 

I'm trying to use TSDuck to analyze SCTE-35 packets to ensure they have been mapped correctly from SCTE-104 inserter for an automation system I'm developing. 

Is there a way to analyze the SCTE-35 packets in the PID associated?

Many thanks, 
Richard

RGCrutwell, 2023-06-22:
I may be able to do this with tsdump using the specified PID in fact looking in the UG

lelegard, 2023-06-23:
Hi @RGCrutwell 

What do you mean by "analyze the SCTE-35 packets"?

If you mean get an hexadecimal dump of the TS packets, yes `tsdump --pid ...` will do the job.

If you would like an analysis of the splice information sections, then use `tstables`. You can get the splice information sections in various formats: text analysis, binary, XML, JSON.

RTFM for more details.

closed 2023-11-26 by github-actions[bot]