tsduck/tsduck #1239: [Q]  Line breaks  in Eit

Holger69m, 2023-06-26:
Hello, in Germany we have several line breaks in the EPG EIT Data.
Example

Navy CIS
First Steps
Crime, USA 2022
Age rating: from 12

Sean Murray (Timothy McGee)
Wilmer Valderrama (Nick Torres)

.....

Here is the result when I used 
tstabcomp -d eit.bin

<text>First StepsCrime, USA 2022Age rating: from 12</text>
<text>Sean Murray(Timothy McGee)Wilmer Valderrama (Nick Torres)</text>

The line breaks are still contained in the .bin file
As soon as I create the XML file, these are missing. Is there a possibility to use this e.g. previously by |#| to replace. Then I could later restore the line breaks in the mysql database.

Then it would look like this
<text>First Steps|#|Crime, USA 2022|#|Age rating: from 12</text>
<text>Sean Murray(Timothy McGee)|#|Wilmer Valderrama (Nick Torres)</text>

so it is easier to evaluate the information

the program works great, this change would make it perfect :)
thanks and greetings from Germany

lelegard, 2023-06-26:
Hi @Holger69m 

Could you please provide a binary sample? Part of the EIT PID with sections like this would be perfect.

It is always better to fix issues based on real-world data. It is also better to add real-world data in the non-regression test suite.

Holger69m, 2023-06-26:
[eit-dump.zip](https://github.com/tsduck/tsduck/files/11867838/eit-dump.zip)

Here again the text created with a Windows EPG tool, but I will automate it with Linux so that the data is automatically updated every night ;)

` 	<title lang="DEU">Navy CIS</title>
	<sub-title lang="DEU">Der erste Fall
Crime-Serie, USA 2022
Altersfreigabe: ab 12</sub-title>
	<desc lang="DEU">Leon Vances Tochter Kayla möchte nach ihrer Ausbildung in Parkers Team voll durchstarten und kann es kaum abwarten, bei ihrem ersten Außeneinsatz praktische Erfahrungen zu sammeln. Als sie den Agenten bei den Ermittlungen rund um den mysteriösen Tod des Chirurgen Dr. Masuda über die Schultern blickt und in Eigenregie riskante Entscheidungen trifft, erlebt sie schon bald die Schattenseiten ihres gefährlichen Jobs ...

Regie: Michael Zinberg
Drehbuch: Yasemin Yilmaz
Komponist: Brian Kirk
Kamera: William Webb
Schnitt: Mitchel Stanley

Darsteller:
Sean Murray (Timothy McGee)
Wilmer Valderrama (Nick Torres)
Katrina Law (Jessica Knight)
Brian Dietzen (Jimmy Palmer)
Diona Reasonover (Kasie Hines)
Rocky Carroll (Leon Vance)
Gary Cole (Alden Parker)
Naomi Grace (Kayla Vance)
Jason Kravits (Gordon Roth)</desc>
	<language>DEU</language>
</programme>`

this record should be present in the attached file
I tried to manipulate the .bin with sed, that was also successful, but tstabcomp aborts with an error message

> EIT2.bin: invalid section

Holger69m, 2023-06-26:
sorry wrong button :D

AlfonZ42, 2023-06-26:
I think the issue is due to coding of newline as `0x0A` (specification says it should be `0x8A`, I believe).

Editing the file with
```sh
xxd EIT_cut.bin EIT_cut.hex
sed -e 's| 0a\([0-9a-f][0-9a-f]\) | 8a\1 |g' -e 's| \([0-9a-f][0-9a-f]\)0a | \18a |g' EIT_cut.hex > EIT_cut-edit.hex
xxd -r EIT_cut-edit.hex EIT_cut-edit.bin
tstabdump --ignore-crc EIT_cut-edit.bin > EIT_cut-edit.txt
```

shows output like:
```
  - Event Id: 7106 (0x1BC2)
    Start UTC: 2023/06/27 19:14:30
    Duration: 01:00:07
    Running status: undefined
    CA mode: free
    - Descriptor 0: Short Event (0x4D, 77), 73 bytes
      Language: DEU
      Event name: "Navy CIS"
      Description: "Der erste Fall
Crime-Serie, USA 2022
Altersfreigabe: ab 12"
    - Descriptor 1: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 0, last: 3
      Language: DEU
      Text: "Leon Vances Tochter Kayla möchte nach ihrer Ausbildung in Parkers Team voll durchstarten und kann es kaum abwarten, bei ihrem ersten Außeneinsatz praktische Erfahrungen zu sammeln. Als sie den Agenten bei den Ermittlungen rund um den mysteriösen T"
    - Descriptor 2: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 1, last: 3
      Language: DEU
      Text: "od des Chirurgen Dr. Masuda über die Schultern blickt und in Eigenregie riskante Entscheidungen trifft, erlebt sie schon bald die Schattenseiten ihres gefährlichen Jobs ...

Regie: Michael Zinberg
Drehbuch: Yasemin Yilmaz
Komponist: Brian Kirk
Kam"
    - Descriptor 3: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 2, last: 3
      Language: DEU
      Text: "era: William Webb
Schnitt: Mitchel Stanley

Darsteller:
Sean Murray (Timothy McGee)
Wilmer Valderrama (Nick Torres)
Katrina Law (Jessica Knight)
Brian Dietzen (Jimmy Palmer)
Diona Reasonover (Kasie Hines)
Rocky Carroll (Leon Vance)
Gary Cole (Alde"
    - Descriptor 4: Extended Event (0x4E, 78), 70 bytes
      Descriptor number: 3, last: 3
      Language: DEU
      Text: "n Parker)
Naomi Grace (Kayla Vance)
Jason Kravits (Gordon Roth)"
    - Descriptor 5: Component (0x50, 80), 12 bytes
      Content/type: 0x0103 (MPEG-2 video, 16:9 aspect ratio without pan vectors, 25 Hz)
      Component tag: 2 (0x02)
      Language: DEU
      Description: "VIDEO"
    - Descriptor 6: Component (0x50, 80), 14 bytes
      Content/type: 0x0205 (MPEG-1 Layer 2 audio, surround sound)
      Component tag: 7 (0x07)
      Language: DEU
      Description: "deutsch"
```

compared to the nonedited one:
```
  - Event Id: 7106 (0x1BC2)
    Start UTC: 2023/06/27 19:14:30
    Duration: 01:00:07
    Running status: undefined
    CA mode: free
    - Descriptor 0: Short Event (0x4D, 77), 73 bytes
      Language: DEU
      Event name: "Navy CIS"
      Description: "Der erste FallCrime-Serie, USA 2022Altersfreigabe: ab 12"
    - Descriptor 1: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 0, last: 3
      Language: DEU
      Text: "Leon Vances Tochter Kayla möchte nach ihrer Ausbildung in Parkers Team voll durchstarten und kann es kaum abwarten, bei ihrem ersten Außeneinsatz praktische Erfahrungen zu sammeln. Als sie den Agenten bei den Ermittlungen rund um den mysteriösen T"
    - Descriptor 2: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 1, last: 3
      Language: DEU
      Text: "od des Chirurgen Dr. Masuda über die Schultern blickt und in Eigenregie riskante Entscheidungen trifft, erlebt sie schon bald die Schattenseiten ihres gefährlichen Jobs ...Regie: Michael ZinbergDrehbuch: Yasemin YilmazKomponist: Brian KirkKam"
    - Descriptor 3: Extended Event (0x4E, 78), 254 bytes
      Descriptor number: 2, last: 3
      Language: DEU
      Text: "era: William WebbSchnitt: Mitchel StanleyDarsteller:Sean Murray (Timothy McGee)Wilmer Valderrama (Nick Torres)Katrina Law (Jessica Knight)Brian Dietzen (Jimmy Palmer)Diona Reasonover (Kasie Hines)Rocky Carroll (Leon Vance)Gary Cole (Alde"
    - Descriptor 4: Extended Event (0x4E, 78), 70 bytes
      Descriptor number: 3, last: 3
      Language: DEU
      Text: "n Parker)Naomi Grace (Kayla Vance)Jason Kravits (Gordon Roth)"
    - Descriptor 5: Component (0x50, 80), 12 bytes
      Content/type: 0x0103 (MPEG-2 video, 16:9 aspect ratio without pan vectors, 25 Hz)
      Component tag: 2 (0x02)
      Language: DEU
      Description: "VIDEO"
    - Descriptor 6: Component (0x50, 80), 14 bytes
      Content/type: 0x0205 (MPEG-1 Layer 2 audio, surround sound)
      Component tag: 7 (0x07)
      Language: DEU
      Description: "deutsch"
```

I wasn't able to test the behaviour of tstabcomp, as I didn't manage to get it to ignore the now-incorrect crc.

Holger69m, 2023-06-26:
oh I see they edited the post.
Version 2 is perfect, very nice, exactly what I tried this morning for 4 hours without success :D

AlfonZ42, 2023-06-26:
Using other replacement character in sed is possible, e.g. `sed -e 's| 0A | 23 |g' EIT_cut.xml > EIT_cut-edit.xml`, which outputs:
```xml
<text>Der erste Fall#Crime-Serie, USA 2022#Altersfreigabe: ab 12</text>
```
if it's easier to process for you later.

Holger69m, 2023-06-26:
> Die Verwendung anderer Ersatzzeichen in sed ist möglich, z. B. `sed -e 's| 0A | 23 |g' EIT_cut.xml > EIT_cut-edit.xml`, was Folgendes ausgibt:
> 
> ```
> <text>Der erste Fall#Crime-Serie, USA 2022#Altersfreigabe: ab 12</text>
> ```
> 
> wenn es für Sie später einfacher zu verarbeiten ist.
> 
> Die Bearbeitung ist jedoch grob, z. B. wird nicht überprüft, ob nur der gewünschte Text eingegeben wurde, das letzte Zeichen der Zeile wird ebenfalls ignoriert.

thanks, but the problem is that this line break {0A} is swallowed by tstabcomp when creating the XML file ;) it exists in the binary file, but no longer in the XML file.

lelegard, 2023-06-26:
Hi @Holger69m,

From your binary file, TSDuck works as expected. This is a bug in your EPG tool, the tool which generates the binary EIT sections.

Let me explain why, based on ETSI EN 300 468.

In short, as mentioned by @AlfonZ42, in DVB strings (all descriptors, not only in EIT), the end-of-line sequence shall be encoded as 0x8A while your EIT sections contain 0x0A for what is apparently expected to be an end-of-line.

All (or most) strings in your EIT sections start with the special code 0x05. According to ETSI EN 300 468, table A.3, this means that the strings are encoded in ISO 8859-9, or Latin-5, where German characters are available. This encoding is represented in table A.6. In this table, we see that byte ranges 00-1F and 7F-9F are greyed. These codes are invalid and must not be present in such strings.

As an exception, in table A.1, we see the "single byte control codes" which are available in all single byte encodings. We see that the range 80-9F is allowed, with specific semantics, including 0x8A for end-of-line.

The remaining codes, 00-1F and 7F are invalid and forbidden. Your 0A bytes are consequently invalid.

> this line break {0A} is swallowed by tstabcomp

This 0A is _not a line break_. This is an invalid character.

When TSDuck decodes a DVB string, it converts it to UTF-8 for XML files or textual display. Invalid byte encodings (including 0A) are ignored because there is no valid way to convert them.

If you think about it, this is completely rational. Byte values 01-1F are used as character table selector. Let's assume you want to encode a string in ISO 6937+ (the default encoding). Let's assume that your text starts with a new line (why not?). Let's assume that you use 0A to encode it as in your EIT's. Then, the binary sequence for that string would start with 0A, right? Then, according to ETSI EN 300 468, a conformant receiver would interpret this 0A as a character table selector and would interpret the rest of the string as ISO 8859-14, or Latin-8 (Celtic). This is not what we want. You can see here one of the reasons why using 0A as a line break in DVB strings is a non-sense.

I suggest to submit a bug report to the provider of your EPG tool. You may provide a link to this issue as explanation.

Holger69m, 2023-06-26:
Yes, I knew that the EPG data was not compliant, but when reading out the data, these line breaks are very advantageous because they cut off the important information perfectly. Without the line break, it's hard to tell where information begins and where it ends.

with the small detour it works perfectly now

lelegard, 2023-06-26:
> these line breaks are very advantageous because they cut off the important information perfectly. 

Agreed, the problem is not the presence of line breaks. The problem is the binary encoding of those line breaks.

AlfonZ42, 2023-06-26:
FWIW, in our local broadcast, I've seen the exact issue mentioned.

A local TV station seems to provide EPG data using UTF-8 (or similar; containing newlines as 0x0A), because the data is used in various different places. 
An example input string of `0x0A <very long description>` (yes, newline at the start of text) is then split into `0x0A <partial description, to fit 255 max length>`, `0x0A <the rest of description>`. 
The splitting is correct as well, it retains the encoding of the string. It's just that the encoding is different than the desired default ISO 6937+. 
It's the missing initial transformation of `0x0A` into `0x8A` that seems to be the core issue.

I have notified our TV station, but the issue happens somewhere in a satellite feed provider, so hopefully they contacted them. 
Seeing as it happens in Germany as well, they might use the same satellite provider, or multiple providers use the HW/SW with the same issue.

lelegard, 2023-06-26:
> A local TV station seems to provide EPG data using UTF-8

Using UTF-8 in DVB strings is perfectly legal. However, according to ETSI EN 300 468, the binary encoding shall start with 0x15. This initial byte indicates that the rest of the binary encoding is UTF-8, where 0x0A is a valid encoding for a line break. If your EPG strings start with 0x15, then this is fine. Otherwise, this is a bug.

My previous post mentioned that 0x0A is invalid in _"single byte encodings"_, which include  ISO 8859-9, the encoding which is used in the provided EIT.

I am not surprised. I have seen so many bugs in the signalization. DVB strings are complicated and many broadcasters like their own inhouse tiny software development, from programmers who never read the DVB spec, instead of using validated professional DVB tools.

That being said, the DVB spec 300 468 is perfectly defined, as most DVB specs. This is not the case in other standards such as ISDB where most descriptors syntax are copied from DVB, but using other encodings for strings, not even defined in the standards, with different implementations per country. The Japanese encoding is complicated but well-defined. In all other ISDB countries, folklore prevails...

closed 2023-11-26 by github-actions[bot]