tsduck/tsduck #1240: Run out of storage: Can we install TSP only

picatchoof, 2023-06-27:
Hi all.
Please can some one advise how can I install minimum of the Tools of TSDuck.
the Embedded Industrial Computer may run out memory So I plan to install TSP Only.
Thnx

lelegard, 2023-06-28:
Hi @picatchoof 

>  may run out memory

Memory or disk storage?

TSDuck was designed as a generic flexible lab tool, for experimentation, typically on desktop systems. There is no plan to shrink it for constrained embedded devices.

On a Raspberry Pi, Arm64 Ubuntu system, after `make install-tools` in a dedicated area, the total installation space is 21 MB.

- Most of the code is in the shared library `libtsduck.so`, which uses 17 MB, 80% of the total disk space. You cannot remove it, nothing would work.
- All executables and plugins are very small: from 67 kB to 132 kB.
  - You can't remove the plugins, unless you really do not need some.
  - You can remove all executables but tsp (35 files, 2.8 MB).
  - However, each of them represent 0.3% to 0.6% of the total. Selecting them for removal may not be worth the effort.
- You can't remove the various configuration files, `.xml` and `.names`, they are required.

Stripping all executable and shared libraries removes a few kB only.

picatchoof, 2023-06-28:
Hi all.
It's good enough to convince me to recover memory elsewhere rather than looking to reduce the TSDuck installation.
Appreciate your quick responses.

Thanx 🙏 a lot

closed 2023-07-06 by lelegard