tsduck/tsduck #1248: Want to out different pid on different IP and Port

prashant-09-tech, 2023-07-03:
Hi,
@lelegard 

I am  receiving MPST unicast stream and  I want to mapping pid  with Ip and port .

Currently I am using this command

tsp -I ip 10201 -P filter --pid 101 --pid 201  -P fork "tsp -P pcrbitrate -P regulate -O ip 127.0.0.1:10207" -P pcrbitrate -P regulate -O ip 127.0.0.1:10206

So, please suggest Which plugin is useful for it.

lelegard, 2023-07-03:
> I am receiving MPST unicast stream and I want to mapping pid with Ip and port .

Please precisely describe what you want to achieve. Is this sending each PID individually to a distinct IP/port? Your command sends the same 2 PIDs on two distinct ports, meaning the two ports receive the same thing.

lelegard, 2023-07-03:
> Yes I want to send each PID individually to distinct Ip and port .

Then, do the filtering on each branch.
~~~
tsp -I ip 10201 \
    -P fork "tsp -P filter --pid 101 -O ip 127.0.0.1:10207" \
    -P filter --pid 201 -O ip 127.0.0.1:10206
~~~

Two points:
1. You can use `-P pcrbitrate` only when you have PCR's. Using it after filtering PID's is useless, unless you are certain that the filtered PID's contain PCR's.
2. As explained many times here and in the user's guide, the plugin `regulate` is only useful to transform an unregulated source (for instance a file) into a regulated output (for instance IP). Using `regulate` on a source which is already regulated is useless at best and counterproductive at worst.

closed 2023-07-03 by prashant-09-tech